PX4/PX4-Autopilot #14805: "ekf2 stop" get stuck in SITL

airopti, 2020-05-01:
**Describe the bug**
I am running SITL, when trying to stop the EKF by `ekf2 stop`, the Gazebo simulation get stuck and there is no any feedback information.  

**System**
- Ubuntu 16.04
- ROS Kinetic
- Gazebo 9

**To Reproduce**
Steps to reproduce the behavior:
1. `roslaunch px4 mavros_posix_sitl.launch`
2. After started the Gazebo simulation, run `ekf2 status`, can get some feedback
3. But, when run `ekf2 stop`, the simulation get stuck and there is no any feedback information.  

[image]

- [log file](https://logs.px4.io/plot_app?log=96ba2306-fdba-489b-b484-4a021fb71a70)
- I have tried Gazebo 7 , 8 and 9, but they all get stuck when running `ekf2 stop`

dagar, 2020-05-01:
@airopti this is because by default PX4 SITL is running lockstep with the simulator, and one of the mechanisms for doing that is waiting for simulated sensor data to make a round trip through the system and go back out as actuator controls.

airopti, 2020-05-02:
But the waiting time seems to be infinite, it just get stuck there, is it normal?

julianoes, 2020-05-04:
@airopti yes that's expected.

Alternatively, you can try without lockstep (see https://dev.px4.io/master/en/simulation/#disable-lockstep-simulation).

airopti, 2020-05-04:
Thank you so much, this solved my problem.

closed 2020-05-04 by airopti