PX4/PX4-Autopilot #14807: airspeed_selector noisy startup

dagar, 2020-05-01:
``` Console
INFO  [airspeed_selector] No airspeed sensor detected. Switch to non-airspeed mode.
WARN  [airspeed_selector] Airspeed: switched from sensor 0 to -1

WARN  [airspeed_selector] Airspeed: switched from sensor -1 to 1
```

@sfuhrer let's adjust this so there aren't unnecessary warnings in good setups.

Antiheavy, 2020-05-01:
keep all that information as INFO for the logs though!

dagar, 2020-05-01:
> keep all that information as INFO for the logs though!

To clarify, I only want to get rid of that unnecessary output on startup. The airspeed selector should be able to cleanly initialize into airspeed mode at first unless the sensors are actually missing.

sfuhrer, 2020-05-02:
Agree, that's annoying, not sure when it started. @dagar have you seen it also outside of SITL? 
https://github.com/PX4/Firmware/pull/14811 would be a potential fix, though a bit hacky.

sfuhrer, 2020-05-07:
Let's close this, as it has been fixed with https://github.com/PX4/Firmware/pull/14811

closed 2020-05-07 by sfuhrer