PX4/PX4-Autopilot #14813: Compile Error [make px4_fmu-v5_default failed]

sarlmolapple, 2020-05-03:
OS:centos,also happens on  v1.10.0 stable

[root@VM_0_15_centos Firmware]# make px4_fmu-v5_default
-- PX4 version: v1.11.0-beta1-140-g6dfe12d
-- PX4 config file: /root/Firmware/boards/px4/fmu-v5/default.cmake
-- PX4 config: px4_fmu-v5_default
-- PX4 platform: nuttx
-- cmake build type: MinSizeRel
-- The CXX compiler identification is GNU 7.2.1
-- The C compiler identification is GNU 7.2.1
-- The ASM compiler identification is GNU
-- Found assembler: /root/gcc-arm-none-eabi-7-2017-q4-major/bin/arm-none-eabi-gcc
-- Check for working CXX compiler using: Ninja
-- Check for working CXX compiler using: Ninja -- works

[224/224] Creating /root/Firmware/build/px4_fmu-v5_default/external/Build/px4io_firmware/px4_io-v2_default.px4
[326/1354] Performing install step for 'px4io_firmware'
FAILED: cd /root/Firmware/build/px4_fmu-v5_default/external/Build/px4io_firmware && USES_TERMINAL_BUILD true && /usr/local/bin/cmake -E touch /root/Firmware/build/px4_fmu-v5_default/external/Stamp/px4io_firmware/px4io_firmware-install
/bin/sh: USES_TERMINAL_BUILD: 未找到命令
[326/1354] Generating nuttx/sched/libsched.a
ninja: build stopped: subcommand failed.
make: *** [px4_fmu-v5_default] 错误 1

julianoes, 2020-05-04:
Just translating for anyone looking at this:

> USES_TERMINAL_BUILD: 未找到命令

This means: USES_TERMINAL_BUILD: command not found.

ibrahim0v, 2020-10-08:
Did you solve this?

ibrahim0v, 2020-10-16:
I solved it by reinstalling the toolchain