PX4/PX4-Autopilot #14820: v1.10.1 px4_fmu-v3 compile error: missing CONFIG_STM32_STM32F4XXX=y on .config

iQyan, 2020-05-04:
**Describe the bug**
A clear and concise description of the bug.

**To Reproduce**
Steps to reproduce the behavior:
1. Install Ubuntu 20.04 on VMWare for Mac.
2. open terminal
3. sudo apt-get install make
4. sudo apt-get install make
5. sudo apt-get install git
6. git clone px4 repository.
7.  cd ./Firmware
8. git submodule init
9. git submodule update --recursive
10. cd ./Tools/setup
11. source ubuntu.sh
12. logout and login again.
13. cd ./Firmware
14. git checkout v1.10.1
15. git submodule update --recursive 
16. make distclean
17. make px4_fmu-v3
(if make px4_sitl, then it compiles successfully)

**Expected behavior**
Compile

**Log Files and Screenshots**
make px4_fmu-v3 upload
-- PX4 version: v1.10.1
-- PX4 config file: /home/user/Documents/Firmware/boards/px4/fmu-v3/default.cmake
-- PX4 config: px4_fmu-v3_default
-- PX4 platform: nuttx
-- PX4 lockstep: disabled
-- cmake build type: MinSizeRel
-- The CXX compiler identification is GNU 7.2.1
-- The C compiler identification is GNU 7.2.1
-- The ASM compiler identification is GNU
-- Found assembler: /opt/gcc-arm-none-eabi-7-2017-q4-major/bin/arm-none-eabi-gcc
-- Check for working CXX compiler: /opt/gcc-arm-none-eabi-7-2017-q4-major/bin/arm-none-eabi-g++
-- Check for working CXX compiler: /opt/gcc-arm-none-eabi-7-2017-q4-major/bin/arm-none-eabi-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /opt/gcc-arm-none-eabi-7-2017-q4-major/bin/arm-none-eabi-gcc
-- Check for working C compiler: /opt/gcc-arm-none-eabi-7-2017-q4-major/bin/arm-none-eabi-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- ccache enabled (export CCACHE_DISABLE=1 to disable)
-- Found PythonInterp: /usr/bin/python3.8 (found version "3.8.2") 
-- Found PY_jinja2: /home/user/.local/lib/python3.8/site-packages/jinja2  
CMake Error at platforms/nuttx/cmake/px4_impl_os.cmake:114 (message):
  Could not determine chip architecture from NuttX config.  You may have to
  add it.
Call Stack (most recent call first):
  CMakeLists.txt:282 (px4_os_determine_build_chip)

------------------------------------------------------
I digged out this issue step by step and found why the error was occurred.

on a file px4_impl_os.cmake,

	function(px4_os_determine_build_chip)

	# determine chip and chip manufacturer based on NuttX config
	if (CONFIG_STM32_STM32F10XX)
		set(CHIP_MANUFACTURER "stm")
		set(CHIP "stm32f1")
	elseif(CONFIG_STM32_STM32F30XX)
		set(CHIP_MANUFACTURER "stm")
		set(CHIP "stm32f3")
	elseif(**CONFIG_STM32_STM32F4XXX**)
		set(CHIP_MANUFACTURER "stm")
		set(CHIP "stm32f4")
	elseif(CONFIG_ARCH_CHIP_STM32F7)
		set(CHIP_MANUFACTURER "stm")
		set(CHIP "stm32f7")
	elseif(CONFIG_ARCH_CHIP_MK66FN2M0VMD18)
		set(CHIP_MANUFACTURER "nxp")
		set(CHIP "k66")
	else()
		message(FATAL_ERROR "Could not determine chip architecture from NuttX config. You may have to add it.")
	endif()

	set(PX4_CHIP ${CHIP} CACHE STRING "PX4 Chip" FORCE)
	set(PX4_CHIP_MANUFACTURER ${CHIP_MANUFACTURER} CACHE STRING "PX4 Chip Manufacturer" FORCE)
	endfunction()

it should have been configured CONFIG_STM32_STM32F4XXX somewhere.
configuration of the file should be shown in .config file as below. (it appears in v1.11.0-beta1)
`CONFIG_STM32_STM32F4XXX=y`

However, on v1.10.1 version, there is no CONFIG_STM32_STM32F4XXX settings on .config file.
It seems that .config file is automatically generated, and it is related to nuttxOS.
But I don't know how to set CONFIG_STM32_STM32F4XXX on v1.10.1.

Interestingly, on v1.11.0-beta1, all is doing well and no issue at all. 
could it be the problem of inconsistency of submodules?(e.g. nuttx OS version etc)

Here is the current submodule information.
git submodule
-3bd51e67e022ce59644d33ebdf6570c2ea88ddb6 Tools/jMAVSim
-169d48217df89922e9fae72ef34fa46ce2e209dd Tools/sitl_gazebo
-08fd0a73045346448adf6969660196228b23e1fa boards/atlflight/cmake_hexagon
-2e5f9d6768b1dbffc006dc2ceeb2bfe120f22163 cmake/configs/uavcan_board_ident
-086f176b677b719c1f5b4267a10553ffc5c3a974 mavlink/include/mavlink/v2.0
-7e446a9976916a7b6fc7266098c67fc6f73a76e0 msg/tools/gencpp
-5736b1f7ad037fb5811a3100ba9da2db0ec1f20a msg/tools/genmsg
 13eeff0ec3f23d84fabf1e9fe216846957ad15cf platforms/nuttx/NuttX/apps (nuttx-7.29)
 427238133be2b0ecd068a11e886ee8fdbc31f6dc platforms/nuttx/NuttX/nuttx (nuttx-7.29-21-g427238133b)
-69e38ecc846df03f0bb71aa136bcc830dd3316a9 src/drivers/gps/devices
-6174b8c10a2dbf47076ca7a7b5820a4c36c8a988 src/drivers/uavcan/libuavcan
-c7d2fe2d4f5954b9042af4edf7e5aeb21dbdd129 src/lib/DriverFramework
-dec1ed0aeb395a885b60b1d61cfb57bd932c7d34 src/lib/ecl
-92d1c8761ec8e5dbe9ba3f756cebd5e90956a1b2 src/lib/matrix
-62d95c870eafb0cccc7bf70bb8a0cbb86f125a0e src/modules/micrortps_bridge/micro-CDR

iQyan, 2020-05-07:
Problem Solved.
You need **Ubuntu 18.04** for PX4 v1.10.1
I follow exactly the same process in Ubuntu 18.04 (VMware) and it works fine!

closed 2020-05-07 by iQyan

feemi, 2020-05-07:
Same error happens in mac OS Catalina for me. It is started after i installed python 3.8.2 package. i see the same version of python in log file.

psp316r, 2020-05-23:
I have the same issue ? any Solution 

-- PX4 version: v1.11.0-beta2-38-g951969ba00
-- PX4 config file: /home/admin1/Desktop/Testing 2/Firmware/boards/px4/fmu-v2/default.cmake
-- PX4 config: px4_fmu-v2_default
-- PX4 platform: nuttx
/home/admin1/Desktop/Testing 2/Firmware/build/px4_fmu-v2_default/NuttX/nuttx/tools/link.sh: line 100: [: /home/admin1/Desktop/Testing: binary operator expected
Makefile:83: /home/admin1/Desktop/Testing: No such file or directory
Makefile:83: 2/Firmware/build/px4_fmu-v2_default/NuttX/apps/Application.mk: No such file or directory
make[3]: *** No rule to make target '2/Firmware/build/px4_fmu-v2_default/NuttX/apps/Application.mk'.  Stop.
make[2]: *** [dirlinks] Error 2
make[1]: *** [dirlinks] Error 2
-- cmake build type: MinSizeRel
-- The CXX compiler identification is GNU 7.2.1
-- The C compiler identification is GNU 7.2.1
-- The ASM compiler identification is GNU
-- Found assembler: /home/admin1/gcc-arm-none-eabi-7-2017-q4-major/bin/arm-none-eabi-gcc
-- Check for working CXX compiler: /home/admin1/gcc-arm-none-eabi-7-2017-q4-major/bin/arm-none-eabi-g++
-- Check for working CXX compiler: /home/admin1/gcc-arm-none-eabi-7-2017-q4-major/bin/arm-none-eabi-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /home/admin1/gcc-arm-none-eabi-7-2017-q4-major/bin/arm-none-eabi-gcc
-- Check for working C compiler: /home/admin1/gcc-arm-none-eabi-7-2017-q4-major/bin/arm-none-eabi-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- ccache enabled (export CCACHE_DISABLE=1 to disable)
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.6.9", minimum required is "3") 
CMake Error at platforms/nuttx/cmake/px4_impl_os.cmake:128 (message):
  Could not determine chip architecture from NuttX config.  You may have to
  add it.
Call Stack (most recent call first):
  CMakeLists.txt:290 (px4_os_determine_build_chip)

-- Configuring incomplete, errors occurred!
See also "/home/admin1/Desktop/Testing 2/Firmware/build/px4_fmu-v2_default/CMakeFiles/CMakeOutput.log".
See also "/home/admin1/Desktop/Testing 2/Firmware/build/px4_fmu-v2_default/CMakeFiles/CMakeError.log".
Error: /home/admin1/Desktop/Testing 2/Firmware/build/px4_fmu-v2_default is not a directory
Makefile:205: recipe for target 'px4_fmu-v2' failed
make: *** [px4_fmu-v2] Error 1

I am getting this error...

julianoes, 2020-05-25:
@psp316r please create a new issue. I don't think your error is the same one.

wjsyz, 2023-07-13:
Try
make distclean
make px4_fmu-v3_default

Mendeler, 2023-12-04:
> Try make distclean make px4_fmu-v3_default

It works!!