PX4/PX4-Autopilot #14824: java.lang.UnsupportedClassVersionError

Bert-Tec, 2020-05-04:
**Describe the bug**
I get an java.lang.UnsupportedClassVersionError Error starting jmavsim via make px4_sitl_default jmavsim. I made sure that I use JDK version 8. I also tried that:
```
sudo apt install openjdk-8-jdk
sudo update-alternatives --config java # choose 8
rm -rf Tools/jMAVSim/out
```
Tried several times to delete all and download and install again. I followed the master version of the docs.

**To Reproduce**
Steps to reproduce the behavior:
`make px4_sitl_default jmavsim`

**Expected behavior**
px4 os connects to sim and sim starts up.

**Log Files and Screenshots**

```
 make px4_sitl_default jmavsim
-- PX4 version: v1.11.0-beta1-151-g8e2c52a31a
-- PX4 config file: /home/xxx/sim/px4/Firmware/boards/px4/sitl/default.cmake
-- PX4 config: px4_sitl_default
-- PX4 platform: posix
-- PX4 lockstep: enabled
-- cmake build type: RelWithDebInfo
-- The CXX compiler identification is GNU 7.5.0
-- The C compiler identification is GNU 7.5.0
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/cc
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Building for code coverage
-- ccache enabled (export CCACHE_DISABLE=1 to disable)
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.6.9", minimum required is "3") 
-- build type is RelWithDebInfo
-- PX4 ECL: Very lightweight Estimation & Control Library v1.9.0-rc1-274-gcda7486
-- Configuring done
-- Generating done
-- Build files have been written to: /home/xxx/sim/px4/Firmware/build/px4_sitl_default
[0/738] git submodule src/drivers/gps/devices
[1/738] git submodule src/lib/ecl
[10/738] git submodule mavlink/include/mavlink/v2.0
[13/738] git submodule Tools/jMAVSim
[737/738] cd /home/xxx/sim/px4/Firmware/build/px4_sitl_default/tmp && /home/xxx/s...none /home/xxx/sim/px4/Firmware /home/xxx/sim/px4/Firmware/build/px4_sitl_default
SITL ARGS
sitl_bin: /home/xxx/sim/px4/Firmware/build/px4_sitl_default/bin/px4
debugger: none
program: jmavsim
model: none
world: none
src_path: /home/xxx/sim/px4/Firmware
build_path: /home/xxx/sim/px4/Firmware/build/px4_sitl_default
empty model, setting iris as default
SITL COMMAND: "/home/xxx/sim/px4/Firmware/build/px4_sitl_default/bin/px4" "/home/xxx/sim/px4/Firmware"/ROMFS/px4fmu_common -s etc/init.d-posix/rcS -t "/home/xxx/sim/px4/Firmware"/test_data
INFO  [px4] Creating symlink /home/xxx/sim/px4/Firmware/ROMFS/px4fmu_common -> /home/xxx/sim/px4/Firmware/build/px4_sitl_default/tmp/rootfs/etc

______  __   __    ___ 
| ___ \ \ \ / /   /   |
| |_/ /  \ V /   / /| |
|  __/   /   \  / /_| |
| |     / /^\ \ \___  |
\_|     \/   \/     |_/

px4 starting.

INFO  [px4] Calling startup script: /bin/sh etc/init.d-posix/rcS 0
INFO  [param] selected parameter default file eeprom/parameters_10016
[param] parameter file not found, creating eeprom/parameters_10016
  SYS_AUTOCONFIG: curr: 0 -> new: 1
* SYS_AUTOSTART: curr: 0 -> new: 10016
  BAT_N_CELLS: curr: 0 -> new: 3
  CAL_ACC0_ID: curr: 0 -> new: 1311244
  CAL_ACC_PRIME: curr: 0 -> new: 1311244
  CAL_GYRO0_ID: curr: 0 -> new: 2294028
  CAL_GYRO_PRIME: curr: 0 -> new: 2294028
  CAL_MAG0_ID: curr: 0 -> new: 197388
  CAL_MAG_PRIME: curr: 0 -> new: 197388
  COM_DISARM_LAND: curr: 2.0000 -> new: 0.5000
  COM_OBL_ACT: curr: 0 -> new: 2
  COM_RC_IN_MODE: curr: 0 -> new: 1
  EKF2_ANGERR_INIT: curr: 0.1000 -> new: 0.0100
  EKF2_GBIAS_INIT: curr: 0.1000 -> new: 0.0100
  COM_ARM_EKF_AB: curr: 0.0017 -> new: 0.0050
  EKF2_REQ_GPS_H: curr: 10.0000 -> new: 0.5000
  MC_PITCH_P: curr: 6.5000 -> new: 6.0000
  MC_PITCHRATE_P: curr: 0.1500 -> new: 0.2000
  MC_ROLL_P: curr: 6.5000 -> new: 6.0000
  MC_ROLLRATE_P: curr: 0.1500 -> new: 0.2000
  MPC_HOLD_MAX_Z: curr: 0.6000 -> new: 2.0000
  MPC_Z_VEL_I: curr: 0.1000 -> new: 0.1500
  MPC_Z_VEL_P: curr: 0.2000 -> new: 0.6000
  MPC_XY_P: curr: 0.9500 -> new: 0.8000
  MPC_XY_VEL_P: curr: 0.0900 -> new: 0.2000
  MPC_XY_VEL_D: curr: 0.0100 -> new: 0.0160
  MPC_SPOOLUP_TIME: curr: 1.0000 -> new: 0.5000
  MPC_TKO_RAMP_T: curr: 3.0000 -> new: 1.0000
  NAV_ACC_RAD: curr: 10.0000 -> new: 2.0000
  NAV_DLL_ACT: curr: 0 -> new: 2
  RTL_DESCEND_ALT: curr: 30.0000 -> new: 5.0000
  RTL_LAND_DELAY: curr: -1.0000 -> new: 5.0000
  RTL_RETURN_ALT: curr: 60.0000 -> new: 30.0000
  SDLOG_MODE: curr: 0 -> new: 1
  SDLOG_PROFILE: curr: 3 -> new: 131
  SDLOG_DIRS_MAX: curr: 0 -> new: 7
  SENS_BOARD_X_OFF: curr: 0.0000 -> new: 0.0000
  SENS_DPRES_OFF: curr: 0.0000 -> new: 0.0010
  TRIG_INTERFACE: curr: 4 -> new: 3
  COM_CPU_MAX: curr: 90.0000 -> new: -1.0000
* RTL_DESCEND_ALT: curr: 5.0000 -> new: 10.0000
* RTL_LAND_DELAY: curr: 5.0000 -> new: 0.0000
  PWM_MAX: curr: 2000 -> new: 1950
  PWM_MIN: curr: 1000 -> new: 1075
  GPS_UBX_DYNMODEL: curr: 7 -> new: 6
* SYS_AUTOCONFIG: curr: 1 -> new: 0
INFO  [dataman] Unknown restart, data manager file './dataman' size is 11798680 bytes
INFO  [simulator] Waiting for simulator to accept connection on TCP port 4560
Buildfile: /home/xxx/sim/px4/Firmware/Tools/jMAVSim/build.xml

make_dirs:
    [mkdir] Created dir: /home/xxx/sim/px4/Firmware/Tools/jMAVSim/out/production/jMAVSim

compile:
    [javac] Compiling 76 source files to /home/xxx/sim/px4/Firmware/Tools/jMAVSim/out/production/jMAVSim

create_run_jar:
      [jar] Building jar: /home/xxx/sim/px4/Firmware/Tools/jMAVSim/out/production/jmavsim_run.jar

copy_res:
     [copy] Copying 17 files to /home/xxx/sim/px4/Firmware/Tools/jMAVSim/out/production

BUILD SUCCESSFUL
Total time: 6 seconds
Exception in thread "main" java.lang.UnsupportedClassVersionError: me/drton/jmavsim/Simulator has been compiled by a more recent version of the Java Runtime (class file version 57.0), this version of the Java Runtime only recognizes class file versions up to 52.0
	at java.lang.ClassLoader.defineClass1(Native Method)
	at java.lang.ClassLoader.defineClass(ClassLoader.java:756)
	at java.security.SecureClassLoader.defineClass(SecureClassLoader.java:142)
	at java.net.URLClassLoader.defineClass(URLClassLoader.java:468)
	at java.net.URLClassLoader.access$100(URLClassLoader.java:74)
	at java.net.URLClassLoader$1.run(URLClassLoader.java:369)
	at java.net.URLClassLoader$1.run(URLClassLoader.java:363)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.net.URLClassLoader.findClass(URLClassLoader.java:362)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:418)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:351)
	at java.lang.Class.forName0(Native Method)
	at java.lang.Class.forName(Class.java:348)
	at org.eclipse.jdt.internal.jarinjarloader.JarRsrcLoader.main(JarRsrcLoader.java:59)
^CERROR [px4] Startup script returned with return value: 2
/home/xxx/sim/px4/Firmware/Tools/sitl_run.sh: line 174: kill: (10905) - No such process
ninja: build stopped: interrupted by user.
Makefile:198: recipe for target 'px4_sitl_default' failed
make: *** [px4_sitl_default] Interrupt
```

**Additional context**
I saw other related issues here, but they didn't helped me.
The log is from a brand new installation, so /out was not existing. Also tried to clean /out at the installations before, but became the same error.

julianoes, 2020-05-05:
And this happens, also if you do `rm -rf Tools/jMAVSim/out` and try again?

Bert-Tec, 2020-05-05:
yes, I tried it again and made sure /out does not exists, see the output following: 
```
xxx@HP-ProBook-450-G5:~/sim/px4/Firmware$ sudo update-alternatives --config java 
There are 3 choices for the alternative java (providing /usr/bin/java).

  Selection    Path                                            Priority   Status
------------------------------------------------------------
  0            /usr/lib/jvm/java-11-openjdk-amd64/bin/java      1111      auto mode
  1            /usr/lib/jvm/java-11-openjdk-amd64/bin/java      1111      manual mode
* 2            /usr/lib/jvm/java-8-openjdk-amd64/jre/bin/java   1081      manual mode
  3            /usr/lib/jvm/jdk-13.0.2/bin/java                 2         manual mode

Press <enter> to keep the current choice[*], or type selection number: 2
xxx@HP-ProBook-450-G5:~/sim/px4/Firmware$ rm -rf Tools/jMAVSim/out
xxx@HP-ProBook-450-G5:~/sim/px4/Firmware$ ls Tools/jMAVSim/
art            build.xml    fix_style.sh           jMAVlib      jMAVSim.ipr  LICENSE  models     src
astylerc_java  environment  jar-in-jar-loader.zip  jMAVSim.iml  lib          mavlink  README.md  tools
xxx@HP-ProBook-450-G5:~/sim/px4/Firmware$ make px4_sitl jmavsim
[0/1] cd /home/xxx/sim/px4/Firmware/build/px4_sitl_default/tmp &...m/px4/Firmware /home/xxx/sim/px4/Firmware/build/px4_sitl_default
SITL ARGS
sitl_bin: /home/xxx/sim/px4/Firmware/build/px4_sitl_default/bin/px4
debugger: none
program: jmavsim
model: none
world: none
src_path: /home/xxx/sim/px4/Firmware
build_path: /home/xxx/sim/px4/Firmware/build/px4_sitl_default
empty model, setting iris as default
SITL COMMAND: "/home/xxx/sim/px4/Firmware/build/px4_sitl_default/bin/px4" "/home/xxx/sim/px4/Firmware"/ROMFS/px4fmu_common -s etc/init.d-posix/rcS -t "/home/xxx/sim/px4/Firmware"/test_data
INFO  [px4] Creating symlink /home/xxx/sim/px4/Firmware/ROMFS/px4fmu_common -> /home/xxx/sim/px4/Firmware/build/px4_sitl_default/tmp/rootfs/etc

______  __   __    ___ 
| ___ \ \ \ / /   /   |
| |_/ /  \ V /   / /| |
|  __/   /   \  / /_| |
| |     / /^\ \ \___  |
\_|     \/   \/     |_/

px4 starting.

INFO  [px4] Calling startup script: /bin/sh etc/init.d-posix/rcS 0
INFO  [param] selected parameter default file eeprom/parameters_10016
[param] parameter file not found, creating eeprom/parameters_10016
  SYS_AUTOCONFIG: curr: 0 -> new: 1
* SYS_AUTOSTART: curr: 0 -> new: 10016
  BAT_N_CELLS: curr: 0 -> new: 3
  CAL_ACC0_ID: curr: 0 -> new: 1311244
  CAL_ACC_PRIME: curr: 0 -> new: 1311244
  CAL_GYRO0_ID: curr: 0 -> new: 2294028
  CAL_GYRO_PRIME: curr: 0 -> new: 2294028
  CAL_MAG0_ID: curr: 0 -> new: 197388
  CAL_MAG_PRIME: curr: 0 -> new: 197388
  COM_DISARM_LAND: curr: 2.0000 -> new: 0.5000
  COM_OBL_ACT: curr: 0 -> new: 2
  COM_RC_IN_MODE: curr: 0 -> new: 1
  EKF2_ANGERR_INIT: curr: 0.1000 -> new: 0.0100
  EKF2_GBIAS_INIT: curr: 0.1000 -> new: 0.0100
  COM_ARM_EKF_AB: curr: 0.0017 -> new: 0.0050
  EKF2_REQ_GPS_H: curr: 10.0000 -> new: 0.5000
  MC_PITCH_P: curr: 6.5000 -> new: 6.0000
  MC_PITCHRATE_P: curr: 0.1500 -> new: 0.2000
  MC_ROLL_P: curr: 6.5000 -> new: 6.0000
  MC_ROLLRATE_P: curr: 0.1500 -> new: 0.2000
  MPC_HOLD_MAX_Z: curr: 0.6000 -> new: 2.0000
  MPC_Z_VEL_I: curr: 0.1000 -> new: 0.1500
  MPC_Z_VEL_P: curr: 0.2000 -> new: 0.6000
  MPC_XY_P: curr: 0.9500 -> new: 0.8000
  MPC_XY_VEL_P: curr: 0.0900 -> new: 0.2000
  MPC_XY_VEL_D: curr: 0.0100 -> new: 0.0160
  MPC_SPOOLUP_TIME: curr: 1.0000 -> new: 0.5000
[... 35 lines cut ...]
BUILD SUCCESSFUL
Total time: 2 seconds
Exception in thread "main" java.lang.UnsupportedClassVersionError: me/drton/jmavsim/Simulator has been compiled by a more recent version of the Java Runtime (class file version 57.0), this version of the Java Runtime only recognizes class file versions up to 52.0
	at java.lang.ClassLoader.defineClass1(Native Method)
	at java.lang.ClassLoader.defineClass(ClassLoader.java:756)
	at java.security.SecureClassLoader.defineClass(SecureClassLoader.java:142)
	at java.net.URLClassLoader.defineClass(URLClassLoader.java:468)
	at java.net.URLClassLoader.access$100(URLClassLoader.java:74)
	at java.net.URLClassLoader$1.run(URLClassLoader.java:369)
	at java.net.URLClassLoader$1.run(URLClassLoader.java:363)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.net.URLClassLoader.findClass(URLClassLoader.java:362)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:418)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:351)
	at java.lang.Class.forName0(Native Method)
	at java.lang.Class.forName(Class.java:348)
	at org.eclipse.jdt.internal.jarinjarloader.JarRsrcLoader.main(JarRsrcLoader.java:59)
^CERROR [px4] Startup script returned with return value: 2
/home/xxx/sim/px4/Firmware/Tools/sitl_run.sh: line 174: kill: (18788) - No such process
ninja: build stopped: interrupted by user.
Makefile:205: recipe for target 'px4_sitl' failed
make: *** [px4_sitl] Interrupt

xxx@HP-ProBook-450-G5:~/sim/px4/Firmware$ 
```

julianoes, 2020-05-05:
Alright, I have to believe you :smile:. Is this on Ubuntu 20.04?

Bert-Tec, 2020-05-05:
Ubuntu 18.04, sorry forget to mention.

julianoes, 2020-05-05:
I just tried it in Ubuntu 18.04 and Ubuntu 20.04 inside VirtualBox and it worked fine.

That's with:
```
java -version
openjdk version "1.8.0_252"
OpenJDK Runtime Environment (build 1.8.0_252-8u252-b09-1~18.04-b09)
OpenJDK 64-Bit Server VM (build 25.252-b09, mixed mode)

uname -a
Linux vbox-ubuntu1804 4.15.0-99-generic #100-Ubuntu SMP Wed Apr 22 20:32:56 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux
```

Bert-Tec, 2020-05-06:
Thanks @julianoes 
Not sure what the problem was, but I tried it with a fresh install of ubuntu 18.04 and it worked.

closed 2020-05-06 by Bert-Tec