PX4/PX4-Autopilot #14829: accel calibration garbage mavlink output on completion

dagar, 2020-05-04:
[image]

MaEtUgR, 2020-05-05:
I've seen this output before but only much later than the calibration so I didn't know where they were coming from. But now it makes sense I recalibrated some time before seeing them so I can confirm. 👍

MaEtUgR, 2020-08-19:
Looks like QGC is handling the messages reported with https://github.com/PX4/Firmware/blob/7b4ae7c6586bcf355beda3741850a0b2c8532cb4/src/modules/commander/accelerometer_calibration.cpp#L335 but for some reason the end gets put out to the user. Line number will change after https://github.com/PX4/Firmware/pull/15235 but presumably it's still there.

dagar, 2020-08-19:
Oh! It's simply the message spilling over. Thanks @MaEtUgR

MaEtUgR, 2020-08-19:
Ah right and before it just got cut instead of split so no one saw it 💡

dagar, 2020-08-19:
Fixed in https://github.com/PX4/Firmware/pull/15235.

https://github.com/PX4/Firmware/pull/15235/commits/6072d7deb0d6ea4ae8225f3d038387f610327c96

commit 6072d7de (2020-08-19): commander: accel calibration reduce status length

 - this keeps the string within a single Mavlink STATUSTEXT
 - fixes #14829
commit f4acc0d5 (2020-08-20): commander: accel calibration reduce status length

 - this keeps the string within a single Mavlink STATUSTEXT
 - fixes #14829
commit a4cf1a70 (2020-08-20): commander: accel calibration reduce status length

 - this keeps the string within a single Mavlink STATUSTEXT
 - fixes #14829
commit f38b10e3 (2020-08-20): commander: accel calibration reduce status length

 - this keeps the string within a single Mavlink STATUSTEXT
 - fixes #14829