PX4/PX4-Autopilot #14830: Firmware and airframe setup issue

amirmoeini, 2020-05-04:
Hello everyone,

I have a pixhawk 2.4.8 and I am trying to flash it with the latest version of QGC in ubuntu 18.04. 

But I am always getting the following error:

QGroundControl can upgrade the firmware on Pixhawk devices, SiK Radios and PX4 Flow Smart Cameras.
All QGroundControl connections to vehicles must be disconnected prior to firmware upgrade.
Please unplug your Pixhawk and/or Radio from USB.
Plug in your device via USB to start firmware upgrade.
Found device: PX4 FMU V2
Connected to bootloader:
Version: 5
Board ID: 255
Flash size: 2080768
Downloading firmware...
From: http://px4-travis.s3.amazonaws.com/Firmware/stable/px4fmu-v3_default.px4
Download complete
MAV_AUTOPILOT = 12
Successfully decompressed parameter_xml
Successfully decompressed airframe_xml
Successfully decompressed image
Erasing previous program...
Erase complete
Programming new version...
Error: Flash failed: Get Command Response: Timeout waiting for bytes to be available at address 0x000004c0
Upgrade cancelled

------------------------------------------
If upgrade failed, make sure to connect directly to a powered USB port on your computer, not through a USB hub. Also make sure you are only powered via USB not battery.

However, I can only flash the firmware from the command line! but after that, when I am setting the airframe the pixhawk does a reboot and when it is turned on it is asking me again to set the airframe and then reboot... . So I assume the parameters are not saved after the reboot and because of that I can never reach the sensor calibration section. Did any of you have any similar experience in the past? Any help would be appreciated.

Thanks

czp0204, 2023-03-06:
Did you fix this issue?

Bhaveshmeghwal21, 2024-02-14:
did you find any solution ??

julianoes, 2024-02-14:
Pixhawk 2.4.8 is not really something that is supported. It's mostly clones where your mileage might vary. Also, this issue is ancient, so time to close it, and not necrobump it.

closed 2024-02-14 by julianoes