PX4/PX4-Autopilot #14831: xformV declaration in gazebo_usv_dynamics_plugin.cpp

robin-shaun, 2020-05-05:
When `make px4_sitl_default gazebo`,  declaration error happens
`/home/robin/px4/1.11/Firmware/Tools/sitl_gazebo/src/gazebo_usv_dynamics_plugin.cpp:251:29: error: missing template arguments before ‘xformV’
     ignition::math::Matrix4 xformV(vq);
/home/robin/px4/1.11/Firmware/Tools/sitl_gazebo/src/gazebo_usv_dynamics_plugin.cpp:285:15: error: ‘xformV’ was not declared in this scope
       bpntW = xformV * bpnt;
` 
Looking into gazebo_usv_dynamics_plugin.cpp, it can be seen that for gazebo version >=8.0 (my Gazebo version is 9.13.0) , the declaration is ` ignition::math::Matrix4 xformV(vq);`, which misses template.
`#if GAZEBO_MAJOR_VERSION >= 8
    ignition::math::Matrix4 xformV(vq);
  #else
    ignition::math::Matrix4<double> xformV(vq);
  #endif`
And when I deleted if-else and changed it to `ignition::math::Matrix4<double> xformV(vq);`, `make px4_sitl_default gazebo` passed. So maybe the if-else can be removed.

xdwgood, 2020-05-05:
same issue : https://github.com/PX4/sitl_gazebo/issues/456

saengphet, 2020-05-05:
Is it the same issue? I have changed a line as mentioned in this and PASSED.
https://discuss.px4.io/t/error-about-gazebo-sitl-building/12019/10

robin-shaun, 2020-05-05:
> Is it the same issue? I have changed a line as mentioned in this and PASSED.
> https://discuss.px4.io/t/error-about-gazebo-sitl-building/12019/10

Yes, it's the same issue. And for the method [https://discuss.px4.io/t/error-about-gazebo-sitl-building/12019/10](https://discuss.px4.io/t/error-about-gazebo-sitl-building/12019/10) describes, if-else is useless and it can be removed.

Jaeyoung-Lim, 2020-05-21:
Fixed by https://github.com/PX4/sitl_gazebo/pull/492

closed 2020-05-21 by Jaeyoung-Lim