PX4/PX4-Autopilot #14835: intermittent SITL test failures

dagar, 2020-05-05:
- https://github.com/PX4/Firmware/pull/14834/checks?check_run_id=646212154
 
[image]

julianoes, 2020-05-05:
```
2020-05-05T13:56:15.2536746Z ==> Running tests for standard_vtol
2020-05-05T13:56:15.2537022Z --> Test case 1 of 2: 'Takeoff and Land' running ...
2020-05-05T13:56:15.2537168Z [px4         ] INFO  [param] selected parameter default file eeprom/parameters_1040
2020-05-05T13:56:15.2537323Z [px4         ] [param] parameter file not found, creating eeprom/parameters_1040
2020-05-05T13:56:15.2538028Z [px4         ]   SYS_AUTOCONFIG: curr: 0 -> new: 1
2020-05-05T13:56:15.2538465Z [px4         ] * SYS_AUTOSTART: curr: 0 -> new: 1040
2020-05-05T13:56:15.2538767Z [px4         ]   BAT_N_CELLS: curr: 0 -> new: 3
2020-05-05T13:56:15.2539047Z [px4         ]   CAL_ACC0_ID: curr: 0 -> new: 1311244
2020-05-05T13:56:15.2539340Z [px4         ]   CAL_ACC_PRIME: curr: 0 -> new: 1311244
2020-05-05T13:56:15.2539628Z [px4         ]   CAL_GYRO0_ID: curr: 0 -> new: 2294028
2020-05-05T13:56:15.2539917Z [px4         ]   CAL_GYRO_PRIME: curr: 0 -> new: 2294028
2020-05-05T13:56:15.2540199Z [px4         ]   CAL_MAG0_ID: curr: 0 -> new: 197388
2020-05-05T13:56:15.2540484Z [px4         ]   CAL_MAG_PRIME: curr: 0 -> new: 197388
2020-05-05T13:56:15.2540764Z [px4         ]   COM_DISARM_LAND: curr: 2.0000 -> new: 0.5000
2020-05-05T13:56:15.2541039Z [px4         ]   COM_OBL_ACT: curr: 0 -> new: 2
2020-05-05T13:56:15.2541314Z [px4         ]   COM_RC_IN_MODE: curr: 0 -> new: 1
2020-05-05T13:56:15.2541607Z [px4         ]   EKF2_ANGERR_INIT: curr: 0.1000 -> new: 0.0100
2020-05-05T13:56:15.2541909Z [px4         ]   EKF2_GBIAS_INIT: curr: 0.1000 -> new: 0.0100
2020-05-05T13:56:15.2542202Z [px4         ]   COM_ARM_EKF_AB: curr: 0.0017 -> new: 0.0050
2020-05-05T13:56:15.2542497Z [px4         ]   EKF2_REQ_GPS_H: curr: 10.0000 -> new: 0.5000
2020-05-05T13:56:15.2542875Z [px4         ]   MC_PITCH_P: curr: 6.5000 -> new: 6.0000
2020-05-05T13:56:15.2543170Z [px4         ]   MC_PITCHRATE_P: curr: 0.1500 -> new: 0.2000
2020-05-05T13:56:15.2543456Z [px4         ]   MC_ROLL_P: curr: 6.5000 -> new: 6.0000
2020-05-05T13:56:15.2543748Z [px4         ]   MC_ROLLRATE_P: curr: 0.1500 -> new: 0.2000
2020-05-05T13:56:15.2544042Z [px4         ]   MPC_HOLD_MAX_Z: curr: 0.6000 -> new: 2.0000
2020-05-05T13:56:15.2544333Z [px4         ]   MPC_Z_VEL_I: curr: 0.1000 -> new: 0.1500
2020-05-05T13:56:15.2544623Z [px4         ]   MPC_Z_VEL_P: curr: 0.2000 -> new: 0.6000
2020-05-05T13:56:15.2544894Z [px4         ]   MPC_XY_P: curr: 0.9500 -> new: 0.8000
2020-05-05T13:56:15.2545185Z [px4         ]   MPC_XY_VEL_P: curr: 0.0900 -> new: 0.2000
2020-05-05T13:56:15.2545480Z [px4         ]   MPC_XY_VEL_D: curr: 0.0100 -> new: 0.0160
2020-05-05T13:56:15.2545777Z [px4         ]   MPC_SPOOLUP_TIME: curr: 1.0000 -> new: 0.5000
2020-05-05T13:56:15.2546075Z [px4         ]   MPC_TKO_RAMP_T: curr: 3.0000 -> new: 1.0000
2020-05-05T13:56:15.2546365Z [px4         ]   NAV_ACC_RAD: curr: 10.0000 -> new: 2.0000
2020-05-05T13:56:15.2546640Z [px4         ]   NAV_DLL_ACT: curr: 0 -> new: 2
2020-05-05T13:56:15.2546932Z [px4         ]   RTL_DESCEND_ALT: curr: 30.0000 -> new: 5.0000
2020-05-05T13:56:15.2547228Z [px4         ]   RTL_LAND_DELAY: curr: -1.0000 -> new: 5.0000
2020-05-05T13:56:15.2547524Z [px4         ]   RTL_RETURN_ALT: curr: 60.0000 -> new: 30.0000
2020-05-05T13:56:15.2547800Z [px4         ]   SDLOG_MODE: curr: 0 -> new: 1
2020-05-05T13:56:15.2548078Z [px4         ]   SDLOG_PROFILE: curr: 3 -> new: 131
2020-05-05T13:56:15.2548356Z [px4         ]   SDLOG_DIRS_MAX: curr: 0 -> new: 7
2020-05-05T13:56:15.2548649Z [px4         ]   SENS_BOARD_X_OFF: curr: 0.0000 -> new: 0.0000
2020-05-05T13:56:15.2548933Z [px4         ]   SENS_DPRES_OFF: curr: 0.0000 -> new: 0.0010
2020-05-05T13:56:15.2549212Z [px4         ]   TRIG_INTERFACE: curr: 4 -> new: 3
2020-05-05T13:56:15.2549509Z [px4         ]   COM_CPU_MAX: curr: 90.0000 -> new: -1.0000
2020-05-05T13:56:15.2549643Z [px4         ] COM_DL_LOSS_T set to 200
2020-05-05T13:56:15.2549922Z [px4         ]   COM_DL_LOSS_T: curr: 10 -> new: 200
2020-05-05T13:56:15.2550054Z [px4         ] COM_RC_LOSS_T set to 10.0
2020-05-05T13:56:15.2550343Z [px4         ]   COM_RC_LOSS_T: curr: 0.5000 -> new: 10.0000
2020-05-05T13:56:15.2550474Z [px4         ] COM_OF_LOSS_T set to 10.0
2020-05-05T13:56:15.2550760Z [px4         ]   COM_OF_LOSS_T: curr: 0.5000 -> new: 10.0000
2020-05-05T13:56:15.2551044Z [px4         ]   IMU_GYRO_RATEMAX: curr: 0 -> new: 400
2020-05-05T13:56:15.2551339Z [px4         ]   MIS_TAKEOFF_ALT: curr: 2.5000 -> new: 20.0000
2020-05-05T13:56:15.2551632Z [px4         ]   MIS_YAW_TMT: curr: -1.0000 -> new: 10.0000
2020-05-05T13:56:15.2552000Z [px4         ]   MPC_ACC_HOR_MAX: curr: 5.0000 -> new: 2.0000
2020-05-05T13:56:15.2552292Z [px4         ]   MPC_TKO_SPEED: curr: 1.5000 -> new: 1.0000
2020-05-05T13:56:15.2552591Z [px4         ]   MPC_VEL_MANUAL: curr: 10.0000 -> new: 3.0000
2020-05-05T13:56:15.2552883Z [px4         ]   MPC_XY_CRUISE: curr: 5.0000 -> new: 3.0000
2020-05-05T13:56:15.2553179Z [px4         ]   MPC_XY_VEL_MAX: curr: 12.0000 -> new: 4.0000
2020-05-05T13:56:15.2553473Z [px4         ]   MPC_Z_VEL_MAX_DN: curr: 1.0000 -> new: 1.5000
2020-05-05T13:56:15.2553766Z [px4         ]   MPC_JERK_MAX: curr: 8.0000 -> new: 4.5000
2020-05-05T13:56:15.2554055Z [px4         ] * NAV_ACC_RAD: curr: 2.0000 -> new: 3.0000
2020-05-05T13:56:15.2554349Z [px4         ] * RTL_LAND_DELAY: curr: 5.0000 -> new: 0.0000
2020-05-05T13:56:15.2554623Z [px4         ]   RTL_TYPE: curr: 0 -> new: 1
[... cut ...]
2020-05-05T13:56:15.2579508Z      - 'Offboard position control': succeeded
2020-05-05T13:56:15.2579726Z   - iris_vision:
2020-05-05T13:56:15.2579972Z      - 'Offboard takeoff and land': succeeded
2020-05-05T13:56:15.2580207Z      - 'Offboard position control': succeeded
2020-05-05T13:56:15.2580433Z   - standard_vtol:
2020-05-05T13:56:15.2580669Z      - 'Takeoff and Land': failed
2020-05-05T13:56:15.2580926Z      - 'Fly square Multicopter Missions': cancelled
2020-05-05T13:56:15.2581046Z Overall result: FAIL
```

dagar, 2020-05-05:
https://github.com/PX4/Firmware/pull/14836/checks?check_run_id=646686108

dagar, 2020-05-06:
https://github.com/PX4/Firmware/runs/649658989

dagar, 2020-05-06:
https://github.com/PX4/Firmware/pull/14846/checks?check_run_id=649765804

dagar, 2020-05-06:
https://github.com/PX4/Firmware/runs/649760590

dagar, 2020-05-06:
https://github.com/PX4/Firmware/runs/650254506

dagar, 2020-05-06:
Environment permission issues? 

`Home directory not accessible: Permission denied`
[image]

dagar, 2020-05-06:
https://github.com/PX4/Firmware/pull/14755/checks?check_run_id=645956081

dagar, 2020-05-06:
[image]
https://github.com/PX4/Firmware/pull/14850/checks?check_run_id=650628730

dagar, 2020-05-06:
[image]

https://github.com/PX4/Firmware/pull/14504/checks?check_run_id=650483012

dagar, 2020-05-07:
[image]
https://github.com/PX4/Firmware/pull/14853/checks?check_run_id=651461557

dagar, 2020-05-07:
[image]

https://github.com/PX4/Firmware/runs/651891362

dagar, 2020-05-07:
[image]
https://github.com/PX4/Firmware/pull/14844/checks?check_run_id=652124079

dagar, 2020-05-07:
[image]
https://github.com/PX4/Firmware/pull/14844/checks?check_run_id=654078516

dagar, 2020-05-08:
https://github.com/PX4/Firmware/pull/14871/checks?check_run_id=656614110

dagar, 2020-05-08:
https://github.com/PX4/Firmware/runs/656856709

dagar, 2020-05-10:
https://github.com/PX4/Firmware/pull/14883/checks?check_run_id=661319771

dagar, 2020-05-10:
https://github.com/PX4/Firmware/pull/14884/checks?check_run_id=661426027

dagar, 2020-05-10:
https://github.com/PX4/Firmware/pull/14885/checks?check_run_id=661428233

julianoes, 2020-05-14:
FYI @kamilritz here are some CI failures collected.

julianoes, 2020-05-19:
I can now reproduce the "connection" problem locally in docker. It has something to do with px4 not terminating properly, or not cleaning up the lock file properly. The problem persists until I remove it by `rm /tmp/px4_lock-0`.

kamilritz, 2020-05-19:
That is awesome. I also experience this sometimes locally. And it seems like the px4 process is still running after the termination of the test.

julianoes, 2020-05-20:
If `px4` is still running then it would ignore `SIGKILL` which is quite unlikely :astonished:

julianoes, 2020-05-20:
@kamilritz it might be the case that the current CI failures are actual failures. I also see them locally, and in the log the ground truth seem to be quite a bit off.

https://logs.px4.io/plot_app?log=6e903edc-d396-4a6f-9971-2b20a93ee7ce

kamilritz, 2020-05-20:
@julianoes But this is not always the case. It is just not repeatable, right?

julianoes, 2020-05-25:
@kamilritz not always but sometimes. It happens to me when I run it locally for 10+ times. The test needs to be repeatable, otherwise we'll just have to disable it again :disappointed: .

kamilritz, 2020-05-25:
@julianoes Is it always the same test case? Which one/s?

julianoes, 2020-05-25:
I believe the vision or vision velocity one. Let me run the test again and see.

kamilritz, 2020-05-25:
Ok, I can have a look this evening. It could be that by speeding up the tests, the robustness decreased.

julianoes, 2020-05-25:
@kamilritz I just ran 20 tries at 20x and it never failed :thinking:

julianoes, 2020-05-26:
Here is an example where the estimation goes wild: 

> However, this still fails sometimes locally, here is an example where the estimation is bonkers:
> https://logs.px4.io/plot_app?log=af8d98b8-3042-4e1d-bf43-ca97a5cf9bd0

julianoes, 2020-05-26:
Here is another strange failure:
https://logs.px4.io/plot_app?log=1ffedfd0-812c-4cb0-8b26-a2e6b2e698fd
```
--> Test case 4 of 4: 'Fly square Multicopter Missions' running ...
creating log directory: logs/2020-05-26T15-01-02Z/02/iris/Fly_square_Multicopter_Missions
Running: /home/julianoes/src/Firmware/build/px4_sitl_default/bin/px4 /home/julianoes/src/Firmware/ROMFS/px4fmu_common -s etc/init.d-posix/rcS -t /home/julianoes/src/Firmware/test_data -d
Logging to logs/2020-05-26T15-01-02Z/02/iris/Fly_square_Multicopter_Missions/log-px4.log
Running: gzserver --verbose /home/julianoes/src/Firmware/Tools/sitl_gazebo/worlds/empty.world
Logging to logs/2020-05-26T15-01-02Z/02/iris/Fly_square_Multicopter_Missions/log-gzserver.log
Running: gz model --spawn-file /home/julianoes/src/Firmware/Tools/sitl_gazebo/models/iris/iris.sdf --model-name iris -x 1.01 -y 0.98 -z 0.83
Logging to logs/2020-05-26T15-01-02Z/02/iris/Fly_square_Multicopter_Missions/log-gzmodelspawn.log
Running: /home/julianoes/src/Firmware/build/px4_sitl_default/mavsdk_tests/mavsdk_tests --url udp://:14540 Fly square Multicopter Missions
Logging to logs/2020-05-26T15-01-02Z/02/iris/Fly_square_Multicopter_Missions/log-mavsdk_tests.log
[px4         ] INFO  [px4] Creating symlink /home/julianoes/src/Firmware/ROMFS/px4fmu_common -> /home/julianoes/src/Firmware/build/px4_sitl_default/tmp/rootfs/etc
[px4         ] ______  __   __    ___ 
[px4         ] | ___ \ \ \ / /   /   |
[px4         ] | |_/ /  \ V /   / /| |
[px4         ] |  __/   /   \  / /_| |
[px4         ] | |     / /^\ \ \___  |
[px4         ] \_|     \/   \/     |_/
[px4         ] px4 starting.
[px4         ] INFO  [px4] Calling startup script: /bin/sh etc/init.d-posix/rcS 0
[px4         ] INFO  [param] selected parameter default file eeprom/parameters_10016
[px4         ] [param] Loaded: eeprom/parameters_10016
[px4         ] COM_DL_LOSS_T set to 100
[px4         ] COM_RC_LOSS_T set to 5.0
[px4         ] COM_OF_LOSS_T set to 5.0
[px4         ] INFO  [dataman] Unknown restart, data manager file './dataman' size is 11798680 bytes
[px4         ] INFO  [simulator] Waiting for simulator to accept connection on TCP port 4560
[px4         ] INFO  [simulator] Simulator connected on TCP port 4560.
[px4         ] INFO  [commander] LED: open /dev/led0 failed (22)
[px4         ] INFO  [commander] Mission #2 loaded, 3 WPs, curr: 0
[px4         ] INFO  [init] Mixer: etc/mixers/quad_w.main.mix on /dev/pwm_output0
[px4         ] INFO  [mavlink] mode: Normal, data rate: 4000000 B/s on udp port 18570 remote port 14550
[px4         ] INFO  [mavlink] mode: Onboard, data rate: 4000000 B/s on udp port 14580 remote port 14540
[px4         ] INFO  [mavlink] mode: Onboard, data rate: 4000 B/s on udp port 14280 remote port 14030
[px4         ] INFO  [logger] logger started (mode=all)
[px4         ] INFO  [logger] Start file log (type: full)
[px4         ] INFO  [logger] [logger] ./log/2020-05-26/13_06_11.ulg
[px4         ] INFO  [logger] Opened full log file: ./log/2020-05-26/13_06_11.ulg
[px4         ] INFO  [mavlink] MAVLink only on localhost (set param MAV_BROADCAST = 1 to enable network)
[px4         ] INFO  [px4] Startup script returned successfully
[px4         ] INFO  [ecl/EKF] 808000: GPS checks passed
[px4         ] INFO  [ecl/EKF] 1844000: EKF aligned, (baro hgt, IMU buf: 18, OBS buf: 14)
[px4         ] INFO  [ecl/EKF] 1844000: reset position to last known position
[px4         ] INFO  [ecl/EKF] 1844000: reset velocity to zero
[px4         ] INFO  [ecl/EKF] 5396000: reset position to GPS
[px4         ] INFO  [ecl/EKF] 5396000: reset velocity to GPS
[px4         ] INFO  [ecl/EKF] 5396000: starting GPS fusion
[px4         ] INFO  [mavlink] partner IP: 127.0.0.1
[gzserver    ] Gazebo multi-robot simulator, version 9.13.0
[gzserver    ] Copyright (C) 2012 Open Source Robotics Foundation.
[gzserver    ] Released under the Apache 2 License.
[gzserver    ] http://gazebosim.org
[mavsdk_tests] Filters: Fly square Multicopter Missions
[mavsdk_tests] [03:06:12|Info ] MAVSDK version: 0.24.0 (mavsdk_impl.cpp:25)
[mavsdk_tests] Waiting for system connect
[mavsdk_tests] [03:06:12|Info ] New system on: 127.0.0.1:14580 (udp_connection.cpp:264)
[mavsdk_tests] [03:06:12|Debug] New: System ID: 1 Comp ID: 1 (mavsdk_impl.cpp:401)
[mavsdk_tests] [03:06:12|Debug] Component Autopilot (1) added. (system_impl.cpp:386)
[mavsdk_tests] [03:06:12|Debug] Discovered 1 component(s) (UUID: 5283920058631409231) (system_impl.cpp:556)
[mavsdk_tests] Waiting for system to be ready
[px4         ] WARN  [mavlink] [timesync] RTT too high for timesync: 4252 ms
[px4         ] WARN  [mavlink] [timesync] RTT too high for timesync: 3652 ms
[px4         ] WARN  [mavlink] [timesync] RTT too high for timesync: 3052 ms
[px4         ] WARN  [mavlink] [timesync] RTT too high for timesync: 2452 ms
[px4         ] WARN  [mavlink] [timesync] RTT too high for timesync: 1852 ms
[px4         ] WARN  [mavlink] [timesync] RTT too high for timesync: 1252 ms
[px4         ] WARN  [mavlink] [timesync] RTT too high for timesync: 652 ms
[px4         ] WARN  [mavlink] [timesync] RTT too high for timesync: 52 ms
[... 148 lines cut ...]
[mavsdk_tests] [03:07:46|Debug] Discovered 1 component(s) (UUID: 5283920058631409231) (system_impl.cpp:556)
[px4         ] WARN  [mavlink] [timesync] RTT too high for timesync: 3472 ms
[px4         ] WARN  [mavlink] [timesync] RTT too high for timesync: 4132 ms
[mavsdk_tests] Waiting for system to be ready
[... cut ...]
[gzserver    ] [Msg] Connected to gazebo master @ http://127.0.0.1:11345
[gzserver    ] [Msg] Publicized address: 10.0.2.15
[gzserver    ] [Msg] Connecting to PX4 SITL using TCP
[gzserver    ] [Msg] Lockstep is enabled
[gzserver    ] [Msg] Speed factor set to: 10
[gzserver    ] [Msg] Using MAVLink protocol v2.0
[gzserver    ] [Err] [gazebo_mavlink_interface.cpp:1389] recvfrom error: Connection reset by peer
[gzserver    ] [Err] [gazebo_mavlink_interface.cpp:1396] Connection closed by client.
```

julianoes, 2020-05-29:
Keeping this open to track more failures.

julianoes, 2020-06-04:
Closing for now until we see it again.

junwoo091400, 2022-10-19:
The issue at https://github.com/PX4/PX4-Autopilot/pull/20227#issuecomment-1284270089 seems to be similar.

And this is also related to the comment from another PR: https://github.com/PX4/PX4-Autopilot/pull/20213#issuecomment-1247757615

It's always the `standard_vtol` :thinking:

junwoo091400, 2022-10-19:
First of all, issues in the CI itself:

1. Core dump is not being uploaded. 
[image]

- https://web.eecs.umich.edu/~sugih/pointers/gdb_core.html
@dagar how could we get this to work? Is core file always generated or only on core dump events?

2. Logs are not uploaded, this would be helpful, as https://github.com/PX4/PX4-Autopilot/issues/14997 utilized it.

- Will add Log uploading capability as an artifact from CI

junwoo091400, 2022-10-26:
## Address Sanitizer strikes again!
Ah I am noticing a pattern here. Exactly as why #20213 PR was created, it's only occurring in the AddressSanitizer build, which happens to be `standard_vtol` at latitutde`-38`.

Some tests that failed:
- https://github.com/PX4/PX4-Autopilot/actions/runs/3317387230
- https://github.com/PX4/PX4-Autopilot/actions/runs/3328711785

[image]

## Current CI failure
And it is weird that it isn't giving any core dump output :thinking: 

[image]

### How the UUID bug failed in previous PR
[image]

Notice how the core dump is provided. Thus the CI failure pattern we are observing now is something different.

junwoo091400, 2022-10-26:
Command to do the build:
```bash
make px4_sitl gazebo mavsdk_tests DONT_RUN=1 PX4_ASAN=1

test/mavsdk_tests/mavsdk_test_runner.py test/mavsdk_tests/configs/sitl.json --model standard_vtol --speed-factor 20 --abort-early --upload
```

junwoo091400, 2022-10-26:
Found the PR that is actually failing, will test again with this: https://github.com/PX4/PX4-Autopilot/pull/20470

https://github.com/PX4/PX4-Autopilot/commit/56436df1e37ab2159752a52236461382c7366360

junwoo091400, 2022-10-26:
Running the docker container for reproducing (hopefully) the difference of environment with the github action:

```bash
docker run -it --privileged --ulimit core=-1 --security-opt seccomp=unconfined -v ~/PX4-Autopilot/:~ px4io/px4-dev-simulation-focal:2021-09-08
```

junwoo091400, 2022-10-26:
Some more clues revealed from https://github.com/Auterion/PX4_firmware_private/pull/2096:

There is now some output

[image]

junwoo091400, 2022-10-27:
Observed the following in the local Docker container based MAVSDK SITL Test:

Command ran:
```bash
root@87e6053cdb91:/home/root/PX4# test/mavsdk_tests/mavsdk_test_runner.py test/mavsdk_tests/configs/sitl.json --model standard_vtol --speed-factor 20 --abort-early --upload --debugger --verbose gdb
```

### Findings
1. The system isn't connecting for more than 25 seconds [here](https://github.com/PX4/PX4-Autopilot/blob/473b471fb6f5b57f6e81ad802594578ea72d0ad3/test/mavsdk_tests/autopilot_tester.cpp#L66).
2. When `--debugger gdb` option is omitted, the MAVSDK test script dies and `gzserver` doesn't terminate like this: 
```bash
Terminating px4
Killing px4
px4 exited with None
```

**This indicates that the PX4 exit process is not cleanly implemented in the MAVSDK script.**

### When ran with GDB Debugger
[image]

```bash
Terminating px4
px4 exited with 0
Terminating gzserver
gzserver exited with -15
gzmodelspawn exited with 0
mavsdk_tests exited with 1
```

junwoo091400, 2022-10-27:
Identified the root cause of the MAVSDK test freezing to this part:

https://github.com/PX4/PX4-Autopilot/blob/5edbc2f80a0b24082711e0d5095cebd3e3860253/test/mavsdk_tests/process_helper.py#L141

Since this function doesn't return / exit (while trying to stop running PX4 instance), other runners can't exit, and thus the whole test script comes to a halt here:
https://github.com/PX4/PX4-Autopilot/blob/5edbc2f80a0b24082711e0d5095cebd3e3860253/test/mavsdk_tests/mavsdk_test_runner.py#L484

junwoo091400, 2022-10-27:
## PX4 Runner not stopping
PX4 Runner thread is not stopping (via `join()`) even after Terminating & Killing is called within `stop()` function.

[image]

**What's weird is that when the program is terminated with the SIGINT signal (`Ctrl + C` on Terminal), px4 runner does indeed exit and the thread terminates (notice the following:)**

```
<process_helper.Px4Runner object at 0x7f72e1213400>  stopping!
px4 exited with -9
stop thread set
thread joined. State: False
```

[image]

junwoo091400, 2022-10-27:
## Subprocess exit signal
According to [subprocess python library documentation](https://docs.python.org/3.4/library/subprocess.html#subprocess.Popen.returncode):
```
A negative value -N indicates that the child was terminated by signal N (POSIX only).
```

And since 9 is [SIGKILL](https://dsa.cs.tsinghua.edu.cn/oj/static/unix_signal.html), which corresponds with the `px4 exited with -9` printout when we trigger the keyboard interrupt,

it seems that px4 is only actually getting killed on a keyboard interrupt event.

junwoo091400, 2022-10-27:
## Different behavior with Valgrind
**Noticed something interesting. Now the `poll()` return value from the PX4 runner is 1, instead of `None` (case for both native run & GDB run).**

[image]

- Valgrind command: `Running: valgrind --track-origins=yes --leak-check=full -v /home/root/PX4/build/px4_sitl_default/bin/px4 /home/root/PX4/build/px4_sitl_default/etc -s etc/init.d-posix/rcS -t /home/root/PX4/test_data -d`
- GDB command: `Running: gdb --args /home/root/PX4/build/px4_sitl_default/bin/px4 /home/root/PX4/build/px4_sitl_default/etc -s etc/init.d-posix/rcS -t /home/root/PX4/test_data -d`
- Raw command: `Running: /home/root/PX4/build/px4_sitl_default/bin/px4 /home/root/PX4/build/px4_sitl_default/etc -s etc/init.d-posix/rcS -t /home/root/PX4/test_data -d`

junwoo091400, 2022-10-27:
https://docs.python.org/3.4/library/subprocess.html#subprocess.Popen

Interesting related resource: https://stackoverflow.com/questions/52224529/how-to-get-asan-to-ignore-child-process

make px4_sitl gazebo mavsdk_tests DONT_RUN=1 PX4_TSAN=1

## Updating used Docker container
As the problem is still so ambiguous, and can't find the root cause, this will be tried with a different docker caontainer (from recently): https://hub.docker.com/layers/px4io/px4-dev-simulation-focal/2022-08-12/images/sha256-62b5277405c1028e4957775fc3c4a33484aae5cddcbb8c269994a62d6dcf8393?context=explore

junwoo091400, 2022-11-15:
> [gzserver    ] [Err] [gazebo_mavlink_interface.cpp:1389] recvfrom error: Connection reset by peer
> [gzserver    ] [Err] [gazebo_mavlink_interface.cpp:1396] Connection closed by client.

[image]

This is still happening. The `recvfrom error & connection closed by client`. No idea still why PX4 is dying 😢. Also, this one was without address sanitizer option, so seems like this can happen on any build

julianoes, 2022-11-18:
@junwoo091400 if you have steps to reproduce the problem locally, I can try that.

junwoo091400, 2022-11-18:
> @junwoo091400 if you have steps to reproduce the problem locally, I can try that.

It would be:
```
# As in SITL definition YaML file, run the container
docker run -it --privileged --ulimit core=-1 --security-opt seccomp=unconfined -v ~/PX4-Autopilot/:~ px4io/px4-dev-simulation-focal:2021-09-08

# Inside container, install MAVSDK
MAVSDK_VERSION=1.3.1 wget "https://github.com/mavlink/MAVSDK/releases/download/v$(cat test/mavsdk_tests/MAVSDK_VERSION)/libmavsdk-dev_$(cat test/mavsdk_tests/MAVSDK_VERSION)_ubuntu20.04_amd64.deb" && dpkg -i "libmavsdk-dev_$(cat test/mavsdk_tests/MAVSDK_VERSION)_ubuntu20.04_amd64.deb"

# Build PX4 Mavsdk tests
make px4_sitl gazebo mavsdk_tests DONT_RUN=1 PX4_ASAN=1

# Run the SITL
test/mavsdk_tests/mavsdk_test_runner.py test/mavsdk_tests/configs/sitl.json --model standard_vtol --speed-factor 20 --abort-early --upload --debugger --verbose
```

You will notice that the failure mode is px4 just hanging & mavsdk runner also hanging (as px4 subprocess didn't die yet) & gzserver reporting no connection to PX4.

Let me know if any part is unclear! I really want to get to the bottom of this 👀👀

julianoes, 2022-11-20:
Hm, so you have been able to reproduce it like that? What about without docker?

junwoo091400, 2022-11-20:
> Hm, so you have been able to reproduce it like that? What about without docker?

Yup, and without docker it is not reproducible. So it has to be with the docker container I think

julianoes, 2022-11-20:
I hate debugging in docker. Ok thanks for the info!

julianoes, 2022-11-20:
It doesn't seem to stall for me. I'm letting it run again and again.

junwoo091400, 2022-11-22:
Maybe related: https://github.com/PX4/PX4-Autopilot/issues/11257

https://github.com/PX4/PX4-Autopilot/issues/11380

Still open: https://github.com/PX4/PX4-Autopilot/issues/11280

junwoo091400, 2022-11-22:
> It doesn't seem to stall for me. I'm letting it run again and again.

Oh ok, I will try again, last time I tried it (unless I messed up the reproducing code up there 😆), I could trigger it consistently.

PR #14944 (2020-05-29): Attempt to fix MAVSDK tests CI
Work in progress to fix #14835