PX4/PX4-Autopilot #14838: Misfire at catapult launch (custom FW version 1.7.2)

MartinStokelj, 2020-05-05:
**Describe the bug**
From 55 test flights we had 2x misfire at catapult launch. It looks like PWM was not send on ESC and motor didn't start.

**To Reproduce**
We do not know how to reproduce.

**Expected behavior**
Motor should always start at catapult launch

**Log Files and Screenshots**
ULG files
[misfire2x.zip](https://github.com/PX4/Firmware/files/4583095/misfire2x.zip)

**Drone (please complete the following information):**
https://www.unmannedtechshop.co.uk/product/unmanned-pixhawk-autopilot-kit/
pixhawk 2 fmuv2
Stack: PX4 Pro
Vehicle: VTOL
FW Version: 1.7.2 custom build

MartinStokelj, 2020-05-18:
Any suggestion what to check in ulg files?

julianoes, 2020-05-18:
I'm not sure either. I can see the acceleration spike which probably should be detected, and at the bottom of the file there are some debug messages as well.

It would be interesting to also see a logfile of a successful launch to compare what is different.

MartinStokelj, 2020-05-18:
> 
> 
> I'm not sure either. I can see the acceleration spike which probably should be detected, and at the bottom of the file there are some debug messages as well.
> 
> It would be interesting to also see a logfile of a successful launch to compare what is different.

Here is one
https://review.px4.io/plot_app?log=fb8b6e04-6f15-4f3d-af5d-467ec1546217
I can give you more if you want

julianoes, 2020-05-18:
I can't see anything obvious, sorry.

@Antiheavy maybe you have some comments or insights on this.

MartinStokelj, 2020-05-18:
> 
> 
> I can't see anything obvious, sorry.
> 
> @Antiheavy maybe you have some comments or insights on this.

Can you suggest what to check if this occurs again?
I thought that is maybe connected to bad GPS connection even if the "COM_ARM_WO_GPS	Allow arming without GPS" is on.
Is it possible that catapult launch shocks GPS to much?

MartinStokelj, 2020-05-19:
@dagar did you had the time to check this?

MartinStokelj, 2020-05-22:
Today we had another misfire at catapult launch

https://review.px4.io/plot_app?log=5276768d-8b93-4237-a8e6-b38f9ed6ee0d
or
[log001.zip](https://github.com/PX4/Firmware/files/4667338/log001.zip)

MartinStokelj, 2020-05-22:
We think that maybe is the param LAUN_CAT_A (Catapult accelerometer threshold) the problem...
We have it on 20, is this too much? I don't know which value do I need to check in ulog files to confirm this (in theory - threshold was not in range)

MartinStokelj, 2020-06-02:
During the last missfire we putted drone on debug link, and called "Reboot" after reboot there was still problem that there was no PWM. So we tryed "px4io stop" and then "px4io start" after that all worked ok.
Since this is old FW anyone know what was the fix if px4io stops working correctly?

dagar, 2020-06-02:
Sorry, there isn't much to go on because the internal px4io state isn't logged. If we could get you synced up with master we could start adding additional px4io instrumentation (logging).

closed 2020-06-11 by MartinStokelj