PX4/PX4-Autopilot #14840: EKF2 yaw estimate diverges on VIO stop/restart - v1.11-beta

mhkabir, 2020-05-06:
**Describe the bug**
The EKF yaw estimate diverges if external vision data is stopped and restarted. This is easily reproducible in hand/bench tests.

**To Reproduce**
Steps to reproduce the behavior:
1. Start sending EV yaw data.
2. Check QGC, EKF yaw estimate looks fine and matches EV.
3. Stop sending EV data.
4. Check QGC, EKF yaw estimate is spinning out of control.
5. Start sending EV data again
6. Check QGC, EKF yaw resets to correct EV value, but then continues spinning out of control.

**Expected behavior**
1. The yaw estimate falls back to compass after EV stop (?)
2. The yaw estimate follows the external vision data after EV restart. 

**Log Files and Screenshots**

[image]

Log from boot: https://logs.px4.io/plot_app?log=c0a43438-fb28-45fd-9a5b-92329748fb93

mhkabir, 2020-05-06:
@kamilritz @bresch Any clues here? We did not have this on v1.10, so seems like something to do with recent changes in ECL.

kamilritz, 2020-05-06:
The reason why it was not able to recover after fusing ev_yaw again, is that it learned a different z gyrobias in the meantime, don't know what was constraining the heading in this period.
I am not sure if there was a designed fall back to another heading source. 
If you need a quick hack you could probably get away with a resetMagHeading [here](https://github.com/PX4/ecl/blob/master/EKF/control.cpp#L400).
@bresch I assume you would want to handle the case it in the mag_control.cpp file.

kamilritz, 2020-05-06:
@mhkabir This issue would probably be better placed in ECL.

mhkabir, 2020-05-06:
@kamilritz thanks. I see the issue now. I'll try to fix it properly in the net 1-2 days, and also do https://github.com/PX4/ecl/issues/800 while I'm at it.

closed 2024-03-08 by mhkabir