PX4/PX4-Autopilot #14847: CI SITL MC_offboard_rpyrt test intermittant failure

dagar, 2020-05-06:
[image]

http://ci.px4.io:8080/blue/organizations/jenkins/PX4_misc%2FFirmware-SITL_tests_ASan/detail/master/1747/pipeline/57
https://logs.px4.io/plot_app?log=423ec840-cc41-45de-8efb-71053cb26936

dagar, 2020-05-06:
http://ci.px4.io:8080/blue/organizations/jenkins/PX4_misc%2FFirmware-SITL_tests/detail/PR-14825/20/pipeline

Pedro-Roque, 2020-05-06:
@dagar  I digged a bit more and it seems that the thread is not starting properly, as when I wait for messages on the `mavros/setpoint_raw/attitude` before arming it locks the process. So there seems to be the root of the problem.

I'm debugging now to see how we can go past this.

dagar, 2020-05-06:
> I digged a bit more and it seems that the thread is not starting properly

Which thread?

Pedro-Roque, 2020-05-06:
> > I digged a bit more and it seems that the thread is not starting properly
> 
> Which thread?

https://github.com/PX4/Firmware/blob/master/integrationtests/python_src/px4_it/mavros/mavros_offboard_yawrate_test.py#L72

twprinston, 2021-08-19:
I have met the same problem. Is there any progress on this matter? Thanks.