PX4/PX4-Autopilot #14851: Poor Fixed wing runway takeoff tracking

ryanjAA, 2020-05-06:
Fixed wing runway takeoff tracking is very poor on various firmwares and aircraft types that i've been testing and had others test.

I've compiled some notes and logs on this and it was even brought up (in a different context) in https://github.com/PX4/Firmware/issues/11785 but the more we look at it, the more this is a true issue and from a practical standpoint is making using autonomous takeoff not feasible given the high risk.

Let me also preface this by saying a many things have been tested and changed and a makeshift solution was come up with as well but that doesn't solve the problem. Wheel Rate settings have been altered to various extreme and subdued amounts (as seen in the logs). This has reduced some things present like sudden full wheel deflection upon mission start but it never fixes the trajectory of the takeoff and that it does not generally point or aim to the takeoff waypoint but goes off course many time more than it does not.

Multiple different aircraft and aircraft types were used, even though the mixer is the same, that is only because we have varying sizes and weights of test aircraft and even foam ones where the setup, weight and geometry on the nose gear and linkages is completely different (which is good to rule out a single hardware type or build as the culprit).

I often have seen the wheel channel output (main 5) swing up and down 500us over neutral, essentially max on both extremes and even when that isnt having the same type of behavior exists at +/200. The scaling factor probably helps in these cases by reducing movement (so long as you are still on course enough or at least not going to crash on takeoff) but the thing to remember is this entire process is only 3-8 seconds long before you're airborne so it's easy to not see or overlook.

The hacked solution? Limit min/max pwm value of the wheel servo either in PX4 or at the servo level to limit how much steering can physically occur. This is not a solution but gets things airborne by quasi limiting the ability to go off course. Not a true solution.

Logs:

https://review.px4.io/plot_app?log=efad594a-afe0-451a-a1dd-a6130e49a03b
This log has an sudden full right wheel right as the mission is initiated.

[image]

https://review.px4.io/plot_app?log=25ce0df8-15d6-4591-8578-83042921cffe - This is similar but not as severe as the previous log (still has huge wheel swings though).

https://review.px4.io/plot_app?log=53e17faa-3f52-4a3f-8e7e-b2d8c8ede219 - This log is interesting that the aircraft was purposely moved to the corner of the runway because the previous flights had it going to the right during the takeoff portion and it sill went much too hard right and then off the runway.

[image]

https://review.px4.io/plot_app?log=ee6bef2c-167c-4d43-ae96-8ac2144df2e1 - This takes off successfully because of the large area, thats the only reason. There is video of the takeoff if need be.

https://review.px4.io/plot_app?log=bc807a89-07c1-44ac-836b-7d9e09ee7d0a - This log actually doenst look as bad except if you zoom out and look at the first waypoint, it's still off to the left.

https://review.px4.io/plot_app?log=668ab1af-c9f8-4c30-b6fb-f26b998c85b4 - This log takes off successfully but zoom into the wheel output. It's successful because the area used was large and it has the time and room to recover and/or just get airborne.

https://review.px4.io/plot_app?log=bac42025-4d70-45d3-ae4c-39c781328b01 This was a taxi test but still shows rapid oscillation (so throttle is only at 30%).
 
[image]

https://review.px4.io/plot_app?log=cbb044ac-211d-49f9-86b9-3d0702427cbe - Same place, post taxi test (same wheel rate settings) and less rapid oscillation but still really bad, had to abort since about to go off runway.

[image]

https://review.px4.io/plot_app?log=08ff0f58-b264-49b0-a694-58eab9b76ffd

https://review.px4.io/plot_app?log=bd117170-1dca-42a8-a309-58b537296419

Below are a collection more, some are very extreme, some less. Multiple combinations of the wheel rate parameters were altered but this still represents multiple settings where oscillation occurred severely.

https://review.px4.io/plot_app?log=7758631a-a5ae-4ead-97b4-f496cfa2a3c8
https://review.px4.io/plot_app?log=10e31631-552f-4585-b1f4-bc73da6ac094
https://review.px4.io/plot_app?log=75486b4a-385b-47a6-a851-c54ae909c077
https://review.px4.io/plot_app?log=d6d43381-14ac-4ba7-9cf2-f9b68d0ed7b7
https://review.px4.io/plot_app?log=b0be9c38-62e5-4cee-9f2e-9c40b603c143
https://review.px4.io/plot_app?log=c1dcab38-8c6b-42a8-91a2-4e89ab1ec26a
https://review.px4.io/plot_app?log=6130efca-f4dd-4a81-b648-f1ed5424f585
 https://review.px4.io/plot_app?log=e7e3027f-8cf0-4ef8-ace5-640383a2775b
https://review.px4.io/plot_app?log=6130efca-f4dd-4a81-b648-f1ed5424f585
https://review.px4.io/plot_app?log=7c70673a-5d26-4996-9de1-35e1b1633d3c

@dagar - as discussed

dagar, 2020-05-06:
Thanks for putting this detailed issue together. Do you have example "good" logs? 

If you happen to have it on hand a video that corresponds to a log never hurts.

ryanjAA, 2020-05-07:
I’ll grab some and track down a video.

ryanjAA, 2020-05-07:
So i went through a lot of logs. There are a few things to look at.

Oscillation (rapid movement of the steering servo) and
Ground based tracking

I think oscillation could be combatted from lowering some of the wheel rate values, which is bad if the ground based tracking is poor. I have done this, even to very severe levels and what happens is if you are tracking poorly you don't have the ability to recover at the wheel rate level.

Back to looking at logs (and we have a lot we can pick through), it's actually easy to see the oscillation in almost all of them but tbh the difference comes back to the tracking side of it. When looking at them and ones with video, you can see that often even though there is oscillation it doesn't have the ability to cause problems because the plane simply takes off but this assumes you are tracking correctly or at least enough to be able to stay on the runway (both cases are present). It is when you are not tracking and there is the oscillation that things can go really bad.

https://review.px4.io/plot_app?log=2e9e96e6-9567-4340-b7e3-49c8fafea980 - This tracks correctly and shows the oscillation settle quite quick. Video link is https://youtu.be/AdpGysyyPkA

https://review.px4.io/plot_app?log=9903b609-587b-44bd-a5e1-7d88fa70ffc8 - This is interesting. The ground plot shows it didn't track perfectly (slight right) and the video suggests that this is technically true with a little oscillation but this is good to show that given the potential takeoff distance, being off course just a little over a long distance can have a potentially bad outcome even though this is not bad. Video is https://youtu.be/4Q7A2tMCXo0

https://review.px4.io/plot_app?log=646cc3a3-9404-49cc-a5d1-26b15e597546 - This doesn't track (although takes off) and you can see the oscillation is more pronounced. Also interesting though is that the moment it gets in the air it rolls right then back left to get on track for the actual takeoff waypoint. Video link is https://youtu.be/baowgtjfdlA

https://review.px4.io/plot_app?log=5263363b-b08e-4172-af5b-a9e64bd5601b
this shows that the plane stays quite close to the center line although the log shows it does go off to the right but that is when airborne. Video link is https://youtu.be/AgAGCJGjLKI

ryanjAA, 2020-05-12:
@dagar Let me know if there anything you want me to have tested. We have excess people capacity given shelter at home requirements with the ability to test.

ryanjAA, 2020-05-19:
@dagar with some looking into this it's starting to be more clear. We should almost treat this as two issues, one is ground steering (which maybe some documentation on reducing oscillation and what the different wheel rate parameters mean in real terms and their effect as the current language is not too informative in a practical sense) but the second is the actual tracking.

If we look at a log above, specifically this: https://review.px4.io/plot_app?log=5263363b-b08e-4172-af5b-a9e64bd5601b

and then look at yaw vs yaw set point we can see that they track quite close. This is in fact true with all the logs posted. Where the difference lies is in the EKF2 heading innovation. In the aforementioned log it is less than -7º during takeoff and then once airborne settles to -6º (approx). 

[image]

[image]
If we look at this on one of the worse logs, https://review.px4.io/plot_app?log=cbb044ac-211d-49f9-86b9-3d0702427cbe We see that while the yaw oscillates during takeoff (something we should remedy via the first point), the EKF2 heading innovation is huge at 35-40º. This is generally only when the vehicle starts moving. Again, these are different aircraft and aircraft types so one thing that is a common is if when the EKF2 heading innovation is large on a mission start, ground tracking decreases substantially. 

So the question is first why (about the EKF2 Heading Innovation) and secondly, how do be remedy this or make it more robust? It does only go up once the vehicle starts moving and while some of the test ratios do increase, nothing is rejected or seen as invalid (>1).

This scenario is the case in pretty much in all logs I've looked at and it's very possible that both the tracking and oscillation issues could exacerbate each other. The oscillation making the plane off course, etc then the estimate gets worse but even with lower oscillating logs anytime the heading innovation is high and therefore tracking is poor so they are connected but not cause and effect.

[image]

ryanjAA, 2020-12-25:
Seems to be better in 1.11.x. Will reopen if any regressions.

closed 2020-12-25 by ryanjAA