PX4/PX4-Autopilot #14854: "make px4_sitl jmavsim" fail no macOS Catalina 10.15.4

kainism, 2020-05-07:
kevin@kevinJdeMacBook-Pro ~ % cd /Users/kevin/Firmware
kevin@kevinJdeMacBook-Pro Firmware % make px4_sitl jmavsim
[0/1] cd /Users/kevin/Firmware/build/p…/kevin/Firmware/build/px4_sitl_default
SITL ARGS
sitl_bin: /Users/kevin/Firmware/build/px4_sitl_default/bin/px4
debugger: none
program: jmavsim
model: none
world: none
src_path: /Users/kevin/Firmware
build_path: /Users/kevin/Firmware/build/px4_sitl_default
empty model, setting iris as default
SITL COMMAND: “/Users/kevin/Firmware/build/px4_sitl_default/bin/px4” “/Users/kevin/Firmware”/ROMFS/px4fmu_common -s etc/init.d-posix/rcS -t “/Users/kevin/Firmware”/test_data
INFO [px4] Creating symlink /Users/kevin/Firmware/ROMFS/px4fmu_common -> /Users/kevin/Firmware/build/px4_sitl_default/tmp/rootfs/etc

| ___ \ \ \ / / / |
| |/ / \ V / / /| |
| __/ / \ / /| |
| | / /^\ \ ___ |
_| / / |_/

px4 starting.

INFO [px4] Calling startup script: /bin/sh etc/init.d-posix/rcS 0
INFO [param] selected parameter default file eeprom/parameters_10016
[param] parameter file not found, creating eeprom/parameters_10016
SYS_AUTOCONFIG: curr: 0 -> new: 1

SYS_AUTOSTART: curr: 0 -> new: 10016
BAT_N_CELLS: curr: 0 -> new: 3
CAL_ACC0_ID: curr: 0 -> new: 1311244
CAL_ACC_PRIME: curr: 0 -> new: 1311244
CAL_GYRO0_ID: curr: 0 -> new: 2294028
CAL_GYRO_PRIME: curr: 0 -> new: 2294028
CAL_MAG0_ID: curr: 0 -> new: 197388
CAL_MAG_PRIME: curr: 0 -> new: 197388
COM_DISARM_LAND: curr: 2.0000 -> new: 0.5000
COM_OBL_ACT: curr: 0 -> new: 2
COM_RC_IN_MODE: curr: 0 -> new: 1
EKF2_ANGERR_INIT: curr: 0.1000 -> new: 0.0100
EKF2_GBIAS_INIT: curr: 0.1000 -> new: 0.0100
COM_ARM_EKF_AB: curr: 0.0017 -> new: 0.0050
EKF2_REQ_GPS_H: curr: 10.0000 -> new: 0.5000
MC_PITCH_P: curr: 6.5000 -> new: 6.0000
MC_PITCHRATE_P: curr: 0.1500 -> new: 0.2000
MC_ROLL_P: curr: 6.5000 -> new: 6.0000
MC_ROLLRATE_P: curr: 0.1500 -> new: 0.2000
MPC_HOLD_MAX_Z: curr: 0.6000 -> new: 2.0000
MPC_Z_VEL_I: curr: 0.1000 -> new: 0.1500
MPC_Z_VEL_P: curr: 0.2000 -> new: 0.6000
MPC_XY_P: curr: 0.9500 -> new: 0.8000
MPC_XY_VEL_P: curr: 0.0900 -> new: 0.2000
MPC_XY_VEL_D: curr: 0.0100 -> new: 0.0160
MPC_SPOOLUP_TIME: curr: 1.0000 -> new: 0.5000
MPC_TKO_RAMP_T: curr: 3.0000 -> new: 1.0000
NAV_ACC_RAD: curr: 10.0000 -> new: 2.0000
NAV_DLL_ACT: curr: 0 -> new: 2
RTL_DESCEND_ALT: curr: 30.0000 -> new: 5.0000
RTL_LAND_DELAY: curr: -1.0000 -> new: 5.0000
RTL_RETURN_ALT: curr: 60.0000 -> new: 30.0000
SDLOG_MODE: curr: 0 -> new: 1
SDLOG_PROFILE: curr: 3 -> new: 131
SDLOG_DIRS_MAX: curr: 0 -> new: 7
SENS_BOARD_X_OFF: curr: 0.0000 -> new: 0.0000
SENS_DPRES_OFF: curr: 0.0000 -> new: 0.0010
TRIG_INTERFACE: curr: 4 -> new: 3
COM_CPU_MAX: curr: 90.0000 -> new: -1.0000
RTL_DESCEND_ALT: curr: 5.0000 -> new: 10.0000
RTL_LAND_DELAY: curr: 5.0000 -> new: 0.0000
PWM_MAX: curr: 2000 -> new: 1950
PWM_MIN: curr: 1000 -> new: 1075
GPS_UBX_DYNMODEL: curr: 7 -> new: 6
Buildfile: /Users/kevin/Firmware/Tools/jMAVSim/build.xml
SYS_AUTOCONFIG: curr: 1 -> new: 0
INFO [dataman] Unknown restart, data manager file ‘./dataman’ size is 11798680 bytes
INFO [simulator] Waiting for simulator to accept connection on TCP port 4560
make_dirs:

compile:

create_run_jar:

copy_res:

BUILD SUCCESSFUL
Total time: 0 seconds
Options parsed, starting Sim.
Starting GUI…
3D [dev] 1.6.0-pre12-daily-experimental daily

2020-05-07 09:48:35.858 java[10760:681714] Apple AWT Internal Exception: NSWindow drag regions should only be invalidated on the Main Thread!
2020-05-07 09:48:35.858 java[10760:681714] *** Terminating app due to uncaught exception ‘NSInternalInconsistencyException’, reason: ‘NSWindow drag regions should only be invalidated on the Main Thread!’
*** First throw call stack:
(
0 CoreFoundation 0x00007fff398e8d07 __exceptionPreprocess + 250
1 libobjc.A.dylib 0x00007fff7260c5bf objc_exception_throw + 48
2 CoreFoundation 0x00007fff3991150c -[NSException raise] + 9
3 AppKit 0x00007fff36b1824c -[NSWindow(NSWindow_Theme) _postWindowNeedsToResetDragMarginsUnlessPostingDisabled] + 310
4 AppKit 0x00007fff36affcb2 -[NSWindow _initContent:styleMask:backing:defer:contentView:] + 1416
5 AppKit 0x00007fff36aff723 -[NSWindow initWithContentRect:styleMask:backing:defer:] + 42
6 libnativewindow_macosx.jnilib 0x000000012a05cf6e Java_jogamp_nativewindow_macosx_OSXUtil_CreateNSWindow0 + 398
7 ??? 0x000000010d3b2407 0x0 + 4516946951
)
libc++abi.dylib: terminating with uncaught exception of type NSException
/Users/kevin/Firmware/Tools/jmavsim_run.sh: line 80: 10760 Abort trap: 6 java -XX:GCTimeRatio=20 -Djava.ext.dirs= -Djavax.accessibility.assistive_technologies=" " -jar jmavsim_run.jar $device $extra_args

[image]

julianoes, 2020-05-07:
Looks like this bug: https://jogamp.org/bugzilla/show_bug.cgi?id=1370

We should try the fix by updating the jogl libs.

You can try this attempt: https://github.com/PX4/Firmware/pull/14855

kainism, 2020-05-07:
> Looks like this bug: https://jogamp.org/bugzilla/show_bug.cgi?id=1370
> 
> We should try the fix by updating the jogl libs.
> 
> You can try this attempt: #14855

Appreciate your kingly reply! 
I just fixed this problem by replacing "adoptopenjdk 8" by development kit downloaded from "https://www.oracle.com/java/technologies/javase-jdk8-downloads.html".  
Again, thanks for your reply!

closed 2020-05-07 by julianoes