PX4/PX4-Autopilot #14856: Unable to run the simulator on OSX Catalina with Gazebo or jmavsim

CoryFoy, 2020-05-07:
**Describe the bug**
I am able to _build_ PX4, but unable to _run_ it with either Gazbo or jmavsim. They both fail with different errors. 

For both I followed the instructions exactly from [the Mac guide](https://dev.px4.io/master/en/setup/dev_env_mac.html) as well as went through the troubleshooting steps, including doing a `make distclean`, setting the `ulimit` and following the instructions for the `cmake` error. 

 * [jmavsim](#jmavsim)
 * [Gazebo](#gazebo)

# jmavsim

For `jmavsim` - Here is the output. Note that the issue seems to be possibly something with the 3D library based on my research. I have tried it both with OpenJDK and Java SE's JDK. 

```
➜  Firmware git:(master) make px4_sitl jmavsim    
[0/1] cd /Users/foyc/Workspace/px4/Firmware/build/px4_sitl_default/tmp && /Users/foyc/Workspace/p...none /Users/foyc/Workspace/px4/Firmware /Users/foyc/Workspace/px4/Firmware/build/px4_sitl_defaul
SITL ARGS
sitl_bin: /Users/foyc/Workspace/px4/Firmware/build/px4_sitl_default/bin/px4
debugger: none
program: jmavsim
model: none
world: none
src_path: /Users/foyc/Workspace/px4/Firmware
build_path: /Users/foyc/Workspace/px4/Firmware/build/px4_sitl_default
empty model, setting iris as default
SITL COMMAND: "/Users/foyc/Workspace/px4/Firmware/build/px4_sitl_default/bin/px4" "/Users/foyc/Workspace/px4/Firmware"/ROMFS/px4fmu_common -s etc/init.d-posix/rcS -t "/Users/foyc/Workspace/px4/Firmware"/test_data
INFO  [px4] Creating symlink /Users/foyc/Workspace/px4/Firmware/ROMFS/px4fmu_common -> /Users/foyc/Workspace/px4/Firmware/build/px4_sitl_default/tmp/rootfs/etc

______  __   __    ___ 
| ___ \ \ \ / /   /   |
| |_/ /  \ V /   / /| |
|  __/   /   \  / /_| |
| |     / /^\ \ \___  |
\_|     \/   \/     |_/

px4 starting.

INFO  [px4] Calling startup script: /bin/sh etc/init.d-posix/rcS 0
INFO  [param] selected parameter default file eeprom/parameters_10016
[param] parameter file not found, creating eeprom/parameters_10016
  SYS_AUTOCONFIG: curr: 0 -> new: 1
* SYS_AUTOSTART: curr: 0 -> new: 10016
  BAT_N_CELLS: curr: 0 -> new: 3
  CAL_ACC0_ID: curr: 0 -> new: 1311244
  CAL_ACC_PRIME: curr: 0 -> new: 1311244
  CAL_GYRO0_ID: curr: 0 -> new: 2294028
  CAL_GYRO_PRIME: curr: 0 -> new: 2294028
  CAL_MAG0_ID: curr: 0 -> new: 197388
  CAL_MAG_PRIME: curr: 0 -> new: 197388
  CBRK_SUPPLY_CHK: curr: 0 -> new: 894281
  COM_DISARM_LAND: curr: 2.0000 -> new: 0.5000
  COM_OBL_ACT: curr: 0 -> new: 2
  COM_RC_IN_MODE: curr: 0 -> new: 1
  EKF2_ANGERR_INIT: curr: 0.1000 -> new: 0.0100
  EKF2_GBIAS_INIT: curr: 0.1000 -> new: 0.0100
  COM_ARM_EKF_AB: curr: 0.0017 -> new: 0.0050
  EKF2_REQ_GPS_H: curr: 10.0000 -> new: 0.5000
  IMU_INTEG_RATE: curr: 200 -> new: 250
  MC_PITCH_P: curr: 6.5000 -> new: 6.0000
  MC_PITCHRATE_P: curr: 0.1500 -> new: 0.2000
  MC_ROLL_P: curr: 6.5000 -> new: 6.0000
  MC_ROLLRATE_P: curr: 0.1500 -> new: 0.2000
  MPC_HOLD_MAX_Z: curr: 0.6000 -> new: 2.0000
  MPC_Z_VEL_I: curr: 0.1000 -> new: 0.1500
  MPC_Z_VEL_P: curr: 0.2000 -> new: 0.6000
  MPC_XY_P: curr: 0.9500 -> new: 0.8000
  MPC_XY_VEL_P: curr: 0.0900 -> new: 0.2000
  MPC_XY_VEL_D: curr: 0.0100 -> new: 0.0160
  MPC_SPOOLUP_TIME: curr: 1.0000 -> new: 0.5000
  MPC_TKO_RAMP_T: curr: 3.0000 -> new: 1.0000
  NAV_ACC_RAD: curr: 10.0000 -> new: 2.0000
[... 148 lines cut ...]

  Could not find a package configuration file provided by "gazebo" with any
  of the following names:

    gazeboConfig.cmake
    gazebo-config.cmake

  Add the installation prefix of "gazebo" to CMAKE_PREFIX_PATH or set
  "gazebo_DIR" to a directory containing one of the above files.  If "gazebo"
  provides a separate development package or SDK, be sure it has been
  installed.

-- Configuring incomplete, errors occurred!
See also "/Users/foyc/Workspace/px4/Firmware/build/px4_sitl_default/build_gazebo/CMakeFiles/CMakeOutput.log".
[27/741] Building CXX object src/lib/ecl/EKF/CMakeFiles/ecl_EKF.dir/ekf.cpp.o
FAILED: external/Stamp/sitl_gazebo/sitl_gazebo-configure 
cd /Users/foyc/Workspace/px4/Firmware/build/px4_sitl_default/build_gazebo && /usr/local/Cellar/cmake/3.17.2/bin/cmake -DCMAKE_INSTALL_PREFIX=/usr/local -DSEND_ODOMETRY_DATA=ON -GNinja /Users/foyc/Workspace/px4/Firmware/Tools/sitl_gazebo && /usr/local/Cellar/cmake/3.17.2/bin/cmake -E touch /Users/foyc/Workspace/px4/Firmware/build/px4_sitl_default/external/Stamp/sitl_gazebo/sitl_gazebo-configure
[33/741] Building CXX object src/lib/ecl/EKF/CMakeFiles/ecl_EKF.dir/covariance.cpp.o
ninja: build stopped: subcommand failed.
make: *** [px4_sitl] Error 1
```

I found https://github.com/PX4/Firmware/issues/5393 from 2016 which talked about needing libgazebo7-dev. It's possible that I missed something in the installation instructions. I wasn't sure if I should install Gazebo from source.

If I can provide any other information or test anything happy to! I know how running an open-source project can be.

kainism, 2020-05-08:
I came across the exactly same error with you!  For the first problem, I solved it by https://github.com/PX4/Firmware/issues/14854.

CoryFoy, 2020-05-13:
(For anyone following up on this, replacing my Java version as listed in #14854 did *not* fix the issue, and I never did get PX4 working at all even trying several things to build from source, so I ended up giving up on it and will try again maybe in a few months with new releases)

julianoes, 2020-05-14:
Sorry, I closed it too early then.

So you're saying Gazebe still does not build? Also not after a `brew upgrade`?

For the jMAVSim problem I made [a pull request](https://github.com/PX4/Firmware/pull/14855) that you could try out. The problem is that no one seems to know a lot about java and the java3d dependencies in order to fix jMAVSim.

CoryFoy, 2020-05-14:
@julianoes Awesome, thanks for reopening. That's correct - neither is building. I tried just now to reinstall `Gazbo` with brew, but I'm getting:

```
/Library/Developer/CommandLineTools/usr/bin/../include/c++/v1/cmath:329:9: error: no member named 'isunordered' in the global namespace
using ::isunordered;
```

which doesn't seem like an issue with PX4 at all, but I don't have time to debug right this second. I'll look at it at the PR you so kindly put up within the next 36 hours and report back!

CoryFoy, 2020-05-17:
Thanks so much again for reopening. I couldn't build Gazebo, but finally did after following https://github.com/Homebrew/brew/issues/5068 which helped out the most. If you have both XCode and CommandLineTools it looks like there are conflicts between them. I ended up deleting CommandLineTools and could build Gazebo again. But I still can't build PX4. I've put the error I get below. I'm chalking this up to probably having a messed up system, so we can just close it unless you see something in here that you think I could be doing wrong.

Thanks!

```
-- The CXX compiler identification is AppleClang 11.0.3.11030032
-- The C compiler identification is AppleClang 11.0.3.11030032
-- The ASM compiler identification is Clang
-- Found assembler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc
-- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++
-- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++ - works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc
-- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc - works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Building for code coverage
-- Found PythonInterp: /usr/local/bin/python3 (found suitable version "3.7.7", minimum required is "3") 
-- build type is RelWithDebInfo
-- PX4 ECL: Very lightweight Estimation & Control Library v1.9.0-rc1-286-g24f2e60
-- Configuring done
-- Generating done
-- Build files have been written to: /Users/foyc/Workspace/px4/Firmware/build/px4_sitl_default
[9/741] git submodule src/drivers/gps/devices
[12/741] git submodule src/lib/ecl
[13/741] git submodule mavlink/include/mavlink/v2.0
[15/741] git submodule Tools/sitl_gazebo
[20/741] Performing configure step for 'sitl_gazebo'
-- install-prefix: /usr/local
-- The C compiler identification is AppleClang 11.0.3.11030032
-- The CXX compiler identification is AppleClang 11.0.3.11030032
-- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc
-- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc - works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++
-- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++ - works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Performing Test COMPILER_SUPPORTS_CXX17
-- Performing Test COMPILER_SUPPORTS_CXX17 - Success
-- Performing Test COMPILER_SUPPORTS_CXX14
-- Performing Test COMPILER_SUPPORTS_CXX14 - Success
-- Performing Test COMPILER_SUPPORTS_CXX11
-- Performing Test COMPILER_SUPPORTS_CXX11 - Success
-- Performing Test COMPILER_SUPPORTS_CXX0X
-- Performing Test COMPILER_SUPPORTS_CXX0X - Success
-- Using C++17 compiler
-- Found Boost: /usr/local/lib/cmake/Boost-1.72.0/BoostConfig.cmake (found suitable version "1.72.0", minimum required is "1.58") found components: system thread filesystem 
CMake Warning (dev) at /usr/local/Cellar/cmake/3.17.2/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:272 (message):
  The package name passed to `find_package_handle_standard_args` (PkgConfig)
  does not match the name of the calling package (gazebo).  This can lead to
  problems in calling code that expects `find_package` result variables
  (e.g., `_FOUND`) to follow a certain pattern.
Call Stack (most recent call first):
  /usr/local/Cellar/cmake/3.17.2/share/cmake/Modules/FindPkgConfig.cmake:45 (find_package_handle_standard_args)
  /usr/local/lib/cmake/gazebo/gazebo-config.cmake:23 (include)
  CMakeLists.txt:32 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Found PkgConfig: /usr/local/bin/pkg-config (found version "0.29.2") 
[... 78 lines cut ...]
    linked by target "gazebo_sonar_plugin" in directory /Users/foyc/Workspace/px4/Firmware/Tools/sitl_gazebo
    linked by target "mav_msgs" in directory /Users/foyc/Workspace/px4/Firmware/Tools/sitl_gazebo
    linked by target "gazebo_geotagged_images_plugin" in directory /Users/foyc/Workspace/px4/Firmware/Tools/sitl_gazebo
    linked by target "gazebo_opticalflow_mockup_plugin" in directory /Users/foyc/Workspace/px4/Firmware/Tools/sitl_gazebo
    linked by target "gazebo_gps_plugin" in directory /Users/foyc/Workspace/px4/Firmware/Tools/sitl_gazebo
    linked by target "gazebo_parachute_plugin" in directory /Users/foyc/Workspace/px4/Firmware/Tools/sitl_gazebo
    linked by target "gazebo_controller_interface" in directory /Users/foyc/Workspace/px4/Firmware/Tools/sitl_gazebo
[... cut ...]
cd /Users/foyc/Workspace/px4/Firmware/build/px4_sitl_default/build_gazebo && /usr/local/Cellar/cmake/3.17.2/bin/cmake -DCMAKE_INSTALL_PREFIX=/usr/local -DSEND_ODOMETRY_DATA=ON -GNinja /Users/foyc/Workspace/px4/Firmware/Tools/sitl_gazebo && /usr/local/Cellar/cmake/3.17.2/bin/cmake -E touch /Users/foyc/Workspace/px4/Firmware/build/px4_sitl_default/external/Stamp/sitl_gazebo/sitl_gazebo-configure
[249/741] Building CXX object src/modules/replay/CMakeFiles/modules__replay.dir/Replay.cpp.o
ninja: build stopped: subcommand failed.
make: *** [px4_sitl] Error 1
```

julianoes, 2020-05-18:
Ok, I think we're one step closer.

> CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
GSTREAMER_LIBRARIES (ADVANCED)

This would hint that some of the gstreamer dependencies are not available. They should have been installed via `brew install px4-sim-gazebo`:

https://github.com/PX4/homebrew-px4/blob/41f7ddd0af9b647470d4ff6f895392ba9e16b948/Formula/px4-sim-gazebo.rb#L12-L17

ryanjAA, 2020-05-18:
On a fresh MacOS 10.15.4 install there are still issues when gstreamer is installed. It will get past the above but fail at:

26/100] Generating /Users/ryandev/src/Firmware/Tools/sitl_gazebo/models/iris/iris.sdf
FAILED: /Users/ryandev/src/Firmware/Tools/sitl_gazebo/models/iris/iris.sdf 
cd /Users/ryandev/src/Firmware/Tools/sitl_gazebo && rm -f /Users/ryandev/src/Firmware/Tools/sitl_gazebo/models/iris/iris.sdf && /usr/local/bin/python3 /Users/ryandev/src/Firmware/Tools/sitl_gazebo/scripts/xacro.py -o /Users/ryandev/src/Firmware/Tools/sitl_gazebo/models/rotors_description/urdf/iris_base.urdf /Users/ryandev/src/Firmware/Tools/sitl_gazebo/models/rotors_description/urdf/iris_base.xacro enable_mavlink_interface:=true enable_ground_truth:=false enable_wind:=false enable_logging:=false rotors_description_dir:=/Users/ryandev/src/Firmware/Tools/sitl_gazebo/models/rotors_description send_vision_estimation:=false send_odometry:=true && gz sdf -p /Users/ryandev/src/Firmware/Tools/sitl_gazebo/models/rotors_description/urdf/iris_base.urdf >> /Users/ryandev/src/Firmware/Tools/sitl_gazebo/models/iris/iris.sdf && rm -f /Users/ryandev/src/Firmware/Tools/sitl_gazebo/models/rotors_description/urdf/iris_base.urdf

dyld: Library not loaded: /usr/local/opt/bullet/lib/libBulletSoftBody.2.88.dylib
  Referenced from: /usr/local/bin/gz
  Reason: image not found
/bin/sh: line 1: 37406 Abort trap: 6           gz sdf -p /Users/ryandev/src/Firmware/Tools/sitl_gazebo/models/rotors_description/urdf/iris_base.urdf >> /Users/ryandev/src/Firmware/Tools/sitl_gazebo/models/iris/iris.sdf
[43/100] Building CXX object CMakeFiles/gazebo_geotagged_images_plugin.dir/src/gazebo_geotagged_images_plugin.cpp.o
ninja: build stopped: subcommand failed.
[737/741] Linking CXX executable bin/px4
FAILED: external/Stamp/sitl_gazebo/sitl_gazebo-build 
cd /Users/ryandev/src/Firmware/build/px4_sitl_default/build_gazebo && /usr/local/Cellar/cmake/3.17.2/bin/cmake --build .
ninja: build stopped: subcommand failed.

• The libBulletSoftBody.2.88.dylib error issue is referenced at https://github.com/osrf/homebrew-simulation/issues/1003 so @CoryFoy once you get streamer installed you'll hit a wall just after at the libBulletSoftBody.2.88.dylib problem.

PX4 builds fine, just not gazebo but I realize this is actually a bigger problem since any of the new Mac computers bought that come with 10.15.x (Catalina) can't be downgraded to 10.14 so no simulation work can be done on them unless dual booting into Linux or an equivalent setup. 

Here's the full output if you're curious. 

make px4_sitl gazebo_plane                               
-- PX4 version: v1.11.0-beta2-21-ga605444462
-- PX4 config file: /Users/ryandev/src/Firmware/boards/px4/sitl/default.cmake
-- PX4 config: px4_sitl_default
-- PX4 platform: posix
-- PX4 lockstep: enabled
-- cmake build type: RelWithDebInfo
-- The CXX compiler identification is AppleClang 11.0.3.11030032
-- The C compiler identification is AppleClang 11.0.3.11030032
-- The ASM compiler identification is Clang
-- Found assembler: /Library/Developer/CommandLineTools/usr/bin/cc
-- Check for working CXX compiler: /Library/Developer/CommandLineTools/usr/bin/c++
-- Check for working CXX compiler: /Library/Developer/CommandLineTools/usr/bin/c++ - works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /Library/Developer/CommandLineTools/usr/bin/cc
-- Check for working C compiler: /Library/Developer/CommandLineTools/usr/bin/cc - works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Building for code coverage
-- Found PythonInterp: /usr/local/bin/python3 (found suitable version "3.7.7", minimum required is "3") 
-- build type is RelWithDebInfo
-- PX4 ECL: Very lightweight Estimation & Control Library v1.9.0-rc1-286-g24f2e60
-- Configuring done
-- Generating done
-- Build files have been written to: /Users/ryandev/src/Firmware/build/px4_sitl_default
[0/741] git submodule src/drivers/gps/devices
[6/741] git submodule src/lib/ecl
[9/741] git submodule mavlink/include/mavlink/v2.0
[13/741] git submodule Tools/sitl_gazebo
[21/741] Performing configure step for 'sitl_gazebo'
-- install-prefix: /usr/local
-- The C compiler identification is AppleClang 11.0.3.11030032
-- The CXX compiler identification is AppleClang 11.0.3.11030032
-- Check for working C compiler: /Library/Developer/CommandLineTools/usr/bin/cc
-- Check for working C compiler: /Library/Developer/CommandLineTools/usr/bin/cc - works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /Library/Developer/CommandLineTools/usr/bin/c++
-- Check for working CXX compiler: /Library/Developer/CommandLineTools/usr/bin/c++ - works
-- Detecting CXX compiler ABI info
[... 134 lines cut ...]
[19/100] Generating /Users/ryandev/src/Firmware/Tools/sitl_gazebo/models/px4flow/px4flow-gen.sdf
/Users/ryandev/src/Firmware/Tools/sitl_gazebo/models/px4flow/px4flow.sdf.jinja -> /Users/ryandev/src/Firmware/Tools/sitl_gazebo/models/px4flow/px4flow-gen.sdf
[20/100] Generating /Users/ryandev/src/Firmware/Tools/sitl_gazebo/models/r200/r200-gen.sdf
[... cut ...]
/bin/sh: line 1: 37406 Abort trap: 6           gz sdf -p /Users/ryandev/src/Firmware/Tools/sitl_gazebo/models/rotors_description/urdf/iris_base.urdf >> /Users/ryandev/src/Firmware/Tools/sitl_gazebo/models/iris/iris.sdf
[43/100] Building CXX object CMakeFiles/gazebo_geotagged_images_plugin.dir/src/gazebo_geotagged_images_plugin.cpp.o
ninja: build stopped: subcommand failed.
[737/741] Linking CXX executable bin/px4
FAILED: external/Stamp/sitl_gazebo/sitl_gazebo-build 
cd /Users/ryandev/src/Firmware/build/px4_sitl_default/build_gazebo && /usr/local/Cellar/cmake/3.17.2/bin/cmake --build .
ninja: build stopped: subcommand failed.
make: *** [px4_sitl] Error 1

CoryFoy, 2020-05-19:
Yep, I'll confirm that after running the brew install and rerunning px4_sitl I see a similar error to above. May just need to try building all this from scratch.

```
➜  Firmware git:(master) make px4_sitl gazebo
[343/494] Performing configure step for 'sitl_gazebo'
-- install-prefix: /usr/local
-- Using C++17 compiler
-- Found Boost: /usr/local/lib/cmake/Boost-1.72.0/BoostConfig.cmake (found suitable version "1.72.0", minimum required is "1.58") found components: system thread filesystem 
CMake Warning (dev) at /usr/local/Cellar/cmake/3.17.2/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:272 (message):
  The package name passed to `find_package_handle_standard_args` (PkgConfig)
  does not match the name of the calling package (gazebo).  This can lead to
  problems in calling code that expects `find_package` result variables
  (e.g., `_FOUND`) to follow a certain pattern.
Call Stack (most recent call first):
  /usr/local/Cellar/cmake/3.17.2/share/cmake/Modules/FindPkgConfig.cmake:45 (find_package_handle_standard_args)
  /usr/local/lib/cmake/gazebo/gazebo-config.cmake:30 (include)
  CMakeLists.txt:32 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Found fcl: /usr/local/Cellar/fcl/0.6.1/include (found suitable version "0.6.1", minimum required is "0.3.2") 
-- Found assimp: /usr/local/include (found version "5.0.0") 
-- Found DART: /usr/local/include (Required is at least version "6") found components: dart 
-- Found Boost: /usr/local/lib/cmake/Boost-1.72.0/BoostConfig.cmake (found suitable version "1.72.0", minimum required is "1.40.0") found components: thread system filesystem program_options regex iostreams date_time 
-- Found Boost: /usr/local/lib/cmake/Boost-1.72.0/BoostConfig.cmake (found version "1.72.0")  
-- Looking for OGRE...
-- Found Ogre Ghadamon (1.9.0)
-- Looking for OGRE_Paging...
-- Found OGRE_Paging: optimized;/usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_8/lib/libOgrePaging.dylib;debug;/usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_8/lib/libOgrePaging.dylib
-- Looking for OGRE_Terrain...
-- Found OGRE_Terrain: optimized;/usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_8/lib/libOgreTerrain.dylib;debug;/usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_8/lib/libOgreTerrain.dylib
-- Looking for OGRE_Property...
-- Found OGRE_Property: optimized;/usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_8/lib/libOgreProperty.dylib;debug;/usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_8/lib/libOgreProperty.dylib
-- Looking for OGRE_RTShaderSystem...
-- Found OGRE_RTShaderSystem: optimized;/usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_8/lib/libOgreRTShaderSystem.dylib;debug;/usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_8/lib/libOgreRTShaderSystem.dylib
-- Looking for OGRE_Volume...
-- Found OGRE_Volume: optimized;/usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_8/lib/libOgreVolume.dylib;debug;/usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_8/lib/libOgreVolume.dylib
-- Looking for OGRE_Overlay...
-- Found OGRE_Overlay: optimized;/usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_8/lib/libOgreOverlay.dylib;debug;/usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_8/lib/libOgreOverlay.dylib
-- Found Protobuf: /usr/local/lib/libprotobuf.dylib (found suitable version "3.11.4", minimum required is "2.3.0") 
-- Config-file not installed for ZeroMQ -- checking for pkg-config
-- Checking for module 'libzmq >= 4'
--   Found libzmq , version 4.3.2
-- Found ZeroMQ: TRUE (Required is at least version "4") 
-- Checking for module 'uuid'
--   Found uuid, version 1.6.2
-- Found UUID: TRUE  
-- Checking for module 'tinyxml2'
--   Found tinyxml2, version 8.0.0
-- Looking for dlfcn.h - found
-- Looking for libdl - found
-- Found DL: TRUE  
-- FreeImage.pc not found, we will search for FreeImage_INCLUDE_DIRS and FreeImage_LIBRARIES
-- Found UUID: TRUE  
-- Checking for module 'gts'
--   Found gts, version 0.7.6
-- Found GTS: TRUE  
-- Checking for module 'libswscale'
--   Found libswscale, version 5.5.100
-- Found SWSCALE: TRUE  
-- Checking for module 'libavdevice >= 56.4.100'
--   Found libavdevice , version 58.8.100
-- Found AVDEVICE: TRUE (Required is at least version "56.4.100") 
-- Checking for module 'libavformat'
--   Found libavformat, version 58.29.100
-- Found AVFORMAT: TRUE  
-- Checking for module 'libavcodec'
--   Found libavcodec, version 58.54.100
-- Found AVCODEC: TRUE  
-- Checking for module 'libavutil'
--   Found libavutil, version 56.31.100
[... 31 lines cut ...]
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/ranlib: file: src/drivers/camera_trigger/libdrivers__camera_trigger.a(seagull_map2.cpp.o) has no symbols
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/ranlib: file: src/drivers/camera_trigger/libdrivers__camera_trigger.a(gpio.cpp.o) has no symbols
[372/494] Linking CXX static library src/lib/bezier/libbezier.a
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/ranlib: file: src/lib/bezier/libbezier.a(BezierQuad.cpp.o) has no symbols
[... cut ...]
/Users/foyc/Workspace/px4/Firmware/Tools/sitl_gazebo/models/c920/c920.sdf.jinja -> /Users/foyc/Workspace/px4/Firmware/Tools/sitl_gazebo/models/c920/c920-gen.sdf
[19/100] Building CXX object CMakeFiles/nav_msgs.dir/Odometry.pb.cc.o
ninja: build stopped: subcommand failed.
FAILED: external/Stamp/sitl_gazebo/sitl_gazebo-build 
cd /Users/foyc/Workspace/px4/Firmware/build/px4_sitl_default/build_gazebo && /usr/local/Cellar/cmake/3.17.2/bin/cmake --build .
ninja: build stopped: subcommand failed.
make: *** [px4_sitl] Error 1
```

julianoes, 2020-05-19:
Looks like the gazebo dependencies are not linking properly:
```
dyld: Library not loaded: /usr/local/opt/bullet/lib/libBulletSoftBody.2.88.dylib
```

It looks like the fix/update is coming here: https://github.com/osrf/homebrew-simulation/issues/1003

jank324, 2020-06-09:
The Gazebo issue looks a lot like something I get every time after I run `brew upgrade`. A reliable workaround for me has been to run `brew reinstall --build-from-source gazebo9` after upgrading Homebrew. This is with Gazebo 9 on macOS 10.15.5.

julianoes, 2020-06-09:
Usually you should not have to rebuild unless the [bottles provided by OSRF](https://github.com/osrf/homebrew-simulation/) are outdated.

However, you do have to do a `make clean` of PX4 and the Gazebo plugins.

ryanjAA, 2020-06-09:
I can confirm this works.

Did the following:

brew upgrade

brew reinstall --build-from-source gazebo9

make clean on PX4 firmware 

Gazebo 9 working under macOS 10.15.4 with PX4 Master

cglusky, 2020-06-16:
Confirm above as well.  macOS 10.5.5

I only have Command Line Tools - no Xcode installed.  

Tried jMAVSim but ended up with too new a jdk I think.

FYI, there does seem to be differences between master and stable in macOS toolchain wiki instructions regarding brew packages .  

Master ```brew cask install xquartz
brew install px4-sim-gazebo```

1.10
```'# Optional, but recommended additional simulation tools:
brew install px4-sim```

CoryFoy, 2020-09-21:
Closing because I won't have the time to loop back on this and no reason for it to sit in the queue if I can't do that!

closed 2020-09-21 by CoryFoy

dagar, 2020-09-21:
FYI with current PX4 (master or stable v1.11) both gazebo and jmavsim are working for me with MacOS Catalina.