PX4/PX4-Autopilot #14857: PX4 do not use RTK GPS heading instead of magnetometer

lukegluke, 2020-05-07:
**Describe the bug**
I would like to use px4 on rover without magnetometer, but with RTK GPS heading from one receiver with two antennas for yaw instead.
Heading is parsed from HDT nmea and is correctly set to heading filed of vehicle_gps_position_s (I’ve checked it). According to [documentation](https://docs.px4.io/master/en/gps_compass/rtk_gps.html#use-rtk-gps-for-yaw) I’ve set set bit position 7 in EKF2_AID_MASK to 1 and EKF2_MAG_TYPE to 5 (None).
But QGC is still showing course by magnetometer.
When I set CAL_MAG0_EN to 0 px4 stops sending attitude, local and global postion.
_SYS_HAS_MAG 0 doesn’t influence on anything (I suppose it is only related to arming checks, right?)_

**Expected behavior**
I expected px4 to use yaw from RTK GPS heading instead of magnetometer.

**Board**
px4_fmu-v2
px4 firmware almost latest master 33dc0fd14699db98181a311285ae8eda4f4983c5 with minor additions

bresch, 2020-05-08:
How did you conclude that the EKF is still using the magnetometer? Could you provide a log?

lukegluke, 2020-05-08:
> How did you conclude that the EKF is still using the magnetometer?

As I said QGC is showing course by magnetometer. When I rotate board by yaw (only board, antennas are fixed) compass is rotating accordingly.
[image]
log: https://logs.px4.io/plot_app?log=806475d7-d4df-4459-ba35-7e82ccbca839

Also when I set CAL_MAG0_EN to 0 EKF stops working, px4 doesn't send attitude, local and global positions.
log: https://logs.px4.io/plot_app?log=a2aabe4d-e302-4bd4-9633-f7d673bc0c77

lukegluke, 2020-05-08:
Notice that when EKF2_MAG_TYPE is 5 (None) yaw is always 0 on startup despite the last board direction, so it looks like EKF2 is somehow trying to use other sources for yaw, but still use magnetometer data.

bresch, 2020-05-12:
@lukegluke I didn't check in your log if the GPS yaw was correctly fused, but the mag isn't fused. The yaw changes when you turn the board because of gyro integration

lukegluke, 2020-05-12:
> The yaw changes when you turn the board because of gyro integration

Oh, ok, that's explain huge bounces in trajectory, especially on turns with none EKF2_MAG_TYPE. Thanks, sorry for misunderstanding. But how to disable gyro integration and use only RTK GPS heading in theory?

bresch, 2020-05-13:
@lukegluke You shouldn't do that, the gyro is used to predict the new heading between two absolute measurement updates. You shouldn't disable the gyro integration but just make the RTK work properly. I tried a few month ago and it worked properly. I don't have time to help you debugging now, but here are a few pointers:
- GPS yaw fusion is done here: https://github.com/PX4/ecl/blob/master/EKF/gps_yaw_fusion.cpp
- fusion is started here: https://github.com/PX4/ecl/blob/24f2e60b7eff664d0035d5f75f410073be28d894/EKF/control.cpp#L528-L547
- in the ulog file, you need to check the `heading` field in `estimation_innovation_variances`, `estimator_innovation_test_ratios`, `estimator_innovations`, ...

Please tell me if you find an issue.

closed 2020-06-22 by bresch