PX4/PX4-Autopilot #14876: Yaw estimate error in Vision based navigation in SITL in v1.10.1 and v1.11.0

robin-shaun, 2020-05-10:
**Describe the bug**
I have developed some demos for PX4 SITL, open sourced in [XTDrone](https://github.com/robin-shaun/XTDrone). For v1.8.0, v1.9.2 and v1.10.0, vision based navigation works well. However, for v1.10.1 and v.1.11.0, yaw estimate error happens. Everything is the same but the version of PX4 is different. I'm not sure why.

**To Reproduce**
For rcS in v1.10.1 and v.11.0
```
param set EKF2_AID_MASK 24
param set EKF2_MAG_TYPE 5
param set EKF2_HGT_MODE 3
```
And then send the ` /mavros/vision_pose/pose` topic

**Expected behavior**
`FCU: Preflight Fail: Yaw estimate error`

mhkabir, 2020-05-10:
It may be this https://github.com/PX4/ecl/pull/807

robin-shaun, 2020-05-10:
Thanks a lot. However, `param set EKF2_MAG_TYPE 5` means that the vehicle doesn't use mag. In this way, only gyro and vision are used for yaw estimation.

mhkabir, 2020-05-10:
EKF2_MAG_TYPE = 5 doesn't function correctly right now for vision based navigation when I last checked. It will cause filter divergence. Try setting it to 0/Auto and apply my patch.

robin-shaun, 2020-05-10:
Thank you very much! When I changed to `param set EKF2_MAG_TYPE 0`, vision-based navigation works well. And I have another question. When I set EKF2_AID_MASK=25, which means GPS is used as well, sometimes `WARN  [ecl/EKF] 331992000: GPS fusion timeout - reset to GPS` happens. Is this a serious problem？

mhkabir, 2020-05-10:
> Does it mean that vision position and GPS position conflict

Yes. You need to use vision velocity if you want to use simultaneously with GPS.

robin-shaun, 2020-05-10:
Yes, I know that from [https://docs.px4.io/master/en/advanced_config/tuning_the_ecl_ekf.html#ekf2_extvis](https://docs.px4.io/master/en/advanced_config/tuning_the_ecl_ekf.html#ekf2_extvis). However, in the qgroundcontrol 3.5.6, I cannot see vision velocity, there is no bit 8. Is it ok to just set the param in rcS.
[image]

bresch, 2022-08-15:
the velocity bit is displayed properly in newer QGC version (with up-to-date metadata)

closed 2022-08-15 by bresch