PX4/PX4-Autopilot #14877: Rover offboard control error in vision-based navigation in  v1.11.0 SITL

robin-shaun, 2020-05-10:
**Describe the bug**
When gps is used, I can control the rover in the offboard mode. However, when vision pose is used without GPS, and setpoint_raw is sent, the rover goes back a little and then doesn't move any more. Does it mean that global position must be supplied. But [https://docs.px4.io/master/en/flight_modes/offboard.html#rover](https://docs.px4.io/master/en/flight_modes/offboard.html#rover) shows that local NED is OK.
[image]

I'm sure the vision pose is OK and local_pose is correct, and the rover can be armed in offboard mode.

```
header: 
  seq: 1932
  stamp: 
    secs: 71
    nsecs: 200032584
  frame_id: "map"
pose: 
  position: 
    x: -0.800802469254
    y: -0.0107255671173
    z: -0.0394087657332
  orientation: 
    x: 0.00714431426672
    y: -0.0149615312873
    z: 0.00369294901337
    w: -0.999855798517
```

```
header: 
  seq: 39
  stamp: 
    secs: 39
    nsecs: 300000000
  frame_id: ''
connected: True
armed: True
guided: True
manual_input: False
mode: "OFFBOARD"
system_status: 4

```

julianoes, 2020-05-13:
I don't think there is enough information here for anyone to fix this. I think it would require more information like steps to reproduce and a log.

And alternatively, feel free to debug this and suggest a fix yourself.

julianoes, 2020-08-24:
No information supplied, closing.

closed 2020-08-24 by julianoes