PX4/PX4-Autopilot #14878: Gazebo cv::undistortPoints compilation error

afwilkin, 2020-05-10:
**Describe the bug**
I am trying to get Gazebo to work. However, there seems to be an issue with OpenCV

**To Reproduce**
Followed install directions [here](https://dev.px4.io/v1.9.0/en/setup/dev_env_linux.html#jmavsimgazebo-simulation)
cd ~/src/Firmware
make px4_sitl gazebo

**Log Files and Screenshots**
> [90/100] Building CXX object OpticalFlow/CMakeFiles/OpticalFlow.dir/src/flow_opencv.cpp.o
FAILED: OpticalFlow/CMakeFiles/OpticalFlow.dir/src/flow_opencv.cpp.o 
/usr/bin/c++  -DLIBBULLET_VERSION=2.87 -DLIBBULLET_VERSION_GT_282 -DOpticalFlow_EXPORTS -I/home/atlaslinux/src/Firmware/Tools/sitl_gazebo/external/OpticalFlow/OpticalFlow_INCLUDE_DIRS -I/home/atlaslinux/src/Firmware/Tools/sitl_gazebo/external/OpticalFlow/include -I/home/atlaslinux/src/Firmware/Tools/sitl_gazebo/external/OpticalFlow/external/klt_feature_tracker/include -isystem /usr/include/gazebo-9 -isystem /usr/include/bullet -isystem /usr/include/simbody -isystem /usr/local/include -isystem /usr/include/sdformat-6.0 -isystem /usr/include/ignition/math4 -isystem /usr/include/OGRE -isystem /usr/include/OGRE/Terrain -isystem /usr/include/OGRE/Paging -isystem /usr/include/ignition/transport4 -isystem /usr/include/ignition/msgs1 -isystem /usr/include/ignition/common1 -isystem /usr/include/ignition/fuel_tools1 -isystem /usr/include/x86_64-linux-gnu/qt5 -isystem /usr/include/x86_64-linux-gnu/qt5/QtCore -isystem /usr/lib/x86_64-linux-gnu/qt5/mkspecs/linux-g++ -isystem /usr/local/include/opencv4 -fpic -fPIC   -std=gnu++17 -MD -MT OpticalFlow/CMakeFiles/OpticalFlow.dir/src/flow_opencv.cpp.o -MF OpticalFlow/CMakeFiles/OpticalFlow.dir/src/flow_opencv.cpp.o.d -o OpticalFlow/CMakeFiles/OpticalFlow.dir/src/flow_opencv.cpp.o -c /home/atlaslinux/src/Firmware/Tools/sitl_gazebo/external/OpticalFlow/src/flow_opencv.cpp
/home/atlaslinux/src/Firmware/Tools/sitl_gazebo/external/OpticalFlow/src/flow_opencv.cpp: In member function ‘virtual int OpticalFlowOpenCV::calcFlow(uint8_t*, const uint32_t&, int&, float&, float&)’:
/home/atlaslinux/src/Firmware/Tools/sitl_gazebo/external/OpticalFlow/src/flow_opencv.cpp:110:7: error: ‘undistortPoints’ is not a member of ‘cv’
   cv::undistortPoints(features_tmp, features_current, camera_matrix, camera_distortion);
       ^~~~~~~~~~~~~~~
[92/100] Building CXX object CMakeFiles/gazebo_gps_plugin.dir/src/gazebo_gps_plugin.cpp.o
In file included from /usr/include/gazebo-9/gazebo/common/CommonIface.hh:22,
                 from /usr/include/gazebo-9/gazebo/common/common.hh:12,
                 from /usr/include/gazebo-9/gazebo/gazebo_core.hh:19,
                 from /usr/include/gazebo-9/gazebo/gazebo.hh:20,
                 from /home/atlaslinux/src/Firmware/Tools/sitl_gazebo/include/common.h:27,
                 from /home/atlaslinux/src/Firmware/Tools/sitl_gazebo/include/gazebo_gps_plugin.h:37,
                 from /home/atlaslinux/src/Firmware/Tools/sitl_gazebo/src/gazebo_gps_plugin.cpp:27:
/usr/include/boost/uuid/sha1.hpp:13:97: note: #pragma message: This header is implementation detail and provided for backwards compatibility.
 #pragma message("This header is implementation detail and provided for backwards compatibility.")
                                                                                                 ^
[93/100] Building CXX object CMakeFiles/gazebo_gst_camera_plugin.dir/src/gazebo_gst_camera_plugin.cpp.o
In file included from /usr/include/gazebo-9/gazebo/common/CommonIface.hh:22,
                 from /usr/include/gazebo-9/gazebo/common/common.hh:12,
                 from /usr/include/gazebo-9/gazebo/gazebo_core.hh:19,
                 from /usr/include/gazebo-9/gazebo/gazebo.hh:20,
                 from /home/atlaslinux/src/Firmware/Tools/sitl_gazebo/include/gazebo_gst_camera_plugin.h:24,
                 from /home/atlaslinux/src/Firmware/Tools/sitl_gazebo/src/gazebo_gst_camera_plugin.cpp:24:
/usr/include/boost/uuid/sha1.hpp:13:97: note: #pragma message: This header is implementation detail and provided for backwards compatibility.
 #pragma message("This header is implementation detail and provided for backwards compatibility.")
                                                                                                 ^
[94/100] Building CXX object OpticalFlow/klt_feature_tracker/CMakeFiles/klt_feature_tracker.dir/src/trackFeatures.cpp.o
ninja: build stopped: subcommand failed.
[739/743] Linking CXX executable bin/px4
FAILED: external/Stamp/sitl_gazebo/sitl_gazebo-build 
cd /home/atlaslinux/src/Firmware/build/px4_sitl_default/build_gazebo && /usr/bin/cmake --build .
ninja: build stopped: subcommand failed.
make: *** [Makefile:207: px4_sitl] Error 1

**Additional context**
I tried to reinstall OpenCV
I tried to `make clean` the project and recompile

**System Info**
Distributor ID:	Ubuntu
Description:	Ubuntu 18.10
Release:	18.10
Codename:	cosmic

Any insight would be great, thank you

afwilkin, 2020-05-11:
So it looks like there might be some issues with my opencv that I installed.
I had downgraded to an older version for a third party library I was trying to build.

I am flushing my system of the install and starting again.

Will report back

afwilkin, 2020-05-11:
I am still experiencing an issue, even after fresh install. 

> [1/9] Building CXX object OpticalFlow/klt_feature_tracker/CMakeFiles/klt_feature_tracker.dir/src/trackFeatures.cpp.o
FAILED: OpticalFlow/klt_feature_tracker/CMakeFiles/klt_feature_tracker.dir/src/trackFeatures.cpp.o 
/usr/bin/c++  -DLIBBULLET_VERSION=2.87 -DLIBBULLET_VERSION_GT_282 -I/home/atlaslinux/src/Firmware/Tools/sitl_gazebo/external/OpticalFlow/external/klt_feature_tracker/include -isystem /usr/include/gazebo-9 -isystem /usr/include/bullet -isystem /usr/include/simbody -isystem /usr/local/include -isystem /usr/include/sdformat-6.0 -isystem /usr/include/ignition/math4 -isystem /usr/include/OGRE -isystem /usr/include/OGRE/Terrain -isystem /usr/include/OGRE/Paging -isystem /usr/include/ignition/transport4 -isystem /usr/include/ignition/msgs1 -isystem /usr/include/ignition/common1 -isystem /usr/include/ignition/fuel_tools1 -isystem /usr/include/x86_64-linux-gnu/qt5 -isystem /usr/include/x86_64-linux-gnu/qt5/QtCore -isystem /usr/lib/x86_64-linux-gnu/qt5/mkspecs/linux-g++ -isystem /usr/include/opencv -fpic   -std=gnu++17 -MD -MT OpticalFlow/klt_feature_tracker/CMakeFiles/klt_feature_tracker.dir/src/trackFeatures.cpp.o -MF OpticalFlow/klt_feature_tracker/CMakeFiles/klt_feature_tracker.dir/src/trackFeatures.cpp.o.d -o OpticalFlow/klt_feature_tracker/CMakeFiles/klt_feature_tracker.dir/src/trackFeatures.cpp.o -c /home/atlaslinux/src/Firmware/Tools/sitl_gazebo/external/OpticalFlow/external/klt_feature_tracker/src/trackFeatures.cpp
/home/atlaslinux/src/Firmware/Tools/sitl_gazebo/external/OpticalFlow/external/klt_feature_tracker/src/trackFeatures.cpp:40:10: fatal error: opencv/core/core.hpp: No such file or directory
 #include <opencv/core/core.hpp>
          ^~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
[5/9] Building CXX object OpticalFlow/CMakeFiles/OpticalFlow.dir/src/flow_opencv.cpp.o
FAILED: OpticalFlow/CMakeFiles/OpticalFlow.dir/src/flow_opencv.cpp.o 
/usr/bin/c++  -DLIBBULLET_VERSION=2.87 -DLIBBULLET_VERSION_GT_282 -DOpticalFlow_EXPORTS -I/home/atlaslinux/src/Firmware/Tools/sitl_gazebo/external/OpticalFlow/OpticalFlow_INCLUDE_DIRS -I/home/atlaslinux/src/Firmware/Tools/sitl_gazebo/external/OpticalFlow/include -I/home/atlaslinux/src/Firmware/Tools/sitl_gazebo/external/OpticalFlow/external/klt_feature_tracker/include -isystem /usr/include/gazebo-9 -isystem /usr/include/bullet -isystem /usr/include/simbody -isystem /usr/local/include -isystem /usr/include/sdformat-6.0 -isystem /usr/include/ignition/math4 -isystem /usr/include/OGRE -isystem /usr/include/OGRE/Terrain -isystem /usr/include/OGRE/Paging -isystem /usr/include/ignition/transport4 -isystem /usr/include/ignition/msgs1 -isystem /usr/include/ignition/common1 -isystem /usr/include/ignition/fuel_tools1 -isystem /usr/include/x86_64-linux-gnu/qt5 -isystem /usr/include/x86_64-linux-gnu/qt5/QtCore -isystem /usr/lib/x86_64-linux-gnu/qt5/mkspecs/linux-g++ -isystem /usr/include/opencv -fpic -fPIC   -std=gnu++17 -MD -MT OpticalFlow/CMakeFiles/OpticalFlow.dir/src/flow_opencv.cpp.o -MF OpticalFlow/CMakeFiles/OpticalFlow.dir/src/flow_opencv.cpp.o.d -o OpticalFlow/CMakeFiles/OpticalFlow.dir/src/flow_opencv.cpp.o -c /home/atlaslinux/src/Firmware/Tools/sitl_gazebo/external/OpticalFlow/src/flow_opencv.cpp
/home/atlaslinux/src/Firmware/Tools/sitl_gazebo/external/OpticalFlow/src/flow_opencv.cpp: In member function ‘virtual int OpticalFlowOpenCV::calcFlow(uint8_t*, const uint32_t&, int&, float&, float&)’:
/home/atlaslinux/src/Firmware/Tools/sitl_gazebo/external/OpticalFlow/src/flow_opencv.cpp:110:7: error: ‘undistortPoints’ is not a member of ‘cv’
   cv::undistortPoints(features_tmp, features_current, camera_matrix, camera_distortion);
       ^~~~~~~~~~~~~~~
[6/9] Linking CXX shared library libgazebo_video_stream_widget.so
ninja: build stopped: subcommand failed.

and when I go to that location

> atlaslinux@atlaslinux-Q551LB:/usr/include/opencv> ls
cvaux.h  cvaux.hpp  cv.h  cv.hpp  cvwimage.h  cxcore.h  cxcore.hpp  cxeigen.hpp  cxmisc.h  highgui.h  ml.h

However, when I go to opencv2 I find what I am looking for 

> atlaslinux@atlaslinux-Q551LB:/usr/include/opencv2/core> ls | grep core
core_c.h
core.hpp

afwilkin, 2020-05-11:
I am wondering if I need to attempt to download an older version of opencv

afwilkin, 2020-05-12:
I changed around some source code, and even after I include the direct path to cv::undistortPoints

> #include <opencv2/imgproc/imgproc.hpp>
#include <opencv2/opencv.hpp>

I am still getting an error. It is like the compiler just doesn't see the file.
I bet that a different version of the file gets included first. 

I will post updates

afwilkin, 2020-05-12:
[image]
this is the function that can not be found. I am including the file, but I am wondering if it is that export macro that could be messing things up

mhkabir, 2020-05-12:
Looks like a setup-specific problem, not a flightstack/toolchain issue. Recommend that you try the officially supported 18.04 LTS.

Feel free to continue the debugging on the forums at discuss.px4.io

closed 2020-05-12 by mhkabir

afwilkin, 2020-05-12:
@mhkabir my apologies for the clutter. I will try 18.04, was not aware that was the officially supported version. Thank you!