PX4/PX4-Autopilot #14886: RGB UI LED does not light up after start-up

roman-dvorak, 2020-05-11:
**Describe the bug**
Hi, I have recognised that RGB UI LED doesn't lightup automatically after startup. This behaviour was observed on about week old master branch (https://github.com/ThunderFly-aerospace/PX4Firmware/commit/070792cbd150729610a6afa2418fce6501724a88). It was tested on CUAV NANO and CUAV V5+ autopilots.  

Today updated master FW from QGC (1.11.0beta) and behaviour was the same. 

The tested airframe was our Auto-G2. But I don't think that it's related to the airframe.

**To Reproduce**
Steps to reproduce the issue:
1. power on autopilot with master FW
2. RGB UI LED doesn't light

**Details**
It looks like that `rgbled_pwm` driver is running after start. 
``` 
nsh> rgbled_pwm status
INFO  [rgbled_pwm] state: ON
INFO  [rgbled_pwm] red: 0, green: 0, blue: 56
```

I have to restart `rgbled_pwm` driver for light up UI LED. 
```
nsh> rgbled_pwm stop

nsh> rgbled_pwm status
WARN  [rgbled_pwm] not started
INFO  [rgbled_pwm] missing command: try 'start', 'status', 'stop'

nsh> rgbled_pwm start

nsh> rgbled_pwm status
INFO  [rgbled_pwm] state: ON
INFO  [rgbled_pwm] red: 0, green: 0, blue: 56
```
Now it starts blinking

dagar, 2020-05-11:
On those CUAV boards it's `rgbled_ncp5623c`. I'm currently looking at a CUAV v5+ that's still "breathing". Is that the light you're missing?

roman-dvorak, 2020-05-13:
Thanks for reply. That light is a great fast debugging tool :)
I have the same behaviour with CUAV V5+ as well as with CUAV nano. 

I can't start the `rgbled_ncp5623c` driver. I probably don't know the correct start parameters. The LED does not flash. And `rgbled_ncp5623c` doesn't start automatically. 

[image]

Paul-DROTEK, 2020-09-10:
Hi there,

Paul from Drotek here, we kind of have the same issue with the Pixhawk3Pro featuring the NCP5623C as the LED driver and using Pixhawk v1.11 and later.
We have to manually start the LED at boot up typing the "rgbled_ncp5623c -I start" command in the MAVLINK console.

Many thnks for your support.

roman-dvorak, 2020-11-06:
Hi, I just found out that it related to use of UAVCAN. If UAVCAN is off, the LED on the autopilot will start flash (after reboot). Is this behavior correct or is it a bug?

I think it should keep blinking at any CAN setting.

From the state described in the original post. Now I am not able to turn on the LED manually with CAN on. I'm getting 
[image]