PX4/PX4-Autopilot #14887: Flight Task - flight tasks update might change current task to None, even due we expect a running task in this point

garfieldG, 2020-05-11:
**Describe the bug**
whenever we wish to change the running task to any of the additional tasks (only Orbit for now) we do it when an active task is already supposed to run https://github.com/PX4/Firmware/blob/2bbdef205e163181cb8c02b08aeceb4c69101b36/src/modules/mc_pos_control/mc_pos_control_main.cpp#L592
https://github.com/PX4/Firmware/blob/2bbdef205e163181cb8c02b08aeceb4c69101b36/src/lib/flight_tasks/FlightTasks.cpp#L20 If something happens and switchTask fails https://github.com/PX4/Firmware/blob/2bbdef205e163181cb8c02b08aeceb4c69101b36/src/lib/flight_tasks/FlightTasks.cpp#L162 there will be no active task once FlightTasks::update() returns, although at this point we do expect an active task to run. 
Another critical thing -https://github.com/PX4/Firmware/blob/2bbdef205e163181cb8c02b08aeceb4c69101b36/src/modules/mc_pos_control/mc_pos_control_main.cpp#L695 with no active task is not a good idea

julianoes, 2020-05-13:
@MaEtUgR could you comment?

BazookaJoe1900, 2020-08-17:
still relevant