PX4/PX4-Autopilot #14888: Ubuntu toolchain setup script fails

Franky333, 2020-05-11:
**Describe the bug**
When running the Ubuntu setup script (Tools/setup/ubuntu.sh) on Ubuntu 20.04, the script stops running at this line: https://github.com/PX4/Firmware/blob/master/Tools/setup/ubuntu.sh#L144

**To Reproduce**
Run the script on Ubuntu 20.04, after installing the NuttX dependencies it stops. When setting GCC_FOUND_VER to 0 in the line linked above, the installation continues as expected.

julianoes, 2020-05-14:
@Franky333 can you check again?

Franky333, 2020-05-15:
@julianoes Not sure why but now I get this error before the toolchain installation: 
`./ubuntu.sh: line 149: [: missing `]'
And then the toolchain gets installed even though I already have it installed.
It seems like I had the same problem in my original report (where I didn't have the toolchain already installed) but instead of printing the error message the script just finished without doing anything else.

julianoes, 2020-05-15:
I just tried this in a fresh Ubuntu 18.04 docker and it worked just fine.

How did you download your `ubuntu.sh` file? Can you try the one downloaded with the repository?

Franky333, 2020-05-15:
I used a fresh download of the repository and launched the script from there.
Maybe the version of bash I am using (5.0.16(1)-release) behaves differently than yours?

hamishwillee, 2020-05-18:
@julianoes I can reproduce in a VM - see duplicate https://github.com/PX4/Firmware/issues/14916

I think it is something to do with the Grep of the version and the addition of "+x" but I don't understand this bit of the code. Hoping you can work out why it is falling over.

julianoes, 2020-05-18:
Sorry, I overlooked the error. Now I can reproduce it.

PR #14922 (2020-05-18): Tools: fix ubuntu setup script
This fixes the error:
`line 149: [: missing `]'`

Fixes #14916, fixes #14888.
closed 2020-05-18 by julianoes