PX4/PX4-Autopilot #14889: Changing EKF2_GPS_POS parameters causes sensor calibration error and failsafe

KimHyungSub, 2020-05-11:
**Describe the bug**
When I changed EKF2_GPS_POS_X, EKF2_GPS_POS_Y, EKF2_GPS_POS_Z parameters, it caused sensor calibration error and failsafe (no global position). But, these parameters do not require reboot based on documentation.

**To Reproduce**
Steps to reproduce the behavior:
1. Execute SITL with gazebo solo
2. assign 2 to  EKF2_GPS_POS_X, EKF2_GPS_POS_Y, and EKF2_GPS_POS_Z parameters
3. takeoff
4. You can see error

**Expected behavior**
Location has to be changed according to these EKF2_GPS_POS parameters.

**Log Files and Screenshots**
- Flight log file: https://drive.google.com/file/d/1gpXsA5eO_idea6A9XAZZvub_E2rZcGVL/view?usp=sharing
- PX4 Flight Review: https://logs.px4.io/plot_app?log=2fa71425-72cf-4656-99a9-4e39acc1b11d

Add screenshots to help explain your problem.
[image]

**Drone (please complete the following information):**
- SITL with with gazebo solo

bresch, 2020-05-12:
Well, I'm not sure if this is really a bug. You tell the estimator that the GPS receiver is 2m away in xyz from the CG. The data that comes in is then incorrectly adjusted and the estimator then diverges. I'm not sure what you would expect.

KimHyungSub, 2020-05-13:
@bresch 
I thought these parameters for compensation of GPS position on body frame like GPS_POS1_X, GPS_POS1_Y, GPS_POS1_Z on ArduPilot.
I wanted to simulate a plane which has a long body frame and GPS is attached on tail of the body.

bresch, 2020-05-14:
>I thought these parameters for compensation of GPS position on body frame like GPS_POS1_X, GPS_POS1_Y, GPS_POS1_Z on ArduPilot.

Yes, they are the same as in ArduPilot: the location in body frame of the GPS antenna.

>I wanted to simulate a plane which has a long body frame and GPS is attached on tail of the body.

Then you need to also offset the GPS data in the simulator, otherwise you won't simulate a vehicle with a GPS attached on the tail of the body.

closed 2020-05-14 by bresch