PX4/PX4-Autopilot #14890: RC override on SITL not working

KimHyungSub, 2020-05-11:
**Describe the bug**
I try to override RC channels on SITL by using mavutil. But, it seems likely to me that PX4 does not properly handle the overridden RC channels. I also noticed that many people already suffered from the same issues (#9782 / https://discuss.px4.io/t/sending-rc-channels-override-to-px4/15120 / https://discuss.px4.io/t/unable-to-send-rc-commands-to-px4/6844).

After I overrode the RC channels, GCS showed correct overridden RC values as below.
[image]

I also confirmed that MavlinkReceiver::handle_message_rc_channels_override() well received the RC override messages and then published uORB message as shown in below. 
[image]
[image]

Can you guys help me to solve the issue? I need to simulate manual control. Without the manual control, we can only test few flight modes which do not require a RC controller.

**Expected behavior**
Drone follows the overridden RC channels.

**Drone (please complete the following information):**
- SITL with gazebo

dagar, 2020-05-12:
Typically MANUAL_CONTROL (https://mavlink.io/en/messages/common.html#MANUAL_CONTROL) is preferred over faking RC.

dagar, 2020-05-12:
> it seems likely to me that PX4 does not properly handle the overridden RC channels

Can you expand on this? What's the actual problem? Did you try going through RC calibration?

Instead of the printf can you try `listener input_rc` and `listener manual_control_setpoint`?

KimHyungSub, 2020-05-13:
@dagar Thanks Dagar! I solved the problem using RC calibration using rc_channels_override_send().

closed 2020-05-13 by KimHyungSub

ceasergarcia, 2024-03-07:
@KimHyungSub I'm having the same issue. After an RC calibration, it is still not handling the override command. Do you mind posting a code snippet on how you created the override command?