PX4/PX4-Autopilot #14891: User can assign a negative value to MIS_TAKEOFF_ALT

KimHyungSub, 2020-05-11:
**Describe the bug**
As you know, MIS_TAKEOFF_ALT parameter indicates a minimum altitude the system will take off to.
I noticed that users can assign a negative value such as -10 to MIS_TAKEOFF_ALT, it causes QGroundControl targeted a weird altitude as shown in below.
[image]

**To Reproduce**
Steps to reproduce the behavior:
1. Execute SITL with gazebo (make px4_sitl gazebo_solo)
2. param set MIS_TAKEOFF_ALT -10
3. Take off on QGroundControl
4. See error

**Expected behavior**
PX4 should deny the negative altitude value. 

**Log Files and Screenshots**
- Flight log file: https://drive.google.com/file/d/1wlp1EJAyxbjMy08LcFUicfMXUbajmE1O/view?usp=sharing
- Share the link to a log: https://logs.px4.io/plot_app?log=cb606ea9-c94a-4597-be40-976df3c2ebd2

**Drone (please complete the following information):**
- SITL with gazebo

dagar, 2020-05-12:
It's limited in metadata (QGC parameter editor), but that's not stored onboard the embedded system. 

https://github.com/PX4/Firmware/blob/466b5db36f949146642145d2d6470ab2975b5ac6/src/modules/navigator/mission_params.c#L52-L53

To be really paranoid an additional check could be added here. https://github.com/PX4/Firmware/blob/4f2d39cb152962832021bb73d09e1f49306704ea/src/modules/navigator/mission.cpp#L1271-L1277

KimHyungSub, 2020-05-12:
@dagar 
Thank you for kind your answer. Could I add the checking code?