PX4/PX4-Autopilot #14900: Servo jittering after disarm

roman-dvorak, 2020-05-13:
**Describe the bug**
After disarm the servo actuator connected to MAIN1 output starts jittering. The intensity of this jittering varies. This state occurs only sometimes. The tested setup was CUAV Nano FMUv5 without IO co-processor. I have observed this behaviour with our Autogyro airframe. But it was repeated with a standard airplane airframe. It doesn't matter what the mixer is.

Firmware was master from 6.5.

**To Reproduce**
Steps to reproduce the behaviour:
1. Connected power on PPM ports and connect some servo to MAIN1 port
2. Switch autopilot on
3. Arm and disarm autopilot
4. See error - servo start jitter

**Video**
This video was recorded after standard disarm through RC
https://youtu.be/r_J6SnLum-c

This signal is only on 1st MAIN output. It looks like the same 'random' signal. It has been verified that this is not a power interference. A variable signal is really output from this port. It looks like the output is set from some memory location, which is overwritten by other values.

We were also surprised that the servos are not 'active' (no pwm output) after pressing the safety button. Only after the ARM. Is this behaviour normal with an autopilot without an IO processor?

dagar, 2020-05-14:
Do you have the corresponding log?

roman-dvorak, 2020-05-14:
I have a log that ends with disarming before starting the output jittering.. I haven't log directly with that jittering (I'll try to make one).

Sometimes it starts immediately after disarm. Sometime it takes some time (for example 10 minutes) and intensity is rising. 

https://review.px4.io/plot_app?log=60ca6141-07e0-4d88-8439-c661b4b81807

roman-dvorak, 2020-05-14:
I have created a new log and video:

https://review.px4.io/plot_app?log=0818c8ca-c9c5-4e3d-a841-8c612e26adef
https://youtu.be/vTO9dVAxjyM

Now I noticed strong noise of the magnetometer.

dagar, 2020-05-14:
I just wanted to confirm that PX4 wasn't sending anything. 

[image]

Does it stop if you unplug the telemetry radio?

roman-dvorak, 2020-05-23:
@julianoes 
Hi, today I got to the oscilloscope and I wanted to measure what the jittering signal look like.

So I connected oscilloscope probe to signal pin of MAIN1 output. Against ground from PPM header. The telemetry modem was disconnected during the measurement.  I armed autopilot and oscilloscope show nice PWM output with 3.3V logic. 

[image]

Then I disarmed autopilot and oscilloscope shows a problematic signal that was causing servo jittering:
[image]

Sometimes it stays for few seconds in logic 1 and sometimes zero-oriented peaks were significantly more frequent (without screenshot)

[image]

Any other pin was in login 0. 
[image]

That's why I think it's some bug.

roman-dvorak, 2020-06-11:
@dagar please, can you advise if it or it is not a software bug. Based on measurements from the last comment? I think there's something uninitialized. Or something is being written to the wrong place in memory.

dagar, 2020-06-11:
This is a CUAV v5 nano (no px4io) and it's only ever the first pin?

roman-dvorak, 2020-06-11:
Yes, it is only on CUAV v5 nano without the IO coprocessor. And it is only on the 1st pin labeled as MAIN1. 

I have trayed upload the same firmware build to CUAV v5+ (with px4io) and this issue was not observed.

Antiheavy, 2020-06-13:
@roman-dvorak do you have a 900 MHz radio (915 MHz) plugged into your system?  does moving the radio or antenna change things?  just curious....

roman-dvorak, 2020-06-13:
During the first test yes. But 433MHz radio from holybro. The disconnecting modem doesn't affect this. It was tested after some of the previous comment.

Nothing was connected to autopilot when measuring with an oscilloscope. I have trayed more power sources. USB from two computers, powering through power module and LiPo. Behavior was the same.

It is interesting that the 1st pin is in log1. And "peaks" are grounded (to log 0). Other outputs are still in log 0.

AlexandreBorowczyk, 2022-11-25:
@roman-dvorak 
Did you manage to find a solution? I believe we have the same issue. I have check with a oscilloscope but the servo motion is the same

roman-dvorak, 2022-11-25:
Hi, no I haven't fixed it and I don't even know if it still occurs with today's FW.
I solved it by not using the first PPM output. Others do not suffer from this problem. Fortunately, we had some outputs free.

AllenHuangDongyue, 2023-06-07:
The same occured on me. Is there any solutions?