PX4/PX4-Autopilot #14903: Takeoff altitude is not correctly respected

julianoes, 2020-05-14:
**Describe the bug**
A multicopter does not go to the correct takeoff altitude but actually lower, presumably it is `MIS_TAKEOFF_ALT` - `NAV_MC_ALT_RAD` because once it "reaches" takeoff altitude it switches to HOLD and doesn't keep climbing.  

**To Reproduce**
SITL:
```
param set NAV_RCL_ACT 0
param set NAV_DLL_ACT 0
param set MIS_TAKEOFF_ALT 2.5
commander takeoff
listener local_position -n 1
```

To "fix" it, set the threshold smaller:
```
param set NAV_MC_ALT_RAD 0.05
```

**Expected behavior**
It should go to the correct altitude as it used to.

**Log Files and Screenshots**
https://logs.px4.io/plot_app?log=eaae3517-b6bd-460a-be32-a2eae6506eb0

[image]

julianoes, 2020-05-14:
FYI @MaEtUgR

hamishwillee, 2020-08-16:
Shouldn't this one be fixed @MaEtUgR ?

julianoes, 2021-05-20:
It's still off by 5cm but it doesn't seem to depend on the mentioned param anymore. Closing.

closed 2021-05-20 by julianoes