PX4/PX4-Autopilot #14909: NAV_DELAY not working properly

julianoes, 2020-05-15:
**Describe the bug**
When a NAV_DELAY mission item is used, a multicopter oscillates back and forth around the waypoint where it is supposed to wait.

**To Reproduce**
Can be reproduced in SITL jMAVSim. Here is a plan file: 
[delay.zip](https://github.com/PX4/Firmware/files/4633983/delay.zip)

**Expected behavior**
It should nicely hold the position.

**Log Files and Screenshots**
https://logs.px4.io/plot_app?log=f284d761-8709-4701-bb26-1dba665b187a

[image]

This is required to resolve https://github.com/PX4/Firmware/issues/14904.

bresch, 2020-05-20:
Note that the issue does not occur if you set the hold time of the WP to 30s instead of adding a delay WP of 30s
[image]

bresch, 2020-05-20:
One strange thing with the delay WP (I used your .plan file), is that the current and next WP longitude and latitude are almost the same, but not exactly:
[image]

dayjaby, 2020-05-20:
> I'm pretty sure this commit doesn't create the issue but before this commit, this behavior was hidden by target *= scale_factor, because the Z vel is close to zero, so the whole vector was set to zero.

@bresch Yes, I agree, but I think it did not set the XY exactly to zero, but smoothed it towards zero. I tried setting XY to zero and it was stuck at ~ NAV_ACC_RAD meters before the waypoint. Looking forward to see your solution :)

dayjaby, 2020-06-04:
@bresch Did you have time yet to tackle this issue? Or can you give guidance on where to look as I don't have experience with the guidance logic in FlightTask.

bresch, 2020-06-25:
@dayjaby I think you should try to understand the difference between a WP with a hold time and a normal waypoint followed by a delay waypoint. Also, is the delay waypoint really useful? Why one should use the delay waypoint instead of a hold time?

dayjaby, 2020-06-26:
@bresch It is the most generic way to make the drone wait. You could for example have a sequence WAYPOINT, IMAGE_START_CAPTURE, DELAY. If WAYPOINT has a hold time, you are not able to perform any other mission items on this very waypoint unless you place multiple waypoints on top of each other - something that Lorenz mentioned in another PR he really does *not* want. He said DELAY is the proper way to handle those scenarios except that it's not working right now.

PR #15222 (2020-06-26): FlightTaskAuto: Update prev and next waypoints depending on validity
This fixes the corner case where a NAV_DELAY command changes the validity of the next WP but not the rest of the triplet. The logic in FlightTask was ignoring this change because the check was only based on WP position change.

fixes #14909 

@julianoes Could you please double check that this fixes the issue?
closed 2020-06-26 by dagar