PX4/PX4-Autopilot #14915: VTOL tiltrotor tilt ("pusher") support in hover #13779

pbecchi, 2020-05-16:
I have a custom design Vtol that i am actually testing as a quadrotor multicopter. The design is asimmetric: 2 small rotor forward and 2 main rotor on main wing that may tilt 90 °
As soon as it flight stable as a quadcopter i want to start tilting the wing and moving forward, keeping the aircraft horizzontal.
I think this should be feasible with the new feature of beta 1.11.
This feature is undocumented ....can somebody tell me how to activate it?
In addition my wing actuator (a powerfull servo) has a total stroke of 180° that through a linkage move the wing 90°.
Linkage and servo arm are aligned at 0° and 90° wing angle, therefore the pwm value is strongly nonlinear with the angle value .
There is any way to give a curve to the actuators output? Like the one you can give to an RC transmitter?
I coud not find anything that show me that possibility.....if not possible, i think i will use and external mcu board that convert actuation value to the necessary pwm values.

Anyhow this feature, if do not exist, will be highly appreciated!

sfuhrer, 2020-05-18:
So you want to use the tilt of your wind to move forward in hover, did I get that correctly? Then you need to set `VT_FWD_THRUST_EN`=4 to have it in auto, position controlled and altitude controlled modes. And probably set `VT_DWN_PITCH_MAX` smaller than it is now.
Then it should be enabled, and you can tune it via the param `VT_FWD_THRUST_SC` .
Note that this is purely for hovering, there is different code that handles the transition part of a tiltrotor VTOL.
About the non-linear PWM and tilt: afaik this is not possible atm, but you could also add it in a customized PX4 firmware.

pbecchi, 2020-05-19:
Thanks!
I will test wing tilt as soon as i finish PD parameters optimisation!
Regarding Vtol this is my understanding:
- i can fly it as a quadcopter for tilt angle <45° and forward speed ≤ 0.5*Vs where Vs is Stall speed.
- above these limit a should start considering it a Vtol in transition.
I have not seen a clear documentation on how Vtol transition is handled.
There should be a way to change mixer following the geometry changes and considering lifting surfaces forces. I have made a matlab analisys of transition trim points as a function of airspeed  and wing tilt that may lead to a number of different mixers, but i dont know how to feed that to PX4....
Any advice? 
For the non linear servo drive, i will make a simple dirty solution with an external mcu that read PWM out of PX4 and generate to correct PWM following a curve.

sfuhrer, 2020-05-26:
@pbecchi atm PX4 only supports static mixers, but we are hoping to improve that soon with a dynamic control allocation, something that's been started with https://github.com/PX4/Firmware/pull/13351. Any help for that is ofc appreciated!
About the current transition handling: It is mainly defined by an airspeed threshold (if you have a sensor). The motors are tilted to some angle (default is 45° I think), and they stay at this tilt until an airspeed threshold is reached.

sfuhrer, 2021-01-18:
FYI, the core algorithms for dynamic control allocation is now merged https://github.com/PX4/PX4-Autopilot/pull/16528

closed 2021-01-18 by sfuhrer