PX4/PX4-Autopilot #14917: ubuntu.sh: is that possible run `source ~/.profile` instead of suggest user reboot/relogin

bys1123, 2020-05-16:
**Describe problem solved by the proposed feature**
Simplify toolchain installation progress.

Current tip: `Reboot or logout, login computer before attempting to build NuttX targets`

hamishwillee, 2020-05-18:
I think we do this because usermod changes don't happen until after reboot/login. So in fact this may only affect the case where you also want to upload the new binary at the same time as build. That said, I suspect that will be a lot of cases.

Did you try building or building/uploading? If it works, perhaps we could be more specific note ie "before attemptiong to upload NuttX targets". Not sure it is worth it though.

bys1123, 2020-05-18:
@hamishwillee The script can run `source ~/.profile` at final to apply the PATH change to avoid reboot/login.
If do not do this, system can not find arm-gcc that just downloaded.
https://github.com/PX4/Firmware/blob/master/Tools/setup/ubuntu.sh#L158-L164

hamishwillee, 2020-05-18:
@bys1123 Can it find it anyway - as per your post https://github.com/PX4/Firmware/issues/14916 ? :-)

My point above though is that we can do this and they will be able to build. But they will still require a reboot if we want them to be able to do uploads because reloading th profile does not fix the usermod problem. So happy to make this change to load the profile, but we still require the reboot. 

Make sense.?

julianoes, 2020-05-18:
@bys1123 
> The script can run source ~/.profile at final to apply the PATH change to avoid reboot/login.

Correct but as @hamishwillee wrote the `usermod` changes require logout/login. Alternatively, you could use `newgrp` but like `source ~/.profile` you have to do it in every console and that's hard to explain to a user. In my opinion it's much easier to just tell users to logout and log back in.

bys1123, 2020-05-18:
@julianoes Understand, thanks for explain.

closed 2020-05-18 by bys1123