PX4/PX4-Autopilot #14918: ImportError('No module named toml',)

liangpu597, 2020-05-17:
My system is : ubuntu 16.04
First， I Use :` pip install toml`
and I clone px4, use:
`make px4_sitl_default gazebo`
and errors:
`Failed to import toml: No module named 'toml'`
`You may need to install it using:`
`    pip3 install --user toml`

and I try open a shell:

> python

> import toml

and Found toml is already installed, but it  errors in px4.

please tell me how to solve? thank you!

bys1123, 2020-05-17:
px4 is using python3

liangpu597, 2020-05-17:
> px4 is using python3

if this, how to solve? thank you!

bys1123, 2020-05-17:
`pip3 install --user toml`

liangpu597, 2020-05-17:
@bys1123 
ok , i follow your way. changed the default python version from 2 to 3, and follow the error pip3 install ***. and try again: `make px4_sitl_default gazebo`.  
and errors:   

> tony@tony-Inspiron-3542:~/PX4_Firmware$ make px4_sitl_default gazebo
> -- PX4 version: v1.11.0-beta1
> -- PX4 config file: /home/tony/PX4_Firmware/boards/px4/sitl/default.cmake
> -- PX4 config: px4_sitl_default
> -- PX4 platform: posix
> -- PX4 lockstep: enabled
> -- cmake build type: RelWithDebInfo
> -- The CXX compiler identification is GNU 5.4.0
> -- The C compiler identification is GNU 5.4.0
> -- The ASM compiler identification is GNU
> -- Found assembler: /usr/bin/cc
> -- Check for working CXX compiler using: Ninja
> -- Check for working CXX compiler using: Ninja -- works
> -- Detecting CXX compiler ABI info
> -- Detecting CXX compiler ABI info - done
> -- Detecting CXX compile features
> -- Detecting CXX compile features - done
> -- Check for working C compiler using: Ninja
> -- Check for working C compiler using: Ninja -- works
> -- Detecting C compiler ABI info
> -- Detecting C compiler ABI info - done
> -- Detecting C compile features
> -- Detecting C compile features - done
> -- Building for code coverage
> -- Found PythonInterp: /usr/bin/python3 (found suitable version "3.5.2", minimum required is "3") 
> -- build type is RelWithDebInfo
> -- PX4 ECL: Very lightweight Estimation & Control Library v1.9.0-rc1-255-g47624a0
> -- Configuring done
> -- Generating done
> -- Build files have been written to: /home/tony/PX4_Firmware/build/px4_sitl_default
> [7/740] git submodule src/drivers/gps/devices
> [10/740] git submodule src/lib/ecl
> [11/740] git submodule mavlink/include/mavlink/v2.0
> [206/740] git submodule Tools/sitl_gazebo
> [214/740] Performing configure step for 'sitl_gazebo'
> -- install-prefix: /usr/local
> -- The C compiler identification is GNU 5.4.0
> -- The CXX compiler identification is GNU 5.4.0
> -- Check for working C compiler using: Ninja
> -- Check for working C compiler using: Ninja -- works
> -- Detecting C compiler ABI info
> -- Detecting C compiler ABI info - done
> -- Detecting C compile features
> -- Detecting C compile features - done
> -- Check for working CXX compiler using: Ninja
> -- Check for working CXX compiler using: Ninja -- works
> -- Detecting CXX compiler ABI info
> -- Detecting CXX compiler ABI info - done
> -- Detecting CXX compile features
> -- Detecting CXX compile features - done
> -- Performing Test COMPILER_SUPPORTS_CXX17
> -- Performing Test COMPILER_SUPPORTS_CXX17 - Success
> -- Performing Test COMPILER_SUPPORTS_CXX14
> -- Performing Test COMPILER_SUPPORTS_CXX14 - Success
> -- Performing Test COMPILER_SUPPORTS_CXX11
> -- Performing Test COMPILER_SUPPORTS_CXX11 - Success
> -- Performing Test COMPILER_SUPPORTS_CXX0X
> -- Performing Test COMPILER_SUPPORTS_CXX0X - Success
> -- Using C++14 compiler
> -- Looking for pthread.h
> -- Looking for pthread.h - found
> -- Looking for pthread_create
> -- Looking for pthread_create - not found
> -- Looking for pthread_create in pthreads
> -- Looking for pthread_create in pthreads - not found
> -- Looking for pthread_create in pthread
[... 124 lines cut ...]
>     linked by target "gazebo_gps_plugin" in directory /home/tony/PX4_Firmware/Tools/sitl_gazebo
>     linked by target "gazebo_usv_dynamics_plugin" in directory /home/tony/PX4_Firmware/Tools/sitl_gazebo
>     linked by target "gazebo_sonar_plugin" in directory /home/tony/PX4_Firmware/Tools/sitl_gazebo
>     linked by target "mav_msgs" in directory /home/tony/PX4_Firmware/Tools/sitl_gazebo
>     linked by target "nav_msgs" in directory /home/tony/PX4_Firmware/Tools/sitl_gazebo
>     linked by target "gazebo_lidar_plugin" in directory /home/tony/PX4_Firmware/Tools/sitl_gazebo
>     linked by target "gazebo_opticalflow_plugin" in directory /home/tony/PX4_Firmware/Tools/sitl_gazebo
>     linked by target "gazebo_mavlink_interface" in directory /home/tony/PX4_Firmware/Tools/sitl_gazebo
>     linked by target "gazebo_uuv_plugin" in directory /home/tony/PX4_Firmware/Tools/sitl_gazebo
>     linked by target "gazebo_vision_plugin" in directory /home/tony/PX4_Firmware/Tools/sitl_gazebo
>     linked by target "gazebo_multirotor_base_plugin" in directory /home/tony/PX4_Firmware/Tools/sitl_gazebo
>     linked by target "gazebo_imu_plugin" in directory /home/tony/PX4_Firmware/Tools/sitl_gazebo
>     linked by target "gazebo_gimbal_controller_plugin" in directory /home/tony/PX4_Firmware/Tools/sitl_gazebo
>     linked by target "physics_msgs" in directory /home/tony/PX4_Firmware/Tools/sitl_gazebo
>     linked by target "gazebo_motor_model" in directory /home/tony/PX4_Firmware/Tools/sitl_gazebo
> 
> -- Configuring incomplete, errors occurred!
> See also "/home/tony/PX4_Firmware/build/px4_sitl_default/build_gazebo/CMakeFiles/CMakeOutput.log".
> See also "/home/tony/PX4_Firmware/build/px4_sitl_default/build_gazebo/CMakeFiles/CMakeError.log".
> [251/740] Building CXX object platform...CMakeFiles/px4_daemon.dir/client.cpp.o
> FAILED: cd /home/tony/PX4_Firmware/build/px4_sitl_default/build_gazebo && /usr/bin/cmake -DCMAKE_INSTALL_PREFIX=/usr/local -DSEND_VISION_ESTIMATION_DATA=ON -GNinja /home/tony/PX4_Firmware/Tools/sitl_gazebo && /usr/bin/cmake -E touch /home/tony/PX4_Firmware/build/px4_sitl_default/external/Stamp/sitl_gazebo/sitl_gazebo-configure
> [251/740] Building CXX object platform...on/CMakeFiles/px4_daemon.dir/pxh.cpp.o
> ninja: build stopped: subcommand failed.
> Makefile:198: recipe for target 'px4_sitl_default' failed
> make: *** [px4_sitl_default] Error 1

bys1123, 2020-05-17:
Suggest use ubuntu 18.04 or 20.04 to build px4 1.10 and later.

liangpu597, 2020-05-17:
> Suggest use ubuntu 18.04 to build px4 1.10 and later.

～～～～ there is any way to solve it ?
I have to change from ubuntu 16.04 to 18.04?

julianoes, 2020-05-18:
@liangpu597 

```
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
GSTREAMER_LIBRARIES (ADVANCED)
```
https://github.com/PX4/Firmware/blob/b5cc3c4ef5d287b9bddebf2d86ef2ae683443df6/Tools/setup/ubuntu.sh#L189-L204

julianoes, 2020-05-27:
I'm going to close this. Please comment again if this is still a problem.

closed 2020-05-27 by julianoes