PX4/PX4-Autopilot #14920: px4vision devkit drone have few parameters seems not fine tuned

bys1123, 2020-05-17:
https://github.com/PX4/Firmware/pull/13683/files#diff-951fb79b652f79e92cd9794c2efca588R85-R97

1. MPC_MANTHR_MIN 0
Set manual throttle MPC_MANTHR_MIN to 0%? Why? Will this cause unsafety?

2. MPC_XY_VEL_P 0.079
MPC_XY_VEL_P only support %.2f

3. MPC_Z_TRAJ_P 0.3
This MPC_Z_TRAJ_P parameter is not exist now.
Related PR: #13596

4. MPC_Z_VEL_I 0.15
MPC_Z_VEL_I max value is 0.1

julianoes, 2020-05-18:
FYI @Jaeyoung-Lim @MaEtUgR

MaEtUgR, 2020-05-18:
@bys1123 Good feedback!

> 1. Set min manual throttle MPC_MANTHR_MIN to 0%? Why?

Only makes sense if `MC_AIRMODE` is set > 0 which we probably don't want for a development platform so we can set that parameter to its default 0.08 and remove it completely from the config at a later point in time. 🟡 

> 2. MPC_XY_VEL_P only support %.2f

In the UI it only shows 2 decimals but internally this value will be used with more precision and the configuration is valid. The necessity for this additional decimal with the exact same tuning goes away with the rescaling I'm proposing here: https://github.com/PX4/Firmware/pull/14823/files#diff-951fb79b652f79e92cd9794c2efca588L95 ✔️ 

> 3. This MPC_Z_TRAJ_P parameter is not exist now.

Yes, we should remove it. 🔴 

> 4. MPC_Z_VEL_I max value is 0.1

Right, I propose to increase the limit because the limit is the default at the same time so there's no margin to increase it even slightly. 🟡 

> 5. IMU_GYRO_CUTOFF set to 100, but IMU_ACCEL_CUTOFF not set.

That makes perfectly sense the accelerometer cutoff can be kept on default. The gyro one was increased for rate control tuning. ✔️ 

> 6. MC_DTERM_CUTOFF now named to IMU_DGYRO_CUTOFF

That was all correctly transitioned: https://github.com/PX4/Firmware/commit/093e9ba1cebbfeddbbc4c7df53f4ef5d27568e12#diff-951fb79b652f79e92cd9794c2efca588 ✔️

bys1123, 2020-06-03:
7. IMU_GYRO_CUTOFF 100 maybe too high
https://logs.px4.io/plot_app?log=0198afe3-7279-4cc0-a438-1479c25a05f6
the vibration is above 70 even with 4S 5200mAh battery.
Only above 58 with recommend 4S 2600mAh battery.

bys1123, 2020-06-19:
@MaEtUgR @Jaeyoung-Lim 
8. MAV_1_MODE  9 
What is 9 for? Seems not on the doc.
Should it be 2 onboard?

Jaeyoung-Lim, 2020-07-09:
@bys1123 Could you make pull requests out of these comments?

Jaeyoung-Lim, 2020-09-11:
@bys1123 Any updates?