PX4/PX4-Autopilot #14924: QGC not flashing PX4 on Cube Yellow but Identifying it as a flashable target for PX4

731avenger, 2020-05-18:
When I attempt to flash a cube yellow with QGC to flash PX4, I get the following error:
```Found device: Hex/ProfiCNC
Connected to bootloader:
Version: 5
Board ID: 120
Flash size: 1998848
Unable to find specified firmware for board type
Upgrade cancelled
------------------------------------------
If upgrade failed, make sure to connect directly to a powered USB port on your computer, not through a USB hub. Also make sure you are only powered via USB not battery.
Upgrade cancelled
------------------------------------------```

This is the case regardless of which version of QGC I use 4.0.5 or nightly. I have been told that the Yellow is PX4 compatible but have been having issues actually flashing one to PX4.

JacobCrabill, 2020-05-18:
The Yellow Cube is not supported by PX4 at present.  ArduPilot and PX4 use essentially the same bootloader, and share the same protocol so that you don't have to re-flash the bootloader in order to switch between ArduPilot and PX4.  The error above is the upload script telling you that it found a bootloader, but does not know what board it is (because PX4 does not have Cube Yellow support).

I am working with other developers on support for the Orange Cube, however; that could potentially be out in a month or so.

JacobCrabill, 2020-08-17:
The Yellow Cube is now available as the PX4 build target `cubepilot_cubeyellow`.

closed 2020-08-17 by JacobCrabill

731avenger, 2020-08-17:
@JacobCrabill cubepilot_cubeyellow is buildable but not useable unless flashing using the SWD tool.  #15513 for additional info.