PX4/PX4-Autopilot #14926: EKF2 altitude jump when GPS checks first pass in flight

dagar, 2020-05-18:
https://logs.px4.io/plot_app?log=677d9349-3e6e-4242-b253-44d1ef488568

[image]

[image]

The z reset counter is 0 the entire flight.

xdwgood, 2020-05-19:
I am interested in this question so I say a few observations:
From the log: 
1: The altitude source used by ekf has not changed during the flight (keep `control_mode_flags.CS_RNG_HGT`)
2:  Several key parameters:   `MPC_ALT_MODE ==1`   Terrain following
`EKF2_RNG_AID==1`
3:`local.ref_alt` change when reset position to gps
Looking forward to see the answers of experts :smiley:

bresch, 2020-05-20:
The vz reset counter increases properly. I think the issue could be in the handler; I'll check that.

[image]

priseborough, 2023-02-09:
This is a result of resetting the covariances for the velocity and horizontal position states and the velocity state vector (which is reset to the GPS velocity) when GPS passes checks.

closed 2023-02-09 by priseborough