PX4/PX4-Autopilot #14933: [VTOL] Dangerous Back Transition

ThomasRigi, 2020-05-19:
@RomanBapst 

**Describe the bug**
If the drone overshoots on a back transition, it initiates a FW style U turn, which already scares me in SITL. 
[image]
During this manoeuver the drone isn't capable of keeping the desired pitch, leading to a large drop in altitude, which makes the drone gain speed, which makes that the back transition doesn't finish.

Additionally, there is no support of the MC motors during the back transition. I would have expected them to [ramp up](https://docs.px4.io/master/en/advanced_config/parameter_reference.html#VT_B_TRANS_RAMP).

**To Reproduce**
It's not completely trivial to reproduce and it's not always entirely equal. Here is my setup with which I found the problem:
[bad_BT.zip](https://github.com/PX4/Firmware/files/4649422/bad_BT.zip)

Firmware  v1.11.0-beta2
-Gazebo: standard_vtol (modified sdf file to have stronger motors)
-Mission: simple mission using normal Takeoff waypoint to take off in FW mode (which is why I needed a stronger motor). Interestingly enough the back transition isn't the same when replacing the Takeoff waypoint with a VTOL takeoff waypoint.
-Parameters: mostly default, need to set CBRK_VTOLARMING = 159753 to allow arming in FW mode.

**Expected behavior**
1. The MC motors should ramp up during the back transition, independent of the take off method used.
2. (Even though it's a logical consequence of using FW L1 position control) IMO the drone shouldn't do this crazy U turn during transition. I'd be in favour of force-ending the transition when you are at the desired position even if the speed is too high.

**Log Files and Screenshots**
The first two logs are both with a FW takeoff, identical setup but the direction of the U turn is different. The third log shows the mission where FW takeoff has been replaced with VTOL Takeoff. It's interesting to see that the MC motors ramp up in this case.

https://logs.px4.io/plot_app?log=241b24bb-248a-4451-a7e0-4638565f9d34
https://logs.px4.io/plot_app?log=abe58451-984d-43e9-a8ac-31308bf0d66e

https://logs.px4.io/plot_app?log=99aa4c67-3e82-47d8-95a3-105c49280e6f

**Drone (please complete the following information):**
SITL Gazebo Standard VTOL with stronger motors

**Additional context**
Position controlled transitions have been introduced here: https://github.com/PX4/Firmware/pull/14405

RomanBapst, 2020-05-19:
@ThomasRigi The MC motors should definitely ramp up during the backtransition, so I'll have a look at that. I also thought about the backtransition, maybe we need to introduce a maximum roll angle or something.

ThomasRigi, 2020-05-20:
@RomanBapst I just found out that the MC motors neither ramp up during the backtransition in v1.10.1 when you do a FW takeoff: https://logs.px4.io/plot_app?log=6ee5578c-d2a2-4ceb-adaa-11ee584b5f85
So it must be something that was broken for quite some time, only that noone tried to use this corner case. 
For reference here a flight with v1.10.1 with a MC takeoff and separated transition command: https://logs.px4.io/plot_app?log=42fa0927-70c7-467e-9d3e-d1ce919592c4 (MC motors ramp up)
and a flight with the VTOL Takeoff item: https://logs.px4.io/plot_app?log=ac6d305d-ea48-4087-bd11-02a9eebe4d29 (MC motors ramp up)

I really don't understand why the backtransitions are treated differently depending on the takeoff which typically has happened a long time ago and doesn't influence the state of the vehicle anymore...

RomanBapst, 2020-05-27:
@ThomasRigi I had a look at this. At first I thought it's the vtol controller to blame but it looks like the issue actually stems from FlightTask. If you check the vehicle local position setpoint vz value, you can see that it's not published. Probably some initialization issue.
[image]

RomanBapst, 2020-05-27:
@ThomasRigi Could you do me a favor and test if https://github.com/PX4/Firmware/pull/14985
fixes this problem?

ThomasRigi, 2020-05-28:
@RomanBapst  Thanks for looking into it! Unfortunately your PR didn't fix it: https://logs.px4.io/plot_app?log=ae2b16f0-9ad7-474d-9c3c-e53a797485e8

Also unfortunately I missed yesterday's dev call due to a bad crash with a new version of our testing drone. :( (most likely a physical failure with the tail, but also with some PX4-related stuff I don't yet understand). Did you discuss some options to not have this bad U-turn on overshoots?

sfuhrer, 2021-01-04:
The MC motor not spinning up should now be fixed with https://github.com/PX4/PX4-Autopilot/pull/16443. 

We though still need to find a solution for the overshooting - it's not just turning around during the backtransition if if overshootes, it then even first in hover goes to the loiter radius before proceeding to the land location.