PX4/PX4-Autopilot #14943: memory overflow during make px4_sitl gazebo

bharath7kumar, 2020-05-20:
RAM Overflows and laptop hangs when I try make px4_sitl gazebo. It overflows every time while  building the file "geo_mag_declination.cpp.o". make px4_sitl jmavsim works well on the other hand. How do I go about this issue

Laptop Specs:

8gb RAM
Intel(R) Core(TM) i7-7700HQ CPU @ 2.80GHz
1 physical processor; 4 cores; 8 threads
OS: Ubuntu 18.04

Logs:

(base) bharath@bharathslaptop:~/Firmware$ make px4_sitl gazebo
-- PX4 version: v1.11.0-beta2-27-ge409ec0038
-- PX4 config file: /home/bharath/Firmware/boards/px4/sitl/default.cmake
-- PX4 config: px4_sitl_default
-- PX4 platform: posix
-- PX4 lockstep: enabled
-- cmake build type: RelWithDebInfo
-- The CXX compiler identification is GNU 7.5.0
-- The C compiler identification is GNU 7.5.0
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/cc
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Building for code coverage
-- ccache enabled (export CCACHE_DISABLE=1 to disable)
-- Found PythonInterp: /home/bharath/anaconda3/bin/python3 (found suitable version "3.6.10", minimum required is "3") 
-- build type is RelWithDebInfo
-- PX4 ECL: Very lightweight Estimation & Control Library v1.9.0-rc1-286-g24f2e60
-- Configuring done
-- Generating done
-- Build files have been written to: /home/bharath/Firmware/build/px4_sitl_default
[0/743] git submodule src/drivers/gps/devices
[1/743] git submodule src/lib/ecl
[2/743] git submodule mavlink/include/mavlink/v2.0
[5/743] git submodule Tools/sitl_gazebo
[10/743] Performing configure step for 'sitl_gazebo'
-- install-prefix: /usr/local
-- The C compiler identification is GNU 7.5.0
-- The CXX compiler identification is GNU 7.5.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Performing Test COMPILER_SUPPORTS_CXX17
-- Performing Test COMPILER_SUPPORTS_CXX17 - Success
-- Performing Test COMPILER_SUPPORTS_CXX14
-- Performing Test COMPILER_SUPPORTS_CXX14 - Success
-- Performing Test COMPILER_SUPPORTS_CXX11
-- Performing Test COMPILER_SUPPORTS_CXX11 - Success
-- Performing Test COMPILER_SUPPORTS_CXX0X
-- Performing Test COMPILER_SUPPORTS_CXX0X - Success
[... 123 lines cut ...]

-- Generating done
-- Build files have been written to: /home/bharath/Firmware/build/px4_sitl_default/build_gazebo
[739/743] Performing build step for 'sitl_gazebo'
[8/100] Generating /home/bharath/Firmw...s/sitl_gazebo/models/c920/c920-gen.sdf
/home/bharath/Firmware/Tools/sitl_gazebo/models/c920/c920.sdf.jinja -> /home/bharath/Firmware/Tools/sitl_gazebo/models/c920/c920-gen.sdf
[9/100] Generating /home/bharath/Firmw...models/3DR_gps_mag/3DR_gps_mag-gen.sdf
/home/bharath/Firmware/Tools/sitl_gazebo/models/3DR_gps_mag/3DR_gps_mag.sdf.jinja -> /home/bharath/Firmware/Tools/sitl_gazebo/models/3DR_gps_mag/3DR_gps_mag-gen.sdf
[10/100] Generating /home/bharath/Firm...models/matrice_100/matrice_100-gen.sdf
/home/bharath/Firmware/Tools/sitl_gazebo/models/matrice_100/matrice_100.sdf.jinja -> /home/bharath/Firmware/Tools/sitl_gazebo/models/matrice_100/matrice_100-gen.sdf
[12/100] Generating /home/bharath/Firm...ls/mb1240-xl-ez4/mb1240-xl-ez4-gen.sdf
/home/bharath/Firmware/Tools/sitl_gazebo/models/mb1240-xl-ez4/mb1240-xl-ez4.sdf.jinja -> /home/bharath/Firmware/Tools/sitl_gazebo/models/mb1240-xl-ez4/mb1240-xl-ez4-gen.sdf
[13/100] Generating /home/bharath/Firm..._gazebo/models/px4flow/px4flow-gen.sdf
/home/bharath/Firmware/Tools/sitl_gazebo/models/px4flow/px4flow.sdf.jinja -> /home/bharath/Firmware/Tools/sitl_gazebo/models/px4flow/px4flow-gen.sdf
[14/100] Generating /home/bharath/Firm..._gazebo/models/pixhawk/pixhawk-gen.sdf
/home/bharath/Firmware/Tools/sitl_gazebo/models/pixhawk/pixhawk.sdf.jinja -> /home/bharath/Firmware/Tools/sitl_gazebo/models/pixhawk/pixhawk-gen.sdf
[18/100] Generating /home/bharath/Firm...s/sitl_gazebo/models/r200/r200-gen.sdf
/home/bharath/Firmware/Tools/sitl_gazebo/models/r200/r200.sdf.jinja -> /home/bharath/Firmware/Tools/sitl_gazebo/models/r200/r200-gen.sdf
[21/100] Generating /home/bharath/Firm...sitl_gazebo/models/sf10a/sf10a-gen.sdf
/home/bharath/Firmware/Tools/sitl_gazebo/models/sf10a/sf10a.sdf.jinja -> /home/bharath/Firmware/Tools/sitl_gazebo/models/sf10a/sf10a-gen.sdf
[41/100] Generating /home/bharath/Firm...Tools/sitl_gazebo/models/iris/iris.sdf

[41/100] Building CXX object CMakeFile...ugin.dir/src/geo_mag_declination.cpp.o

I interrupted at this point to get the logs just before it hangs

julianoes, 2020-05-20:
Generally, this is a RAM issue. So you should be able to work around it by having more RAM available.

Alternatively, you could try to build with `-j2` instead of e.g. `-j8`, however, I need to find out how that argument can be passed to the Gazebo build.

bharath7kumar, 2020-05-20:
I was able to build with the same laptop a few months back. But it is not working now. I tried make px4_sitl gazebo -j1, it still used all the eight threads.

julianoes, 2020-05-20:
@dagar how can we pass the `-jX` argument to the sitl_gazebo build?

dagar, 2020-05-20:
> @dagar how can we pass the `-jX` argument to the sitl_gazebo build?

Passed through here somehow. https://github.com/PX4/Firmware/blob/6c16a29d26b2c79d63f1459be361c8972399d9ce/platforms/posix/cmake/sitl_target.cmake#L34-L47

BazookaJoe1900, 2020-05-21:
The high needs of RAM when compiling the SITL happens to me too. In my case I am working with VM that if I don't give the VM more then 12GB of RAM its start to use cache.
Also the duration of building the SITL can be few long minutes.
is that normal? is there might be some faulty build argument that can be changed?
I tried to find one but couldn't.

julianoes, 2020-05-25:
> Passed through here somehow. 

@dagar that's the part I'm confused about :smile:

julianoes, 2020-05-25:
I'm generally confused about the parallel build. I've tried going to the basics and removed `ninja` and disabled `ccache` and just `make px4_sitl -j1` already used 2-4 cores building :confused:.

julianoes, 2020-05-25:
Ok, I have a fix coming up that actually allows to configure how parallel the build should happen.

julianoes, 2020-06-04:
With #14971 you can now build with fewer cores in order to avoid running out of RAM:

```
make px4_sitl gazebo -2
```

closed 2020-06-04 by julianoes

shu-bamma, 2020-10-18:
> With #14971 you can now build with fewer cores in order to avoid running out of RAM:
> 
> ```
> make px4_sitl gazebo -2
> ```

same issue...dosen't help my case.

cdbharath, 2020-10-18:
> > With #14971 you can now build with fewer cores in order to avoid running out of RAM:
> > ```
> > make px4_sitl gazebo -2
> > ```
> 
> same issue...dosen't help my case.

pull the commit that is mentioned in this thread. It worked for my case.

shu-bamma, 2020-10-18:
> > > With #14971 you can now build with fewer cores in order to avoid running out of RAM:
> > > ```
> > > make px4_sitl gazebo -2
> > > ```
> > 
> > 
> > same issue...dosen't help my case.
> 
> pull the commit that is mentioned in this thread. It worked for my case.

@cdbharath  thanks for the response..i am pretty new to this stuff so bear with me.
you mean the latest one right??
Makefile: fix j for clang-tidy calls

dagar, 2020-10-18:
@shu-bamma from a clean state (make distclean) can you build and capture the full output with the error message?

shu-bamma, 2020-10-18:
> @shu-bamma from a clean state (make distclean) can you build and capture the full output with the error message?

> RAM Overflows and laptop hangs when I try make px4_sitl gazebo. It overflows every time while building the file "gazebo_wind_plugin.cpp.o". make px4_sitl jmavsim works well on the other hand. How do I go about this issue
> 
> Laptop Specs:
> 
> 8gb RAM
> Intel(R) Core(TM) i7-7700HQ CPU @ 2.80GHz
> 1 physical processor; 4 cores; 8 threads
> OS: Ubuntu 18.04
> 
> Logs:
> 
> ```
> bharath@bharathslaptop:~/Firmware$ make px4_sitl gazebo
> -- PX4 version: v1.11.0-beta2-27-ge409ec0038
> -- PX4 config file: /home/bharath/Firmware/boards/px4/sitl/default.cmake
> -- PX4 config: px4_sitl_default
> -- PX4 platform: posix
> -- PX4 lockstep: enabled
> -- cmake build type: RelWithDebInfo
> -- The CXX compiler identification is GNU 7.5.0
> -- The C compiler identification is GNU 7.5.0
> -- The ASM compiler identification is GNU
> -- Found assembler: /usr/bin/cc
> -- Check for working CXX compiler: /usr/bin/c++
> -- Check for working CXX compiler: /usr/bin/c++ -- works
> -- Detecting CXX compiler ABI info
> -- Detecting CXX compiler ABI info - done
> -- Detecting CXX compile features
> -- Detecting CXX compile features - done
> -- Check for working C compiler: /usr/bin/cc
> -- Check for working C compiler: /usr/bin/cc -- works
> -- Detecting C compiler ABI info
> -- Detecting C compiler ABI info - done
> -- Detecting C compile features
> -- Detecting C compile features - done
> -- Building for code coverage
> -- ccache enabled (export CCACHE_DISABLE=1 to disable)
> -- Found PythonInterp: /usr/bin/python3 (found suitable version "3.6.9", minimum required is "3") 
> -- build type is RelWithDebInfo
> -- PX4 ECL: Very lightweight Estimation & Control Library v1.9.0-rc1-286-g24f2e60
> -- Configuring done
> -- Generating done
> -- Build files have been written to: /home/bharath/Firmware/build/px4_sitl_default
> [0/743] git submodule src/drivers/gps/devices
> [1/743] git submodule src/lib/ecl
> [2/743] git submodule mavlink/include/mavlink/v2.0
> [5/743] git submodule Tools/sitl_gazebo
> [12/743] Performing configure step for 'sitl_gazebo'
> -- install-prefix: /usr/local
> -- The C compiler identification is GNU 7.5.0
> -- The CXX compiler identification is GNU 7.5.0
> -- Check for working C compiler: /usr/bin/cc
> -- Check for working C compiler: /usr/bin/cc -- works
> -- Detecting C compiler ABI info
> -- Detecting C compiler ABI info - done
> -- Detecting C compile features
> -- Detecting C compile features - done
> -- Check for working CXX compiler: /usr/bin/c++
> -- Check for working CXX compiler: /usr/bin/c++ -- works
> -- Detecting CXX compiler ABI info
> -- Detecting CXX compiler ABI info - done
> -- Detecting CXX compile features
> -- Detecting CXX compile features - done
> -- Performing Test COMPILER_SUPPORTS_CXX17
> -- Performing Test COMPILER_SUPPORTS_CXX17 - Success
> -- Performing Test COMPILER_SUPPORTS_CXX14
> -- Performing Test COMPILER_SUPPORTS_CXX14 - Success
> -- Performing Test COMPILER_SUPPORTS_CXX11
[... 122 lines cut ...]
> [12/100] Generating /home/bharath/Firm...sitl_gazebo/models/sf10a/sf10a-gen.sdf
> /home/bharath/Firmware/Tools/sitl_gazebo/models/sf10a/sf10a.sdf.jinja -> /home/bharath/Firmware/Tools/sitl_gazebo/models/sf10a/sf10a-gen.sdf
> [13/100] Generating /home/bharath/Firm...s/sitl_gazebo/models/c920/c920-gen.sdf
> /home/bharath/Firmware/Tools/sitl_gazebo/models/c920/c920.sdf.jinja -> /home/bharath/Firmware/Tools/sitl_gazebo/models/c920/c920-gen.sdf
> [14/100] Generating /home/bharath/Firm...ls/mb1240-xl-ez4/mb1240-xl-ez4-gen.sdf
> /home/bharath/Firmware/Tools/sitl_gazebo/models/mb1240-xl-ez4/mb1240-xl-ez4.sdf.jinja -> /home/bharath/Firmware/Tools/sitl_gazebo/models/mb1240-xl-ez4/mb1240-xl-ez4-gen.sdf
> [15/100] Generating /home/bharath/Firm...models/3DR_gps_mag/3DR_gps_mag-gen.sdf
> /home/bharath/Firmware/Tools/sitl_gazebo/models/3DR_gps_mag/3DR_gps_mag.sdf.jinja -> /home/bharath/Firmware/Tools/sitl_gazebo/models/3DR_gps_mag/3DR_gps_mag-gen.sdf
> [16/100] Generating /home/bharath/Firm..._gazebo/models/px4flow/px4flow-gen.sdf
> /home/bharath/Firmware/Tools/sitl_gazebo/models/px4flow/px4flow.sdf.jinja -> /home/bharath/Firmware/Tools/sitl_gazebo/models/px4flow/px4flow-gen.sdf
> [17/100] Generating /home/bharath/Firm..._gazebo/models/pixhawk/pixhawk-gen.sdf
> /home/bharath/Firmware/Tools/sitl_gazebo/models/pixhawk/pixhawk.sdf.jinja -> /home/bharath/Firmware/Tools/sitl_gazebo/models/pixhawk/pixhawk-gen.sdf
> [18/100] Generating /home/bharath/Firm...models/matrice_100/matrice_100-gen.sdf
> /home/bharath/Firmware/Tools/sitl_gazebo/models/matrice_100/matrice_100.sdf.jinja -> /home/bharath/Firmware/Tools/sitl_gazebo/models/matrice_100/matrice_100-gen.sdf
> [28/100] Generating /home/bharath/Firm...s/sitl_gazebo/models/r200/r200-gen.sdf
> /home/bharath/Firmware/Tools/sitl_gazebo/models/r200/r200.sdf.jinja -> /home/bharath/Firmware/Tools/sitl_gazebo/models/r200/r200-gen.sdf
> [37/100] Generating /home/bharath/Firm...Tools/sitl_gazebo/models/iris/iris.sdf
> 
> [37/100] Building CXX object CMakeFile...lugin.dir/src/gazebo_wind_plugin.cpp.o
> ```
> 
> I interrupted at this point to get the logs just before it hangs

@dagar this is the EXACT error and machine configurations i have , then in #14971 @julianoes claimed to have fixed it but you pointed out about the CI failing and from then on i did not follow your chats. In this fix he has made a total of 4 commits, which one should i pull was my doubt (if any of them actually fix the problem that is)
Thanks

julianoes, 2021-04-21:
I only see this now. @shu-bamma you're right. It does not actually work for the Gazebo case.

julianoes, 2021-04-21:
It is now using the number of cores on your system, see: https://github.com/PX4/PX4-Autopilot/commit/52bb99688f4c828aa95e5e337269880cfca94f50

offjangir, 2021-06-06:
The build is not over yet, let the machine run for a while I have good harware specs it took about  40-45 mins after it hangs . make worked without specifying cores. Sit back and let it process.