PX4/PX4-Autopilot #14945: Mixer corner case with too low collective thrust and no airmode?

MaEtUgR, 2020-05-20:
I had airmode disabled (`MC_AIRMODE` set to 0) and minimum thrust in stabilized on zero (`MPC_MANTHR_MIN` set to 0).

**Expected behavior**
The vehicle should fly normally as long as the collective thrust is high enough and have loose rate and attitude tracking temporarily when the thrust is too low to have full torque authority because two motors cannot ramp any lower.

**Describe the bug**
Instead it lost attitude tracking completely not even compensating with the two motors that were still in actuation range and never regained any authority even though I increased throttle to more than hover thrust. It even commanded the motors to stop completely.

**To Reproduce**
Steps to reproduce the behavior:
1. Flash master
2. Set `MC_AIRMODE` to 0 (default)
3. Set `MPC_MANTHR_MIN` to 0
4. Hover in stabilized mode
5. Start pushing roll back and forth
6. Lower thrust all the way to zero just very quickly and pull it up immediately again
7. See drone flipping multiple times and crashing

**Log Files and Screenshots**
https://logs.px4.io/plot_app?log=60177bc8-95ce-481f-9454-79ef03267a7c

[image]

**Drone (please complete the following information):**
- Holybro QAV250
- I was testing https://github.com/PX4/Firmware/pull/14896, https://github.com/PX4/Firmware/pull/14823, https://github.com/PX4/Firmware/pull/14738, https://github.com/PX4/Firmware/pull/14551 based on 793eb821531191409aba3034606d6fad12d3be3a and am very convinced the problem is unrelated to the tested prs.

**Initial investigation**
My initial investigation brought me to the mixer since the rate and attitude setpoints and reactions looked very plausible but the `actuator_control` goes to -3.75 which I'm not completely sure if it's in the expected range and the actuator outputs first all go to idle and then completely to zero commanding the motors to stop without disarm, kill, land detection or failure detection.

FYI @bresch @bkueng

MaEtUgR, 2020-05-25:
In the meantime, I found out that the complete divergence must have been caused by the failure detector triggering and I overlooked it in the first pass because the status changes only for one quick sample:
[image]

MaEtUgR, 2020-05-27:
I did some SITL testing in stabilized and acro and indeed the mixer keeps all motors idle if you give zero thrust. It doesn't allow the two motors that are not saturated to compensate anything.
[image]

bresch, 2020-05-27:
@MaEtUgR thanks for the tests, I'll have a look

bresch, 2020-06-12:
@MaEtUgR After thinking about this issue, I think this isn't a but but was the correct behavior: a minimum thrust is required to operate without airmode. No airmode is a "safe" mode where the divergence of the attitude rate controller doesn't generate a dangerous behavior as long as the throttle is low: keeping the stick low can prevent a flip on ground.
This is also why the description of [MPC_MANTHR_MIN](https://dev.px4.io/master/en/advanced/parameter_reference.html#MPC_MANTHR_MIN) says that the value should only be set to zero if airmode is enabled.

closed 2026-08-12 by github-actions[bot]