PX4/PX4-Autopilot #14953: Typhoon h480 in SITL has a lot of vibrations in flight

Jaeyoung-Lim, 2020-05-21:
**Describe the bug**
The typhoon h480 model in SITL has been showing a lot of vibrations in flight lately.

**To Reproduce**
```
make px4_sitl gazebo_typhoon_h480
```

**Log Files and Screenshots**
[image]
Log: https://review.px4.io/plot_app?log=12f06674-3498-4a09-81f6-29b946a0b708

dagar, 2020-05-21:
What's it like with the gazebo rate increased?

When did this last work?

Jaeyoung-Lim, 2020-05-21:
@dagar I think it was like this for quite some time...

My initial suspicion is that the landing gear joint position control has something to do with the oscillations. Disabling the gimbal joint controllers certainly improve the situation, but doesn't solve it completely

Jaeyoung-Lim, 2020-08-23:
This is still the case

canberkdurmus, 2021-01-16:
We can close this issue since it is been resolved in #16519

closed 2021-01-16 by Jaeyoung-Lim