PX4/PX4-Autopilot #14954: Suggest: Add details about how to using hand-launch takeoff mode in the user guide.

guizilaile23, 2020-05-21:
First, The RWTO_TKOFF was set to disable.
then, after I turn the mode to takeoff, the motor will not spin, and landing was detected after 2s.

so I change the RWTO_TKOFF=1, and after I turn the mode to takeoff, the motor works.
because my FW does not have any landing gear, so I throw my FW, just like the hand-launch method, the FW works.  

so, how to use hand-launch takeoff mode correctly ?

And after 4 hours searching on Google and PX4 guide, I found nothing. 
I still don't know why hand-launch not work, and how to disable landing detector. 
And I'm not dare to ask questions, because it's may makes me look like a fool.

In the introduction of takeoff mode ("https://docs.px4.io/master/en/flight_modes/takeoff.html")
nothing details information about how to use hand-launch takeoff mode exist.

hope you could give the details about how to use hand-launch takeoff mode.
i.e.  step1. armed the vehicle,
       step2. turn the mode to TAKEOFF ( at this time, the motor should spin.......)
       step3. throw the vehicle in 5s after turned the mode.

julianoes, 2020-05-25:
FYI @hamishwillee

hamishwillee, 2020-05-26:
@guizilaile23 I hate looking like a fool too. But the fact is, that you don't, this is not documented.

@RomanBapst @Antiheavy Can either of you comment on how hand-launch works on PX4? I have not done it. All we have in [takeoff mode docs](https://docs.px4.io/master/en/flight_modes/takeoff.html) is:

> In catapult/hand launch mode the vehicle will perform a full throttle climbout (ramp up to RWTO_MAX_THR in about 2 seconds). Once the altitude error < FW_CLMBOUT_DIFF, regular navigation will proceed.

My guess is that you arm the vehicle, face into the wind (?), throw it on a 30 degree angle (?), and thta if armed the vehicle will detect the acceleration is over a certain amount and immediately kick in the throttle. 
- Is any of that close? If not, what do you do.
- Does arming still auto-disarm on fixed-wing after 10 seconds
- Is there a size/weight limit that is max recommended for hand launch?
- Any other considerations, settings you particularly need for this?

Further, what about catapult launch? How does that differ, if at all. Any other considerations? Any catapults we "recommend"?

Antiheavy, 2020-05-26:
There are multiple ways to do an automated hand launched takeoff with PX4:

**1) Use Mission Takeoff waypoint:** this is what we do on our hand launched fixed wing and I think this is how most people do it.  You must have a Takeoff waypoint at the start of the mission (or at least it needs to be the active waypoint when the mission starts.  Arm the vehicle while in Mission mode AND the Takeoff waypoint is active.  Once flight is detected (not landed) after being thrown the vehicle climbs out (using the minimum pitch parameter in the the Takeoff waypoint (terrible way to do it BTW! see note below) until it reaches < FW_CLMBOUT_DIFF (also a terrible way to do it! should be a height above launch param, not height below waypoint param, but oh well...)

**2) Use Takeoff flight mode:**  When the vehicle is armed in Takeoff flight mode, it waits to detect flight (not landing) then the vehicle will climb to it's minimum loiter altitude and then hold/loiter in place when it reaches that altitude. I haven't personally tested this in a long time so I'm not sure if it is working properly anymore.  I think QGC might actually block this flight mode button for fixed wing now (maybe for good reason? see note below).
_Note: I think there is still a hard-coded 10 degrees takeoff pitch for Takeoff mode which may not be appropriate for many vehicles.  I know there is a stale PR or three hanging around that tried to add a parameter to control takeoff pitch: FW_TKO_PITCH_MIN.  The PR we rolled into our firmware was based on this: https://github.com/PX4/Firmware/pull/9243  Not sure why it was never merged upstream..._

**3) Use Launch detection with either of the above options:**  we use this to start the motor before throwing.  We pump/shake the aircraft forward to trigger launch detection, then when the motor has reached full throttle (1-2 seconds) we throw the aircraft using the full thrust of the propeller to assist immediate climb out.  Works MUCH better than letting the throttle ramp up after the vehicle has been thrown.

guizilaile23, 2020-05-26:
Thanks, everyone, I found the way to use hand-launch takeoff in doc of ArduPilot,(I'm still using PX4 now) and now I understand the reason why my motor not spinning. I think the procedure is basically the same. I will try it this weekend. thanks again.
by the way, the hand-launch takeoff introduction of  ArduPilot is worth learning from, I think. (https://ardupilot.org/plane/docs/takeoff-landingpage.html).

hamishwillee, 2020-05-27:
OK discussion to continue in https://github.com/PX4/px4_user_guide/pull/744

hamishwillee, 2020-08-30:
I think this is covered now: 
- http://docs.px4.io/master/en/flying/fixed_wing_takeoff.html#catapulthand-launch-mode
- http://docs.px4.io/master/en/flight_modes/takeoff.html#fixed_wing

closed 2020-08-30 by hamishwillee