PX4/PX4-Autopilot #14961: drivers/platform/Kconfig' not found error.

heshananupama, 2020-05-23:
[image]
Every time I try to build a firmware, it just end up with this error. I followed the web while building the firmware but it complains me about this. What would be the reason?

dagar, 2020-05-23:
Which version of PX4 is this?

Try
``` Bash
make distclean
make px4_fmu-v4_default
```

heshananupama, 2020-05-23:
its PX4 version: v1.11.0-beta2-38-g951969ba00 It didn't fix. Still the build just stop stating the same error :(

dagar, 2020-05-23:
> its PX4 version: v1.11.0-beta2-38-g951969ba00 It didn't fix. Still the build just stop stating the same error :(

Can you check the toolchain installation? Is that the default terminal and shell or something different? From the firmware checkout you can run `./Tools/setup/OSX.sh`.

heshananupama, 2020-05-23:
Its my mac terminal. After I run the shell script you said following is the output. But still the error is there. 
[image]

wanderzell, 2020-05-23:
I have same error. While trying to build a firmware for Kakute F7

Is there any places where one could just go and download the Firmware to flash and use? I spent entire evening on installing "developers toolchain". Yet my only intention was to install the firmware and try how it flies on a racer.

wanderzell, 2020-05-24:
Ignore my last comment - found firmware downloads from https://github.com/PX4/Firmware/releases

Look for small link "Assets" where it has more than 2

Would help if all releases have builds for all targets. Will save people from wasted effort setting up development environment that they will never really use.

dagar, 2020-05-24:
@heshananupama let's see if we can capture more targeted error output to find the problem. I can tell you it isn't a problem in general as we have automated Mac builds. http://ci.px4.io:8080/blue/organizations/jenkins/PX4_misc%2FFirmware-compile_mac/detail/master/1917/pipeline

``` Console
make distclean
V=1 make px4_fmu-v4_default
```

heshananupama, 2020-05-24:
@wanderzell where did you find the firmware? I downloaded the source from assets but still could not figure out the place of the firmware

heshananupama, 2020-05-24:
@dagar Do we need to export some environment variables before running the make command? 
Following are the entire output that I am getting
`(base) Heshans-MacBook-Pro:Firmware heshananu$ V=1 make px4_fmu-v4_default
-- PX4 version: v1.11.0-beta2-38-g951969ba00
-- PX4 config file: /Users/heshananu/Desktop/hes/Firmware/boards/px4/fmu-v4/default.cmake
-- PX4 config: px4_fmu-v4_default
-- PX4 platform: nuttx
-- cmake build type: MinSizeRel
-- The CXX compiler identification is GNU 8.2.1
-- The C compiler identification is GNU 8.2.1
-- The ASM compiler identification is GNU
-- Found assembler: /usr/local/bin/arm-none-eabi-gcc
-- Check for working CXX compiler: /usr/local/bin/arm-none-eabi-g++
-- Check for working CXX compiler: /usr/local/bin/arm-none-eabi-g++ - works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /usr/local/bin/arm-none-eabi-gcc
-- Check for working C compiler: /usr/local/bin/arm-none-eabi-gcc - works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Found PythonInterp: /Library/Frameworks/Python.framework/Versions/3.8/bin/python3 (found suitable version "3.8.1", minimum required is "3") 
-- build type is MinSizeRel
-- PX4 ECL: Very lightweight Estimation & Control Library v1.9.0-rc1-295-g600240d
-- Found PythonInterp: /Library/Frameworks/Python.framework/Versions/3.8/bin/python3 (found version "3.8.1") 
-- Using C++11
-- Release build type: MinSizeRel
-- NuttX verbose build enabled
-- ROMFS: px4fmu_common
-- ROMFS: Adding rc.board_defaults
-- ROMFS: Adding rc.board_sensors
-- ROMFS: Adding rc.board_extras
-- ROMFS: Adding rc.board_mavlink
-- Found SVD: 
-- Configuring done
-- Generating done
-- Build files have been written to: /Users/heshananu/Desktop/hes/Firmware/build/px4_fmu-v4_default
Scanning dependencies of target git_nuttx
Scanning dependencies of target git_nuttx_apps
Scanning dependencies of target px4_config_file_target
Scanning dependencies of target uorb_headers
[  0%] git submodule platforms/nuttx/NuttX/nuttx
[  0%] git submodule platforms/nuttx/NuttX/apps
[  0%] Built target px4_config_file_target
[  0%] Generating uORB topic headers
Scanning dependencies of target ver_gen
[  0%] Generating git version header
[  0%] Built target git_nuttx_apps
[  0%] Built target git_nuttx
Scanning dependencies of target mixer_gen
Scanning dependencies of target mixer_gen_6dof
[  0%] Generating mixer_multirotor.generated.h
[  0%] Generating mixer_multirotor_6dof.generated.h
[  0%] Built target mixer_gen_6dof
[  0%] Generating mixer_multirotor_normalized.generated.h
Scanning dependencies of target git_gps_devices
[  0%] git submodule src/drivers/gps/devices
[  0%] Built target git_gps_devices
Scanning dependencies of target px4_uavcan_dsdlc
[  0%] Built target mixer_gen
[  0%] PX4 UAVCAN dsdl compiler
Scanning dependencies of target git_uavcan
[  0%] git submodule src/drivers/uavcan/libuavcan
[  0%] Built target git_uavcan
Scanning dependencies of target libuavcan_dsdlc
[  0%] Running dsdl compiler
[  0%] Built target px4_uavcan_dsdlc
[  0%] Built target ver_gen
Scanning dependencies of target git_ecl
Scanning dependencies of target git_mavlink_v2
[  1%] git submodule src/lib/ecl
[  1%] git submodule mavlink/include/mavlink/v2.0
[  1%] Built target git_ecl
[  1%] Built target git_mavlink_v2
Scanning dependencies of target airframes_xml
Scanning dependencies of target parameters_xml
[  1%] Creating airframes.xml
[  1%] Generating serial_params.c
[  1%] Built target libuavcan_dsdlc
[  1%] Built target airframes_xml
[  1%] Generating parameters.xml
[  1%] Built target uorb_headers
Scanning dependencies of target nuttx_copy_and_apps_target
[  1%] Built target parameters_xml
[  1%] Built target nuttx_copy_and_apps_target
Scanning dependencies of target nuttx_config_kconfig_target
[  2%] Built target nuttx_config_kconfig_target
Scanning dependencies of target nuttx_copy_config_dir_target
[  3%] Built target nuttx_copy_config_dir_target
Scanning dependencies of target nuttx_config_target
[  3%] Copying NuttX compressed config nsh and inflating (make olddefconfig)
drivers/Kconfig:433: 'drivers/platform/Kconfig' not found (in 'source drivers/platform/Kconfig'). Check that environment variables are set correctly (e.g. $srctree, which is unset or blank). Also note that unset environment variables expand to the empty string.
make[4]: *** [do_olddefconfig] Error 1
make[3]: *** [NuttX/nuttx/.config] Error 2
make[3]: *** Deleting file `NuttX/nuttx/.config'
make[2]: *** [NuttX/CMakeFiles/nuttx_config_target.dir/all] Error 2
make[1]: *** [all] Error 2
make: *** [px4_fmu-v4_default] Error 2
(base) Heshans-MacBook-Pro:Firmware heshananu$ 
`

wanderzell, 2020-05-24:
> @wanderzell where did you find the firmware file? I downloaded the source from assets but still could not figure out the place of the firmware

I ended up grabbing a build from v1.11.0 Beta 1 release (not the latest)

Latest release has only source code but If you scroll down a bit then you see that some releases have more assets attached - code and ready built firmwares.

dagar, 2020-05-24:
> @dagar Do we need to export some environment variables before running the make command?
> Following are the entire output that I am getting

No you don't need to, but it's possible you have something already set that's breaking it. 

Your python path is a bit surprising (`/Library/Frameworks/Python.framework/Versions/3.8/bin/python3`). We need to try and capture real error output from the failed `make olddefconfig` NuttX build step.

After the failed build is there a `nuttx_olddefconfig.log` in the build directory? 
``` Console
% make px4_fmu-v4_default
-- PX4 version: v1.11.0-beta2-38-g81c4429596
-- PX4 config file: /Users/dagar/git/Firmware/boards/px4/fmu-v4/default.cmake
-- PX4 config: px4_fmu-v4_default
-- PX4 platform: nuttx
-- cmake build type: MinSizeRel
-- The CXX compiler identification is GNU 7.3.1
-- The C compiler identification is GNU 7.3.1
-- The ASM compiler identification is GNU
-- Found assembler: /usr/local/bin/arm-none-eabi-gcc
-- Check for working CXX compiler: /usr/local/bin/arm-none-eabi-g++
-- Check for working CXX compiler: /usr/local/bin/arm-none-eabi-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /usr/local/bin/arm-none-eabi-gcc
-- Check for working C compiler: /usr/local/bin/arm-none-eabi-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- ccache enabled (export CCACHE_DISABLE=1 to disable)
-- Found PythonInterp: /usr/local/bin/python3 (found suitable version "3.7.6", minimum required is "3")
-- build type is MinSizeRel
-- PX4 ECL: Very lightweight Estimation & Control Library v1.9.0-rc1-295-g600240d
-- Found PythonInterp: /usr/local/bin/python3 (found version "3.7.6")
-- Using C++11
-- Release build type: MinSizeRel
-- ROMFS: px4fmu_common
-- ROMFS: Adding rc.board_defaults
-- ROMFS: Adding rc.board_sensors
-- ROMFS: Adding rc.board_extras
-- ROMFS: Adding rc.board_mavlink
-- Found SVD: /Users/dagar/git/Firmware/../cmsis-svd/data/STMicro/STM32F427.svd
-- Configuring done
-- Generating done
-- Build files have been written to: /Users/dagar/git/Firmware/build/px4_fmu-v4_default
[0/1335] git submodule platforms/nuttx/NuttX/nuttx
[1/1335] git submodule src/lib/ecl
[3/1335] git submodule src/drivers/gps/devices
[8/1335] git submodule mavlink/include/mavlink/v2.0
[9/1335] git submodule src/drivers/uavcan/libuavcan
[13/1335] git submodule platforms/nuttx/NuttX/apps
[1333/1335] Linking CXX executable px4_fmu-v4_default.elf
Memory region         Used Size  Region Size  %age Used
           flash:     1723149 B      2032 KB     82.81%
            sram:       32284 B       192 KB     16.42%
          ccsram:          0 GB        64 KB      0.00%
[1335/1335] Creating /Users/dagar/git/Firmware/build/px4_fmu-v4_default/px4_fmu-v4_default.px4

% find . -name nuttx_olddefconfig.log
./build/px4_fmu-v4_default/NuttX/nuttx/nuttx_olddefconfig.log
```

We might need to add some debugging here to see what's going on. https://github.com/PX4/Firmware/blob/1c0925a189990f29f21c8c3eb2bdd594d52cb984/platforms/nuttx/NuttX/tools/kconfig-conf#L16

heshananupama, 2020-05-25:
@dagar Following is the output. So I had format my Laptop and reinstalled everything. Still the error is same. 
`heshanperera@Heshans-MacBook-Pro build % cat ./px4_fmu-v4_default/NuttX/nuttx/nuttx_olddefconfig.log
Create .version
LN: arch/arm/src/board to /Users/heshanperera/Desktop/px4/Firmware/build/px4_fmu-v4_default/NuttX/nuttx/../nuttx-config//src
LN: include/arch to arch/arm/include
LN: arch/arm/src/chip to arch/arm/src/stm32
LN: /Users/heshanperera/Desktop/px4/Firmware/build/px4_fmu-v4_default/NuttX/nuttx/drivers/platform to /Users/heshanperera/Desktop/px4/Firmware/build/px4_fmu-v4_default/NuttX/nuttx/../nuttx-config/drivers
LN: include/arch/board to /Users/heshanperera/Desktop/px4/Firmware/build/px4_fmu-v4_default/NuttX/nuttx/../nuttx-config//include
LN: include/arch/chip to arch/arm/include/stm32
LN: platform/board to /Users/heshanperera/Desktop/px4/Firmware/build/px4_fmu-v4_default/NuttX/apps/platform/dummy
DEBUG: kconfiglib kconfig-conf wrapper, arguments: --olddefconfig Kconfig
heshanperera@Heshans-MacBook-Pro build % 
`
The debugging line is already there in my kcongif-conf file

heshananupama, 2020-05-27:
@wanderzell Thank you, I found the files under assets. However the file has a .px4 extension right? How would we create the .hex file or the .bin file that could be uploaded to Microcontroller.

pepjo, 2020-06-07:
Hi,

I am running linux (manjaro) and I am having exactly the same problem when trying to build the kakute F7 firmware. I was trying to build from master, commit: `af5c48b8699b8ef1ee274e0268678fb0dd7db212`

In particular, I get:
```
[...]
Scanning dependencies of target nuttx_config_kconfig_target
[  1%] Built target nuttx_config_kconfig_target
Scanning dependencies of target nuttx_copy_config_dir_target
[  2%] Built target nuttx_copy_config_dir_target
Scanning dependencies of target nuttx_config_target
[  2%] Copying NuttX compressed config nsh and inflating (make olddefconfig)
drivers/Kconfig:433: 'drivers/platform/Kconfig' not found (in 'source drivers/platform/Kconfig'). Check that environment variables are set correctly (e.g. $srctree, which is unset or blank). Also note that unset environment variables expand
 to the empty string.
make[4]: *** [tools/Makefile.unix:568: do_olddefconfig] Error 1
make[3]: *** [NuttX/CMakeFiles/nuttx_config_target.dir/build.make:86: NuttX/nuttx/.config] Error 2
make[3]: *** Deleting file 'NuttX/nuttx/.config'
make[2]: *** [CMakeFiles/Makefile2:11936: NuttX/CMakeFiles/nuttx_config_target.dir/all] Error 2
make[1]: *** [Makefile:126: all] Error 2
make: *** [Makefile:215: holybro_kakutef7_default] Error 2
```