PX4/PX4-Autopilot #14962: Could not determine chip architecture from NuttX config.  You may have to   add it.

psp316r, 2020-05-23:
When I am Trying to build using command  **make px4_fmu-v2** I am getting the following Error:

[1/1043] Copying NuttX compressed conf... nsh and inflating (make olddefconfig)
FAILED: NuttX/nuttx/.config NuttX/nuttx_olddefconfig.stamp 
cd "/home/admin1/Desktop/Testing 2/Firmware/build/px4_fmu-v5_default/NuttX/nuttx" && /usr/bin/cmake -E copy_if_different /home/admin1/Desktop/Testing\ 2/Firmware/boards/px4/fmu-v5/nuttx-config/nsh/defconfig /home/admin1/Desktop/Testing\ 2/Firmware/build/px4_fmu-v5_default/NuttX/nuttx/.config && "/home/admin1/Desktop/Testing 2/Firmware/platforms/nuttx/NuttX/tools/px4_nuttx_make_olddefconfig.sh" > nuttx_olddefconfig.log && /usr/bin/cmake -E touch /home/admin1/Desktop/Testing\ 2/Firmware/build/px4_fmu-v5_default/NuttX/nuttx_olddefconfig.stamp
/home/admin1/Desktop/Testing 2/Firmware/build/px4_fmu-v5_default/NuttX/nuttx/tools/link.sh: line 100: [: /home/admin1/Desktop/Testing: binary operator expected
Makefile:83: /home/admin1/Desktop/Testing: No such file or directory
Makefile:83: 2/Firmware/build/px4_fmu-v5_default/NuttX/apps/Application.mk: No such file or directory
make[3]: *** No rule to make target '2/Firmware/build/px4_fmu-v5_default/NuttX/apps/Application.mk'.  Stop.
make[2]: *** [dirlinks] Error 2
make[1]: *** [dirlinks] Error 2
[2/1043] Performing configure step for 'px4io_firmware'
FAILED: external/Stamp/px4io_firmware/px4io_firmware-configure 
cd "/home/admin1/Desktop/Testing 2/Firmware/build/px4_fmu-v5_default/external/Build/px4io_firmware" && /usr/bin/cmake -DCONFIG=px4_io-v2_default -GNinja "/home/admin1/Desktop/Testing 2/Firmware" && /usr/bin/cmake -E touch "/home/admin1/Desktop/Testing 2/Firmware/build/px4_fmu-v5_default/external/Stamp/px4io_firmware/px4io_firmware-configure"
-- PX4 version: v1.11.0-beta2-38-g951969ba00
-- PX4 config file: /home/admin1/Desktop/Testing 2/Firmware/boards/px4/io-v2/default.cmake
-- PX4 config: px4_io-v2_default
-- PX4 platform: nuttx
-- cmake build type: MinSizeRel
-- ccache enabled (export CCACHE_DISABLE=1 to disable)
CMake Error at platforms/nuttx/cmake/px4_impl_os.cmake:128 (message):
  Could not determine chip architecture from NuttX config.  You may have to
  add it.
Call Stack (most recent call first):
  CMakeLists.txt:290 (px4_os_determine_build_chip)

-- Configuring incomplete, errors occurred!
See also "/home/admin1/Desktop/Testing 2/Firmware/build/px4_fmu-v5_default/external/Build/px4io_firmware/CMakeFiles/CMakeOutput.log".
See also "/home/admin1/Desktop/Testing 2/Firmware/build/px4_fmu-v5_default/external/Build/px4io_firmware/CMakeFiles/CMakeError.log".
[3/1043] Generating parameters.xml
ninja: build stopped: subcommand failed.
Makefile:198: recipe for target 'px4_fmu-v5_default' failed
make: *** [px4_fmu-v5_default] Error 1

What to do now any Solution ?, I have went through the documentation and installed all the dependencies of NuttX and other. But nothing is working.

mcsauder, 2020-05-23:
Hi @psp316r, the first part of this issue says that you are trying to build `fmu-v2`, but all of the errors say `fmu-v5`. Could you first confirm that your have run 'git submodule update --init --recursive` in your repo (to ensure that all of the submodule data is present in your local codebase), and then can you confirm which version you are trying to build, `v2` or `v5` so that the problem can be narrowed down?  Please also post your OS/build environment specifics to help narrow things to your build issues.  I built both v2 and v5 in Ubuntu 20.04 just a few moments ago to see if I could spot anything but am not seeing any issues on my new installation after utilizing the build scripts to install dependencies.

psp316r, 2020-05-23:
Hi @mcsauder. I have ran `git submodule update --init --recursive` and I building it for **V2**. And I am Using **Ubuntu 18.04**

Anyway Just Isolved the Issue it is due missing of NuttX Configuration, Since I am running on the Ubuntu 18.04 I used the [ ubuntu_sim_nuttx.sh ](https://raw.githubusercontent.com/PX4/Devguide/master/build_scripts/ubuntu_sim_nuttx.sh) to fix this issue.

1103706199, 2021-05-11:
Is it solved now? I had the same problem

psp316r, 2021-05-14:
I have solved the issue it is due to the missing NuttX Configuration, Since I am running on Ubuntu 18.04 I ran the ubuntu_sim_nuttx.sh to fix this issue.

closed 2021-05-14 by psp316r