PX4/PX4-Autopilot #14963: UDP cannot be larger than 14550 in v1.11 SITL

robin-shaun, 2020-05-23:
I use the same udp settings `<arg name="fcu_url" default="udp://:14551@127.0.0.1:34594"/>`, in v1.8.0 mavros connected True,  but in v1.11.0 mavros connected false.
Therefore, now I can only use 11 uavs (14540~14550) in v1.11, but in v1.8.0 I can use much more. I don't know how to solve it.
Two launch files for v1.8 and v1.11 are in the attachment.
[multi_uav_launch.zip](https://github.com/PX4/Firmware/files/4671193/multi_uav_launch.zip)

julianoes, 2020-05-25:
@Jaeyoung-Lim didn't we address this recently?

Jaeyoung-Lim, 2020-05-25:
@julianoes Yes we did, but more than 10 uavs the offboard remote udp port is fixed in `14549` as in https://github.com/PX4/Firmware/blob/master/ROMFS/px4fmu_common/init.d-posix/rcS#L112

closed 2020-05-25 by julianoes

robin-shaun, 2020-05-27:
Hello, I think the problem hasn't be solved. My rcS is the newest version. You said without using roslaunch, but I think the problem is the connection to mavros. In fact qgc can connect to more than 10 vehicles, but mavros cannot.
[image]

```
robin@robin-G3-3590:~/px4$ rostopic echo /iris_11/mavros/state
header: 
  seq: 0
  stamp: 
    secs: 2
    nsecs: 534000000
  frame_id: ''
connected: False
armed: False
guided: False
manual_input: False
mode: ''
system_status: 0
---
```

Jaeyoung-Lim, 2020-05-27:
@robin-shaun Okay, the problem seems to be that multiple mavros nodes are unable to use the same udp port (`14549`). As you can see in the rcS, the offboard remote port ranges from `14540~14549`

Why not use the `udp_offboard_port_local` or `udp_payload_port` that get different ports for each instances?