PX4/PX4-Autopilot #14964: "make px4_sitl gazebo" fails to build

KrisRadowski, 2020-05-23:
when I try to run gazebo (on Ubuntu 18.04) in headless mode, build fails on generating `/home/kris/Firmware/Tools/sitl_gazebo/models/iris/iris.sdf` with this error:
`FAILED: /home/kris/Firmware/Tools/sitl_gazebo/models/iris/iris.sdf
cd /home/kris/Firmware/Tools/sitl_gazebo && rm -f /home/kris/Firmware/Tools/sitl_gazebo/models/iris/iris.sdf && /usr/bin/python3 /home/kris/Firmware/Tools/sitl_gazebo/scripts/xacro.py -o /home/kris/Firmware/Tools/sitl_gazebo/models/rotors_description/urdf/iris_base.urdf /home/kris/Firmware/Tools/sitl_gazebo/models/rotors_description/urdf/iris_base.xacro enable_mavlink_interface:=true enable_ground_truth:=false enable_wind:=false enable_logging:=false rotors_description_dir:=/home/kris/Firmware/Tools/sitl_gazebo/models/rotors_description send_vision_estimation:=false send_odometry:=true && gz sdf -p /home/kris/Firmware/Tools/sitl_gazebo/models/rotors_description/urdf/iris_base.urdf >> /home/kris/Firmware/Tools/sitl_gazebo/models/iris/iris.sdf && rm -f /home/kris/Firmware/Tools/sitl_gazebo/models/rotors_description/urdf/iris_base.urdf

gz: symbol lookup error: /usr/lib/x86_64-linux-gnu/libgazebo_common.so.9: undefined symbol: _ZN8ignition10fuel_tools12ClientConfig12SetUserAgentERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE
[16/78] Building CXX object CMakeFiles/gazebo_uuv_plugin.dir/src/gazebo_uuv_plugin.cpp.o
ninja: build stopped: subcommand failed.
[2/9] Generating git version header
FAILED: external/Stamp/sitl_gazebo/sitl_gazebo-build
cd /home/kris/Firmware/build/px4_sitl_default/build_gazebo && /usr/bin/cmake --build .
ninja: build stopped: subcommand failed.
Makefile:205: recipe for target 'px4_sitl' failed
make: *** [px4_sitl] Error 1`

How to fix this?

Jaeyoung-Lim, 2020-05-23:
@KrisRadowski Looks like you are missing `libgazebo_common.so`

KrisRadowski, 2020-05-23:
> @KrisRadowski Looks like you are missing `libgazebo_common.so` Have you followed the toolchain installation instruction in the px4 documentation?

yes, I even make every step once again but it still fails; on top of that actually the offending library is not missing
```
kris@DESKTOP-NQV6QDH:~/Firmware$ ls -l /usr/lib/x86_64-linux-gnu/ | grep -i libgazebo_common
lrwxrwxrwx  1 root root       21 Apr  3 22:25 libgazebo_common.so -> libgazebo_common.so.9
lrwxrwxrwx  1 root root       26 Apr  3 22:25 libgazebo_common.so.9 -> libgazebo_common.so.9.13.0
-rw-r--r--  1 root root  1219224 Apr  3 22:25 libgazebo_common.so.9.13.0
```

closed 2020-05-23 by KrisRadowski