PX4/PX4-Autopilot #14975: feature request : Add CRUISE mode for fixedwings

moreba1, 2020-05-26:
Hi

we need a cruise mode for fixed-wing . it is like Position mode however airspeed must be fixed and be Independent from throttle stick . this mode can decrease workload of the pilot and prevent from stall.

bresch, 2020-05-26:
@moreba1 Do you need this because you need to use the throttle stick for something else? Usually, flying at cruise speed is made easy using a spring-loaded throttle stick.

moreba1, 2020-05-26:
> 
> 
> @moreba1 Do you need this because you need to use the throttle stick for something else? Usually, flying at cruise speed is made easy using a spring-loaded throttle stick.

No, I will use a tablet for GCS and don't use RC . in some situations there is a need to change mode to position for switching a VTOL between Fixedwing and Multirotor.

bresch, 2020-05-26:
@moreba1 Ok, so you have a joystick, right?

I'm trying to understand why you would need this mode.
> this mode can decrease workload of the pilot

If you don't want to let the pilot change the speed, you can set cruise = min = max airspeed. Does he need to change it sometimes?

> and prevent from stall

If TECS (altitude + speed controller) is tuned properly, the plane won't stall, even with the throttle stick at 0 and full pitch up.

moreba1, 2020-05-26:
> @moreba1 Ok, so you have a joystick, right?

Yes

> If you don't want to let the pilot change the speed, you can set cruise = min = max airspeed. Does he need to change it sometimes?

Yes, we can limit the range of airspeed. and it will fly with the minimum airspeed at 0 throttle.

bresch, 2020-05-26:
Cool, I'll close the issue then. Feel free to reopen if something is still not clear.

closed 2020-05-26 by bresch