PX4/PX4-Autopilot #14981: Drone flies towards old mission waypoint on arm, when no current=1 is set on new mission

Crowdedlight, 2020-05-26:
**Describe the bug**
When a mission is stopped midway, and the drone then lands, receives a new mission with all items having ``current=0`` parameters and is armed again, the drone immediately flies towards the waypoint from the old mission, instead of executing the first item on the new mission. This crashed our drone as it tilts and tries to fly forward before properly "takeoff" to a safe altitude, thus catching the ground with the landing struts.  

The resetting of ``mission_current`` seems to happen upon arming, but there seems to be some sort of race-condition, as the drone reports ``mission_current`` to reset to 0 upon arming, but the log shows the drone still gets a setpoint for the old waypoint and tries to fly that, before the ``mission_current`` index reset is realised.

Thomas Stauber spotted that our velocity controller asked for a 3m/s horizontal velocity immediately upon arming, which indicated that the drone was flying towards an old mission item.
[image] 

**To Reproduce**
Steps we did to reproduce the behaviour:
1. Fly normal mission
2. Before normal mission finishes, switch mode to "land now" or "RTL" and land the drone
3. Start new flight with identical mission. (It uploads the same mission again, but doesn't clear the old one first)
4. Arm and send "start_mission" command
5. Drone attempts to fly to the waypoint it was at in the last mission and thus flip in the direction of that waypoint.

**Expected behavior**
Upon uploading a new mission with no ``current=1`` parameters set, we would expect the drone to reset the current-index of mission items to 0. Intuitively I would expect the index to get reset when the PX4 accepts the new mission and sends ``MAV_MISSION_ACCEPTED`` and not first on arming the drone. 

**Log Files and Screenshots**
flight log: https://review.px4.io/plot_app?log=116b78e1-7732-4e6e-acf8-88e640d8cdfc 
video of the takeoff and flip: https://nextcloud.sdu.dk/index.php/s/46H7m9TXgXkQqQF

log from our GCS rosbag, showing the ``mission_current`` is not reset until arming, but keeps the old missions current index. The new mission had been uploaded and accepted by the PX4 before this snippet from the log:
```
---
data: 0
---
data: 0
---            <----------- a second before it changes to 15, the flightmode of the FC is set to "Auto - Mission". It appears the current_waypoint change is due to the flightmode being changed
data: 15
---
data: 15
---
data: 15
---
data: 15
---
data: 15
---            <----------- Right at the same time it goes from 15 to 0, is when the drone report it is "armed". So it seems like the index isn't reset until the drone is armed
data: 0
---
data: 0
---
data: 0
---
```

**Drone (please complete the following information):**
- Hexacopter using Pixhawk 2.1 Cube
- (The long legs shown on the video are due to testing, we know they are long, ugly and doesn't help make it stable)
- SD-card for PX4: I will update with exact details tomorrow when I am back in the office, but I believe it was a Scandisk 8/16GB class 10. 

**Additional context**

As soon as we added ``current=1`` on the first item of the new mission we can no longer reproduce this error and we are using that fix for the time being.

julianoes, 2020-05-27:
@RomanBapst does this sound familiar?

julianoes, 2020-05-27:
> As soon as we added current=1 on the first item of the new mission we can no longer reproduce this error and we are using that fix for the time being.

This makes sense, and should always be done. However, if it is not set, PX4 should either discard the mission or automatically set 0 to be current, right?

julianoes, 2020-09-01:
Sorry, I obviously don't have time for this one at the moment.

Crowdedlight, 2020-09-04:
Fair enough. I would agree that it makes sense for PX4 to set current=0, if it isn't specified in the received mission.