PX4/PX4-Autopilot #14982: Error in spawning multi uav

mohdwaseem27, 2020-05-26:
I'm new to this, I'm trying to spawn multi uav 
I launched it by `roslaunch px4 multi_uav_mavros_sitl.launch` this
and I get this error

`SUMMARY
========

CLEAR PARAMETERS
 * /uav0/mavros/
 * /uav1/mavros/
 * /uav2/mavros/

PARAMETERS
 * /gazebo/enable_ros_network: True
 * /rosdistro: melodic
 * /rosversion: 1.14.5
 * /uav0/mavros/cmd/use_comp_id_system_control: False
 * /uav0/mavros/conn/heartbeat_rate: 1.0
 * /uav0/mavros/conn/system_time_rate: 1.0
 * /uav0/mavros/conn/timeout: 10.0
 * /uav0/mavros/conn/timesync_rate: 10.0
 * /uav0/mavros/distance_sensor/hrlv_ez4_pub/field_of_view: 0.0
 * /uav0/mavros/distance_sensor/hrlv_ez4_pub/frame_id: hrlv_ez4_sonar
 * /uav0/mavros/distance_sensor/hrlv_ez4_pub/id: 0
 * /uav0/mavros/distance_sensor/hrlv_ez4_pub/orientation: PITCH_270
 * /uav0/mavros/distance_sensor/hrlv_ez4_pub/send_tf: True
 * /uav0/mavros/distance_sensor/hrlv_ez4_pub/sensor_position/x: 0.0
 * /uav0/mavros/distance_sensor/hrlv_ez4_pub/sensor_position/y: 0.0
 * /uav0/mavros/distance_sensor/hrlv_ez4_pub/sensor_position/z: -0.1
 * /uav0/mavros/distance_sensor/laser_1_sub/id: 3
 * /uav0/mavros/distance_sensor/laser_1_sub/orientation: PITCH_270
 * /uav0/mavros/distance_sensor/laser_1_sub/subscriber: True
 * /uav0/mavros/distance_sensor/lidarlite_pub/field_of_view: 0.0
 * /uav0/mavros/distance_sensor/lidarlite_pub/frame_id: lidarlite_laser
 * /uav0/mavros/distance_sensor/lidarlite_pub/id: 1
 * /uav0/mavros/distance_sensor/lidarlite_pub/orientation: PITCH_270
 * /uav0/mavros/distance_sensor/lidarlite_pub/send_tf: True
 * /uav0/mavros/distance_sensor/lidarlite_pub/sensor_position/x: 0.0
 * /uav0/mavros/distance_sensor/lidarlite_pub/sensor_position/y: 0.0
 * /uav0/mavros/distance_sensor/lidarlite_pub/sensor_position/z: -0.1
 * /uav0/mavros/distance_sensor/sonar_1_sub/id: 2
 * /uav0/mavros/distance_sensor/sonar_1_sub/orientation: PITCH_270
 * /uav0/mavros/distance_sensor/sonar_1_sub/subscriber: True
 * /uav0/mavros/fake_gps/eph: 2.0
 * /uav0/mavros/fake_gps/epv: 2.0
 * /uav0/mavros/fake_gps/fix_type: 3
 * /uav0/mavros/fake_gps/geo_origin/alt: 408.0
 * /uav0/mavros/fake_gps/geo_origin/lat: 47.3667
 * /uav0/mavros/fake_gps/geo_origin/lon: 8.55
 * /uav0/mavros/fake_gps/gps_rate: 5.0
 * /uav0/mavros/fake_gps/mocap_transform: True
 * /uav0/mavros/fake_gps/satellites_visible: 5
 * /uav0/mavros/fake_gps/tf/child_frame_id: fix
 * /uav0/mavros/fake_gps/tf/frame_id: map
 * /uav0/mavros/fake_gps/tf/listen: False
 * /uav0/mavros/fake_gps/tf/rate_limit: 10.0
 * /uav0/mavros/fake_gps/tf/send: False
 * /uav0/mavros/fake_gps/use_mocap: True
 * /uav0/mavros/fake_gps/use_vision: False
 * /uav0/mavros/fcu_protocol: v2.0
 * /uav0/mavros/fcu_url: udp://:14540@loca...
 * /uav0/mavros/gcs_url: 
 * /uav0/mavros/global_position/child_frame_id: base_link
 * /uav0/mavros/global_position/frame_id: map
 * /uav0/mavros/global_position/gps_uere: 1.0
 * /uav0/mavros/global_position/rot_covariance: 99999.0
 * /uav0/mavros/global_position/tf/child_frame_id: base_link
 * /uav0/mavros/global_position/tf/frame_id: map
 * /uav0/mavros/global_position/tf/global_frame_id: earth
 * /uav0/mavros/global_position/tf/send: False
[... 944 lines cut ...]
[ INFO] [1590529212.286608202, 2.412000000]: VER: 2.1: Flight software:     010b0000 (1c0925a189000000)
[ INFO] [1590529212.286812400, 2.412000000]: VER: 2.1: Middleware software: 010b0000 (1c0925a189000000)
[ INFO] [1590529212.286998641, 2.412000000]: VER: 2.1: OS software:         050300ff (0000000000000000)
[ INFO] [1590529212.287178291, 2.412000000]: VER: 2.1: Board hardware:      00000001
[ INFO] [1590529212.287357003, 2.412000000]: VER: 2.1: VID/PID:             0000:0000
[ INFO] [1590529212.287518391, 2.412000000]: VER: 2.1: UID:                 4954414c44494e50
[ INFO] [1590529212.506116968, 2.632000000]: VER: 3.1: Capabilities         0x000000000000e4ef
[ INFO] [1590529212.506417768, 2.632000000]: VER: 3.1: Flight software:     010b0000 (1c0925a189000000)
[ INFO] [1590529212.506507520, 2.632000000]: VER: 3.1: Middleware software: 010b0000 (1c0925a189000000)
[ INFO] [1590529212.506596335, 2.632000000]: VER: 3.1: OS software:         050300ff (0000000000000000)
[ INFO] [1590529212.506687986, 2.632000000]: VER: 3.1: Board hardware:      00000001
[ INFO] [1590529212.506768495, 2.632000000]: VER: 3.1: VID/PID:             0000:0000
[ INFO] [1590529212.506854789, 2.632000000]: VER: 3.1: UID:                 4954414c44494e51
INFO  [ecl/EKF] 5192000: reset position to GPS
INFO  [ecl/EKF] 5192000: reset velocity to GPS
INFO  [ecl/EKF] 5192000: starting GPS fusion
INFO  [ecl/EKF] 5192000: reset position to GPS
INFO  [ecl/EKF] 5192000: reset velocity to GPS
INFO  [ecl/EKF] 5192000: starting GPS fusion
[ INFO] [1590529226.381593872, 16.412000000]: WP: mission received
[ INFO] [1590529226.609486547, 16.632000000]: WP: mission received
`
can some one please help me out of this 
However I the gazebo is spwaning 3 uavs's 
Thanks

julianoes, 2020-05-27:
@Jaeyoung-Lim have you seen this before?
`ERROR [px4] is_server_running: failed to create lock file: /tmp/px4_lock-0`

Jaeyoung-Lim, 2020-05-27:
@mohdwaseem27 Are you running this in a container?

mohdwaseem27, 2020-05-27:
@Jaeyoung-Lim sorry I don't exactly know what a container is 

I'm using this on a dual boot along side windows

closed 2020-05-30 by mohdwaseem27

HenryHuYu, 2020-06-26:
did u get this solved?

mohdwaseem27, 2020-06-26:
@HenryHuYu  Yes, but I don't know what the problem was. I restarted my system and the problem was solved

SEUZTh, 2020-12-13:
请问您解决了没？

SEUZTh, 2020-12-13:
The lock is occupied. You need to release it. If you reboot the PC, the lock wiil be released.

groers, 2022-03-02:
> The lock is occupied. You need to release it. If you reboot the PC, the lock wiil be released.

Thanks bro. After the reboot, the same question is solved.  谢谢兄弟重启后问题解决了

julianoes, 2022-03-06:
If you don't want to reboot, you probably just have to remove the lock file:
```
sudo rm -rf /tmp/px4_lock-0
```

groers, 2022-03-06:
这是来自QQ邮箱的自动回复邮件。你好，你的邮件已收到。