PX4/PX4-Autopilot #14984: Error when running ubuntu_sim_ros_melodic.sh script

daylanc, 2020-05-27:
Hello,

Not sure if I'm posting this in the correct section, but I am running into issues when running the following command: `bash ubuntu_sim_ros_melodic.sh` script from the following installation website: https://dev.px4.io/master/en/setup/dev_env_linux_ubuntu.html

I successfully installed Gazebo and JMAVSim using the `bash ./Tools/setup/ubuntu.sh` script, and seem to have installed ROS Melodic from the first part of the `ubuntu_sim_ros_melodic.sh` script, but run into the following errors:

[image]
[image]

Hopefully someone could help me understand what dependencies i am missing or if I missed something during the installation steps.

Thanks!

julianoes, 2020-05-27:
It says `No module named future`.

Can you try `pip install --user future`.

daylanc, 2020-05-27:
@julianoes Thanks for the reply. after running your suggested command, it seems like I could install 1 of 6 packages. Here's the next error I get:

[image]

TSC21, 2020-05-27:
> @julianoes Thanks for the reply. after running your suggested command, it seems like I could install 1 of 6 packages. Here's the next error I get:
> 
> [image]

Are you sure that the install scripts didn't run into any errors?

TSC21, 2020-05-27:
Also why are you building MAVROS from source? If you want to do that you need to follow the propper procedures so that the dependencies are installed: https://github.com/mavlink/mavros/tree/master/mavros#source-installation

daylanc, 2020-05-27:
@TSC21 I was following the Environment Setup page to get started: https://dev.px4.io/master/en/setup/dev_env_linux_ubuntu.html

Not sure if this is the correct way to get started with PX4 and simulation or not

TSC21, 2020-05-27:
> @TSC21 I was following the Environment Setup page to get started: https://dev.px4.io/master/en/setup/dev_env_linux_ubuntu.html
> 
> Not sure if this is the correct way to get started with PX4 and simulation or not

OK and where in that documentation you read that MAVROS is to be installed from source?

daylanc, 2020-05-27:
@TSC21 I wasn't trying to install MAVROS from source. That's the error I received when I ran `bash ubuntu_sim_ros_melodic.sh`

TSC21, 2020-05-27:
> @TSC21 I wasn't trying to install MAVROS from source. That's the error I received when I ran `bash ubuntu_sim_ros_melodic.sh`

Ok. So does any other error appear before the setup reaches to that stage?

daylanc, 2020-05-27:
Here's my entire output:

[image]
[image]
[image]
[image]

TSC21, 2020-05-27:
`rosdep: command not found` means that you probably don't have the ROS distro `setup.bash` properly source or even not properly installed.

TSC21, 2020-05-27:
Also, try to manually install and setup rosdep - check section 1.6.1 in http://wiki.ros.org/melodic/Installation/Ubuntu.

daylanc, 2020-05-27:
@TSC21 manually installing rosdep seemed to be the solution. All 6 packages were built successfully. I wonder why this wasn't done with the `bash ubuntu_sim_ros_melodic.sh` script? I started this install with a clean install of Ubuntu 18.04.

closed 2020-05-27 by daylanc

TSC21, 2020-05-27:
> @TSC21 manually installing rosdep seemed to be the solution. All 6 packages were built successfully. I wonder why this wasn't done with the `bash ubuntu_sim_ros_melodic.sh` script? I started this install with a clean install of Ubuntu 18.04.

@hamishwillee

hamishwillee, 2020-05-28:
Can someone here create a PR to update it as needed - since you know what needs to happen. I'm happy to test it afterwards.

hamishwillee, 2020-06-02:
@daylanc @TSC21 Neither of you able to update the script?

TSC21, 2020-06-09:
Fix in https://github.com/PX4/Devguide/pull/1045. @hamishwillee sorry for the delay.

hamishwillee, 2020-06-10:
Thanks Nuno!

jmurraylouw, 2021-04-20:
This is still a problem. I just ran `bash ubuntu_sim_ros_melodic.sh` on a clean install on Ubuntu 18 and it gives the same error.

Running:
```
sudo rosdep init
rosdep update
```
also solved the problem.

Maybe the updated script wasn't uploaded to where it is download from? (`wget https://raw.githubusercontent.com/PX4/Devguide/master/build_scripts/ubuntu_sim_ros_melodic.sh`)