PX4/PX4-Autopilot #14988: Error when compiling PX4 SITL

XueZhihan, 2020-05-27:
Hey all. I'm trying to compile PX4 on Ubuntu 20.04 and get this error:

<pre><font color="#4E9A06"><b>zhihan@zhihan-MS-7B17</b></font>:<font color="#3465A4"><b>~/PX4/Firmware</b></font>$ make posix_sitl_ekf2  none_iris
ninja: Entering directory `/home/zhihan/PX4/Firmware/build/posix_sitl_ekf2&apos;
[1/295] Building C object src/modules/mavlin...s__mavlink__mavlink_tests.dir/__/mavlink.c.o
<font color="#CC0000">FAILED: </font>src/modules/mavlink/mavlink_tests/CMakeFiles/modules__mavlink__mavlink_tests.dir/__/mavlink.c.o 
/usr/bin/ccache /usr/bin/cc -DCONFIG_ARCH_BOARD_SITL -DMODULE_NAME=\&quot;mavlink_tests\&quot; -DPX4_MAIN=mavlink_tests_app_main -D__DF_LINUX -D__PX4_LINUX -D__PX4_POSIX -D__STDC_FORMAT_MACROS -Dnoreturn_function=&quot;__attribute__((noreturn))&quot; -I. -Isrc -Isrc/lib -Isrc/modules -I../../src -I../../src/drivers/boards/sitl -I../../src/include -I../../src/lib -I../../src/lib/DriverFramework/framework/include -I../../src/lib/matrix -I../../src/modules -I../../src/platforms -I../../platforms/posix/include -Iexternal/Install/include -I../../mavlink/include/mavlink -g -std=gnu99 -fno-common -Wall -Warray-bounds -Wdisabled-optimization -Werror -Wextra -Wfatal-errors -Wfloat-equal -Wformat-security -Winit-self -Wlogical-op -Wmissing-declarations -Wmissing-field-initializers -Wpointer-arith -Wshadow -Wuninitialized -Wunknown-pragmas -Wunused-variable -Wno-implicit-fallthrough -Wno-unused-parameter -Wunused-but-set-variable -Wformat=1 -Wdouble-promotion -Wbad-function-cast -Wstrict-prototypes -Wmissing-prototypes -Wnested-externs -fvisibility=hidden -include visibility.h -fno-strict-aliasing -fomit-frame-pointer -fno-math-errno -funsafe-math-optimizations -ffunction-sections -fdata-sections -fno-strength-reduce -fno-builtin-printf -fdiagnostics-color=always -O2 -g -DNDEBUG   -fno-strict-aliasing -fomit-frame-pointer -fno-math-errno -funsafe-math-optimizations -ffunction-sections -fdata-sections -fno-strength-reduce -fno-builtin-printf -fdiagnostics-color=always -Wno-sign-compare -DMAVLINK_FTP_UNIT_TEST -DMavlinkStream=MavlinkStreamTest -DMavlinkFTP=MavlinkFTPTest -std=gnu99 -MD -MT src/modules/mavlink/mavlink_tests/CMakeFiles/modules__mavlink__mavlink_tests.dir/__/mavlink.c.o -MF src/modules/mavlink/mavlink_tests/CMakeFiles/modules__mavlink__mavlink_tests.dir/__/mavlink.c.o.d -o src/modules/mavlink/mavlink_tests/CMakeFiles/modules__mavlink__mavlink_tests.dir/__/mavlink.c.o   -c ../../src/modules/mavlink/mavlink.c
In file included from <b>../../mavlink/include/mavlink/v2.0/standard/../protocol.h:83</b>,
                 from <b>../../mavlink/include/mavlink/v2.0/standard/standard.h:30</b>,
                 from <b>../../mavlink/include/mavlink/v2.0/standard/mavlink.h:32</b>,
                 from <b>../../src/modules/mavlink/mavlink_bridge_header.h:89</b>,
                 from <b>../../src/modules/mavlink/mavlink.c:46</b>:
<b>../../mavlink/include/mavlink/v2.0/standard/../mavlink_helpers.h:</b> In function ‘<b>mavlink_finalize_message_buffer</b>’:
<b>../../mavlink/include/mavlink/v2.0/standard/../mavlink_helpers.h:248:24:</b> <font color="#CC0000"><b>error: </b></font>taking address of packed member of ‘<b>struct __mavlink_message</b>’ may result in an unaligned pointer value [<font color="#CC0000"><b>-Werror=address-of-packed-member</b></font>]
  248 |  crc_accumulate_buffer(<font color="#CC0000"><b>&amp;msg-&gt;checksum</b></font>, _MAV_PAYLOAD(msg), msg-&gt;len);
      |                        <font color="#CC0000"><b>^~~~~~~~~~~~~~</b></font>
compilation terminated due to -Wfatal-errors.
cc1: all warnings being treated as errors
[2/295] Building CXX object src/drivers/dist...sf0x/CMakeFiles/drivers__sf0x.dir/sf0x.cpp.o
<font color="#CC0000">FAILED: </font>src/drivers/distance_sensor/sf0x/CMakeFiles/drivers__sf0x.dir/sf0x.cpp.o 
/usr/bin/ccache /usr/bin/c++  -DCONFIG_ARCH_BOARD_SITL -DMODULE_NAME=\&quot;sf0x\&quot; -DPX4_MAIN=sf0x_app_main -D__DF_LINUX -D__PX4_LINUX -D__PX4_POSIX -D__STDC_FORMAT_MACROS -Dnoreturn_function=&quot;__attribute__((noreturn))&quot; -I. -Isrc -Isrc/lib -Isrc/modules -I../../src -I../../src/drivers/boards/sitl -I../../src/include -I../../src/lib -I../../src/lib/DriverFramework/framework/include -I../../src/lib/matrix -I../../src/modules -I../../src/platforms -I../../platforms/posix/include -Iexternal/Install/include -g -fno-exceptions -fno-rtti -std=gnu++11 -fno-threadsafe-statics -DCONFIG_WCHAR_BUILTIN -D__CUSTOM_FILE_IO__ -fcheck-new -Wall -Warray-bounds -Wdisabled-optimization -Werror -Wextra -Wfatal-errors -Wfloat-equal -Wformat-security -Winit-self -Wlogical-op -Wmissing-declarations -Wmissing-field-initializers -Wpointer-arith -Wshadow -Wuninitialized -Wunknown-pragmas -Wunused-variable -Wno-implicit-fallthrough -Wno-unused-parameter -Wunused-but-set-variable -Wformat=1 -Wdouble-promotion -Wno-missing-field-initializers -Wno-overloaded-virtual -Wreorder -Wno-format-truncation -fvisibility=hidden -include visibility.h -fno-strict-aliasing -fomit-frame-pointer -fno-math-errno -funsafe-math-optimizations -ffunction-sections -fdata-sections -fno-strength-reduce -fno-builtin-printf -fdiagnostics-color=always -O2 -g -DNDEBUG   -fno-strict-aliasing -fomit-frame-pointer -fno-math-errno -funsafe-math-optimizations -ffunction-sections -fdata-sections -fno-strength-reduce -fno-builtin-printf -fdiagnostics-color=always -Wno-sign-compare -std=gnu++11 -MD -MT src/drivers/distance_sensor/sf0x/CMakeFiles/drivers__sf0x.dir/sf0x.cpp.o -MF src/drivers/distance_sensor/sf0x/CMakeFiles/drivers__sf0x.dir/sf0x.cpp.o.d -o src/drivers/distance_sensor/sf0x/CMakeFiles/drivers__sf0x.dir/sf0x.cpp.o -c ../../src/drivers/distance_sensor/sf0x/sf0x.cpp
In file included from <b>/usr/include/string.h:495</b>,
                 from <b>../../src/drivers/distance_sensor/sf0x/sf0x.cpp:51</b>:
In function ‘<b>char* strncpy(char*, const char*, size_t)</b>’,
    inlined from ‘<b>SF0X::SF0X(const char*, uint8_t)</b>’ at <b>../../src/drivers/distance_sensor/sf0x/sf0x.cpp:196:9</b>:
<b>/usr/include/x86_64-linux-gnu/bits/string_fortified.h:106:34:</b> <font color="#CC0000"><b>error: </b></font>‘char* __builtin_strnc<b>py(char*, const char*, long unsigned int)</b>’ specified bound 20 equals destination size [<font color="#CC0000"><b>-Werror=stringop-truncation</b></font>]
  106 |   return <font color="#CC0000"><b>__builtin___strncpy_chk (__dest, __src, __len, __bos (__dest))</b></font>;
      |          <font color="#CC0000"><b>~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~</b></font>
compilation terminated due to -Wfatal-errors.
cc1plus: all warnings being treated as errors
[3/295] Building CXX object src/drivers/dist.../CMakeFiles/drivers__tfmini.dir/tfmini.cpp.o
<font color="#CC0000">FAILED: </font>src/drivers/distance_sensor/tfmini/CMakeFiles/drivers__tfmini.dir/tfmini.cpp.o 
/usr/bin/ccache /usr/bin/c++  -DCONFIG_ARCH_BOARD_SITL -DMODULE_NAME=\&quot;tfmini\&quot; -DPX4_MAIN=tfmini_app_main -D__DF_LINUX -D__PX4_LINUX -D__PX4_POSIX -D__STDC_FORMAT_MACROS -Dnoreturn_function=&quot;__attribute__((noreturn))&quot; -I. -Isrc -Isrc/lib -Isrc/modules -I../../src -I../../src/drivers/boards/sitl -I../../src/include -I../../src/lib -I../../src/lib/DriverFramework/framework/include -I../../src/lib/matrix -I../../src/modules -I../../src/platforms -I../../platforms/posix/include -Iexternal/Install/include -g -fno-exceptions -fno-rtti -std=gnu++11 -fno-threadsafe-statics -DCONFIG_WCHAR_BUILTIN -D__CUSTOM_FILE_IO__ -fcheck-new -Wall -Warray-bounds -Wdisabled-optimization -Werror -Wextra -Wfatal-errors -Wfloat-equal -Wformat-security -Winit-self -Wlogical-op -Wmissing-declarations -Wmissing-field-initializers -Wpointer-arith -Wshadow -Wuninitialized -Wunknown-pragmas -Wunused-variable -Wno-implicit-fallthrough -Wno-unused-parameter -Wunused-but-set-variable -Wformat=1 -Wdouble-promotion -Wno-missing-field-initializers -Wno-overloaded-virtual -Wreorder -Wno-format-truncation -fvisibility=hidden -include visibility.h -fno-strict-aliasing -fomit-frame-pointer -fno-math-errno -funsafe-math-optimizations -ffunction-sections -fdata-sections -fno-strength-reduce -fno-builtin-printf -fdiagnostics-color=always -O2 -g -DNDEBUG   -fno-strict-aliasing -fomit-frame-pointer -fno-math-errno -funsafe-math-optimizations -ffunction-sections -fdata-sections -fno-strength-reduce -fno-builtin-printf -fdiagnostics-color=always -Wno-sign-compare -std=gnu++11 -MD -MT src/drivers/distance_sensor/tfmini/CMakeFiles/drivers__tfmini.dir/tfmini.cpp.o -MF src/drivers [... cut ...]
  248 |  crc_accumulate_buffer(<font color="#CC0000"><b>&amp;msg-&gt;checksum</b></font>, _MAV_PAYLOAD(msg), msg-&gt;len);
      |                        <font color="#CC0000"><b>^~~~~~~~~~~~~~</b></font>
compilation terminated due to -Wfatal-errors.
cc1plus: all warnings being treated as errors
[10/295] Building CXX object src/modules/uOR...__uORB__uORB_tests.dir/uORB_tests_main.cpp.o
ninja: build stopped: subcommand failed.
make: *** [Makefile:148：posix_sitl_ekf2] Error 1</pre>

Hopefully someone could help me out of this.
Thanks.

julianoes, 2020-05-29:
Thanks for the issue.

This is a warning. You can disable that it errors by removing:
https://github.com/PX4/Firmware/blob/7eee9deb4c8ee08ed66019c2342007991fe59819/cmake/px4_add_common_flags.cmake#L71

It looks like you are compiling an older version of PX4 because you are using `make posix_sitl_ekf2` instead of `make px4_sitl`. Ubuntu 20.04 support was only added recently for newer PX4 versions.

XueZhihan, 2020-05-31:
It works! Thank you so much!

closed 2020-05-31 by XueZhihan