PX4/PX4-Autopilot #14989: PX4 Sensor Orientations Documentation

HecticSalad, 2020-05-28:
This is more of a request for some more thorough documentation on Sensor Orientation in the PX4 docs. On this page: https://docs.px4.io/master/en/config/flight_controller_orientation.html, under the _Calculating Orientation_ section, when rotating the Pixhawk about more than one axis, I was confused about whether your reference frame changes after the first rotation. This lead to more confusion because, if the reference frame does change after the first rotation, what is the correct order of rotations?

I have since found out that the reference frame refers to the reference frame of the vehicle (not the FCU), so when you rotate the Pixhawk 90 degrees about the x axis (roll), performing a subsequent 90 degree rotation about the z axis (yaw) is about the vehicles reference frame, NOT the Pixhawk's (now changed) reference frame.

The docs give an example of a rotation about one axis, but this is insufficient information if the user wants to rotate about 2 or 3 axes and can be very time-consuming and frustrating figuring out how to do it.

Thanks in advance.

julianoes, 2020-05-29:
Thanks for the note. Correct, it's extrinsic rotation.

We would of course very appreciate a pull request to improve the docs here:
https://github.com/PX4/px4_user_guide/blob/master/en/config/flight_controller_orientation.md

FYI @hamishwillee

hamishwillee, 2020-06-01:
Thanks very much. How about a very simple [change the docs](http://docs.px4.io/master/en/config/flight_controller_orientation.html#calculating-orientation) (changes highlighted in bold).

> YAW, PITCH and/or ROLL offsets **of the flight controller** are calculated relative to the **vehicle** forward-facing-upright orientation (clock-wise rotation around the Z, Y and X axis, respectively).

@HecticSalad would that have removed the ambiguity?

hamishwillee, 2020-06-10:
@julianoes I fixed this in https://github.com/PX4/px4_user_guide/pull/750 - can you review that?

julianoes, 2020-06-10:
Ok, I made a suggestion.

HecticSalad, 2020-06-10:
@julianoes is there any chance you can put up an example image that covers two or more axes of rotation? The current one is just about the yaw axis which is not enough.

hamishwillee, 2020-06-11:
@HecticSalad Or perhaps you could ? Happy to take a PR.

closed 2020-06-11 by hamishwillee

junwoo091400, 2022-11-07:
> 

That would certainly be helpful! @McKayWhight do you still have an image? You can definitely make a PR for that :wink: 

I noticed this issue while preparing docs for the https://github.com/PX4/PX4-Autopilot/pull/20247