PX4/PX4-Autopilot #14994: 1.11 till master may 28´th on PixRacer gyro 0 and acc 0 get wrong values

taileron, 2020-05-28:
Pixracer gets strange raw values from gyro 0 and acc 0 so gyro cal always ends with “motion retrying” error.
(with 1.10 it the whole cal procedure could be finished with success)
Reducing imu_gyro_ratemax didn´t solve it.
gyro0 and acc0 disabled doesn´t help as well, the cal procedure still tries to use gyro0  
Here are the sensors values from console:
sensors status
INFO [sensors] gyro status:
INFO [ecl/validation] validator: best: 1, prev best: 1, failsafe: YES (4 events)
INFO [ecl/validation] sensor #0, prio: 1, state: TOUT ECNT
INFO [ecl/validation] val: 0.4073, lp: 0.1746 mean dev: 0.0076 RMS: 1.0619 conf: 0.0000
INFO [ecl/validation] val: 1.2216, lp: -0.1350 mean dev: 0.0624 RMS: 1.2962 conf: 0.0000
INFO [ecl/validation] val: -0.2300, lp: -0.0765 mean dev: -0.0917 RMS: 1.1297 conf: 0.0000
INFO [ecl/validation] sensor #1, prio: 100, state: OK
INFO [ecl/validation] val: -0.0058, lp: 0.0028 mean dev: -0.0000 RMS: 0.0023 conf: 1.0000
INFO [ecl/validation] val: -0.0019, lp: -0.0011 mean dev: 0.0000 RMS: 0.0019 conf: 1.0000
INFO [ecl/validation] val: 0.0027, lp: 0.0015 mean dev: 0.0000 RMS: 0.0014 conf: 1.0000
INFO [sensors] accel status:
INFO [ecl/validation] validator: best: 1, prev best: 1, failsafe: YES (4 events)
INFO [ecl/validation] sensor #0, prio: 1, state: TOUT ECNT
INFO [ecl/validation] val: -0.1547, lp: 6.1306 mean dev: 0.6197 RMS: 10.1262 conf: 0.0000
INFO [ecl/validation] val: 24.8830, lp: 6.1518 mean dev: -0.2533 RMS: 9.6765 conf: 0.0000
INFO [ecl/validation] val: 13.0815, lp: 6.6939 mean dev: 0.1947 RMS: 9.7891 conf: 0.0000
INFO [ecl/validation] sensor #1, prio: 100, state: OK
INFO [ecl/validation] val: 0.2108, lp: 0.2164 mean dev: 0.0005 RMS: 0.0379 conf: 1.0000
INFO [ecl/validation] val: 0.1070, lp: 0.0857 mean dev: 0.0001 RMS: 0.0378 conf: 1.0000
INFO [ecl/validation] val: -9.5672, lp: -9.6160 mean dev: -0.0002 RMS: 0.0269 conf: 1.0000
INFO [sensors] mag status:
INFO [ecl/validation] validator: best: -1, prev best: -1, failsafe: NO (0 events)

INFO [sensors] Airspeed status:
INFO [ecl/validation] no data

INFO [vehicle_acceleration] selected sensor: 2359306 (1)
INFO [vehicle_acceleration] bias: [0.000 0.000 0.000]
INFO [vehicle_acceleration] sample rate: 799.041 Hz

INFO [vehicle_angular_velocity] selected sensor: 2359306 (1)
INFO [vehicle_angular_velocity] bias: [0.000 0.000 0.000]
INFO [vehicle_angular_velocity] sample rate: 799.041 Hz

INFO [vehicle_air_data] selected barometer: 3997714 (0)
vehicle_air_data: cycle: 12001 events, 822591us elapsed, 68.54us avg, min 20us max 622us 93.233us rms
INFO [ecl/validation] validator: best: 0, prev best: 0, failsafe: NO (0 events)
INFO [ecl/validation] sensor #0, prio: 75, state: OK
INFO [ecl/validation] val: 101380.0000, lp: 101378.8125 mean dev: -0.0567 RMS: 4.3212 conf: 1.0000
INFO [ecl/validation] val: 30.7600, lp: 30.7019 mean dev: 0.0802 RMS: 0.0169 conf: 1.0000
INFO [ecl/validation] val: 0.0000, lp: 0.0000 mean dev: 0.0000 RMS: 0.0000 conf: 1.0000
nsh> listener sensor_gyro

TOPIC: sensor_gyro 2 instances

Instance 0:
sensor_gyro_s
timestamp: 399377360 (0.008741 seconds ago)
timestamp_sample: 399376921 (439 us before timestamp)
device_id: 3801098 (Type: 0x3A, SPI:1 (0x00))
x: -2.3780
y: 2.7612
z: 0.8742
temperature: 0.0000

Instance 1:
sensor_gyro_s
timestamp: 399392838 (0.004444 seconds ago)
timestamp_sample: 399392599 (239 us before timestamp)
device_id: 2359306 (Type: 0x24, SPI:1 (0x00))
x: 0.0080
y: 0.0018
z: 0.0081
temperature: 47.070

dagar, 2020-05-28:
Thanks for reporting. Does that sensor (icm20608g) ever give you valid data?

dagar, 2020-05-28:
Could you try this branch? https://github.com/PX4/Firmware/tree/pr-invensense_imu_reset

It monitors for consecutive failures and issues a full sensor reset if necessary.

taileron, 2020-05-28:
> icm20608g
in the versions before 1.11 the sensor has worked it was only often necessary to calibrate at different temperatures

dagar, 2020-05-28:
It's a completely new driver using the sensor much differently. We now use the sensor's FIFO and SPI DMA to transfer raw 8 kHz data and do all the filtering PX4 side.

Can you try the branch?

taileron, 2020-05-29:
currently I´m fixed to Mac OS 10.12 (Sierra) and 10.15 (Catalina) and I´m still trying to get the build procedure working above PX 4 V1.8 
The gyro 0 x,y,z values are absolutely random and fluctuate very quickly between -5 and 5 accels 0 between -50 and 50

dagar, 2020-05-29:
I'll open a PR and point you to a binary.

taileron, 2020-05-29:
Got the build procedure in 10.12 working that branch behaves identical:
or can I manually do this reset?
temperature of accel 0 doesn´t read as well.
listener sensor_accel

TOPIC: sensor_accel 2 instances

Instance 0:
 sensor_accel_s
	timestamp: 145735613  (0.009395 seconds ago)
	timestamp_sample: 145735160  (453 us before timestamp)
	device_id: 3801098 (Type: 0x3A, SPI:1 (0x00)) 
	x: -20.7062
	y: -12.1798
	z: -17.7917
	temperature: 0.0000

Instance 1:
 sensor_accel_s
	timestamp: 145752130  (0.004013 seconds ago)
	timestamp_sample: 145751594  (536 us before timestamp)
	device_id: 2359306 (Type: 0x24, SPI:1 (0x00)) 
	x: -0.2159
	y: 0.1276
	z: -9.7805
	temperature: 40.6004

taileron, 2020-05-30:
This board had two magnetometers till v 1.10.1 now only one of them is found. 
(Tried an alternative GCC toolchain 9.2.1, then the random gyro 0 values get a bit lower.) 
I understand all advantages of the faster driver structure but as long as some imu sensors give strange values they should be able to ignored like the mag otherwise the cal procedure keeps failing.

taileron, 2020-05-30:
Now I got the chance to test another PixRacer unit R14 the previous was a very old R12
With master 1.11 it worked few minutes and afterwards the same strange values on gyro 0 accel 0 shown as on R12 from start.
With pr-invensense_imu_reset branch (build works only with GCC 7.2.1) it works but shows failsafe events in regular timeframes. 
One of the mags is not found anymore same as on R14 board.

i would like to do a flight test, what can happen if such an imu reset is performed in flight?

LorenzMeier, 2020-05-31:
@dagar Have you taken the sensor reset on that board away on boot? There was a *required* sensor reset (resetting the sensor power supply, NOT a register reset) being done on boot because some of these boards boot with browned out sensors.

Almost all of the "quirks" in software boot on PX4 are there for a reason to deal with broken hardware, so it's important to not to try to be clever with "cleanups".

dagar, 2020-05-31:
> This board had two magnetometers till v 1.10.1 now only one of them is found.

The pixracer has 2 internal mags, a dedicated mag (hmc5883 or lis3mdl) plus the ak8963 mag included in the mpu9250. The mpu9250 mag is now disabled by default because the internal i2c slave disrupts the scheduling (which we can see now), and it's also low quality.

> i would like to do a flight test, what can happen if such an imu reset is performed in flight?

It will switch over to the secondary IMU.

dagar, 2020-05-31:
> Now I got the chance to test another PixRacer unit R14 the previous was a very old R12
> With master 1.11 it worked few minutes but afterwards there are the same strange values on gyro 0 and accel 0 shown as on R12 from start.

@taileron anything else you can share about your usage? How are you powering these boards? Can you verify which have the icm20608g (older pixracers) and which have the icm20602? It you share a log from boot (SDLOG_MODE 1) I'll be able to tell.

dagar, 2020-05-31:
> @dagar Have you taken the sensor reset on that board away on boot? There was a _required_ sensor reset (resetting the sensor power supply, NOT a register reset) being done on boot because some of these boards boot with browned out sensors.

@LorenzMeier no definitely not, and in fact I wanted to expand that mechanism so it can be more tightly integrated with actual sensor init. It's worth double checking it's working as intended.

> Almost all of the "quirks" in software boot on PX4 are there for a reason to deal with broken hardware, so it's important to not to try to be clever with "cleanups".

These checks are evolving to catch more subtle failures that show up later in usage, not as a band-aid for bad hardware. For example this issue with the icm20689 in all CUAV fmu-v5 hardware (https://github.com/PX4/Firmware/issues/13083) that's plagued us long before the new drivers came into place. We want to try and recover if possible, but properly flagging and propagating errors from various sources (eg FIFO data integrity checks) is just as important.

dagar, 2020-05-31:
@taileron I've tried about 5 random pixracers I have on hand, but so far haven't reproduced any failures or bogus data. Most are mRo pixracers < 1 year, but 1 is an old AUAV pixracer with the icm20608g. I'll let them run a bit longer.

Any thing else you can share about your setup? What else is connected? Using a power module, USB power, or both?

dagar, 2020-05-31:
I've found an older board with an icm20608g that's able to reproduce the problem (after running continuously for quite a while). Debugging...

dagar, 2020-05-31:
@taileron could you give https://github.com/PX4/Firmware/pull/15004 a try?

Binary: https://github.com/PX4/Firmware/suites/739248622/artifacts/7483749
https://github.com/PX4/Firmware/actions/runs/120856803

taileron, 2020-06-01:
the old R12 has a 20608g as well. #15004 doesn't change the behavior of imu 0 but sometimes emu 1 fails too. the FC are connected via USB but via wifi and power module it´s identical.
Sometimes there are 70 failsafe events around the start procedure. 
https://review.px4.io/plot_app?log=8fbbe75b-7ef2-44b4-8f0f-92b4838c93ee
Imu rate max = 400
https://review.px4.io/plot_app?log=324c18ec-09a6-400c-8ed2-64adb6e875e8
imu rate max = 0 (789.7Hz)

The R14 currently works with emu_reset branch

dagar, 2020-06-01:
I can confirm the icm20608g data is still bad in your R12 logs and now looks good in the R14. What's noteworthy is that it's bad initially, and not the same as what I thought might be a similar problem with the ICM20689 going bad later.

Next for the R12 I'll look at implementing the icm20608g self test procedure.

> is there a trick to edit "extras.txt" over mavlink, cat /fs/microsd/etc/extras.txt only shows it ?
Not that I'm aware of, but I haven't tried it. If this isn't working please open a separate issue and we'll take a look at what's missing in the Mavlink FTP code.

taileron, 2020-06-04:
Glued Pixracer R14 board now flies great (mission still needs a bit of tuning to avoid oscillations in strong wind) missing entries could be solved within parameters:
https://review.px4.io/plot_app?log=fc50db22-3070-4083-9090-a0a0d62d257f

How can I proceed further to get V1.11 running on the R12 board, which is my only one that is not damaged by the makeshift repair.

dagar, 2020-06-05:
Could you please try the current https://github.com/PX4/Firmware/pull/15004 again? I believe I found the problem with older ICM20608g units.

taileron, 2020-06-05:
works now .... calibration finished without errors
listener sensor_gyro

TOPIC: sensor_gyro 2 instances

Instance 0:
 sensor_gyro_s
	timestamp: 489968009  (0.009620 seconds ago)
	timestamp_sample: 489967769  (240 us before timestamp)
	device_id: 3801098 (Type: 0x3A, SPI:1 (0x00)) 
	x: -0.0038
	y: 0.0057
	z: -0.0024
	temperature: 53.9841
	error_count: 0

Instance 1:
 sensor_gyro_s
	timestamp: 489987386  (0.009146 seconds ago)
	timestamp_sample: 489987152  (234 us before timestamp)
	device_id: 2359306 (Type: 0x24, SPI:1 (0x00)) 
	x: -0.0109
	y: -0.0141
	z: -0.0040
	temperature: 50.2329
	error_count: 0

dagar, 2020-06-06:
> #15004 works now on R12 .... calibration finished without errors - are there further test needed?

Nothing further here, but please continue testing master and reporting issues as we close in on a v1.11.0 release. 

Thanks for the help debugging this.

closed 2020-06-06 by dagar